classicrocker883/MRiscoCProUI #142: [BUG]  Bed collision - Micro bed probe 2 not detecting bed

6flip9, 2024-06-08:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

When probing bed with Micro Probe v2 the probe is not detecting bed leading to bed collision.  Probe can manually stow and probe without issue but detection is not occuring during probe of bed. 

Aquila_GD32_BMP_BLT-ProUI-EX-05-23.bin

### Printer Model

Voxelab Aquila

### Model Type

original 

### Your Mainboard

Aquila GD32

### Other Mainboard Type

ba

### Add-ons that could be involved

Biqu Micro probe v2 

### ProUI-EX | Extra Features?

Yes

### Bed Leveling

BLT - BL Touch Bilinear mesh

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

n/a

classicrocker883, 2024-06-09:
where do you have it plugged in? the 2-pins (I believe they are black/white) must go where the Z-min old Z endstop switch connector, **not** in the same row as the BL-touch connector on the mainboard.

classicrocker883, 2024-07-02:
is this https://github.com/MarlinFirmware/Marlin/issues/27148 related to your issue?

closed 2025-02-23 by github-actions[bot]