classicrocker883/MRiscoCProUI #143: [BUG] Mesh inset settings do nothing

Arson31, 2024-06-11:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

Updated to MAY 2024 Rev.2 build. (Aquila_GD32_MM-ProUI-EX-05-23.bin)

I set the X maximum to 199.0 because I'm using a Satsana shroud for the Haldis Red Lizard Q1 like I have with the previous builds. After setting and saving the settings the shroud still crashes onto the LCD screen mounted vertically to the right of my Aquila.

I set the X maximum to 160 (which should be dead center) and the shroud still crashes into the screen. 

### Printer Model

Voxelab Aquila

### Model Type

OG

### Your Mainboard

Aquila GD32

### Add-ons that could be involved

Satsana shroud for the Haldis Red Lizard Q1

### ProUI-EX | Extra Features?

Yes

### Bed Leveling

MM - Manual Mesh Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2024-06-12:
how about the Physical Settings, under the Control menu? shouldn't this be adjusted instead of the Mesh Inset?

Arson31, 2024-06-14:
@classicrocker883 I set the X max on physical settings to 180, stored settings, and it still crashes into the lcd screen.

Arson31, 2024-06-15:
@classicrocker883 The mesh X maximum was set to 199.0 in the Mesh Inset settings so I guess that's the limit. It never crashed into the LCD screen when I set the mesh X maximum to 199.0; I was able to create the mesh in the past.

classicrocker883, 2024-06-16:
I had made some changes regarding mesh_inset code, how it is defined. I was able to test with probe firmware fine, so I have a suspicion  of what it can be.

could you tell me which firmware release *did* work fine last that you know?

Arson31, 2024-06-18:
@classicrocker883 I think it was the November release of last year, I couldn't remember it's been a while since I updated the firmware.

I'm using a 3x3 mesh and I tested [Aquila_GD32_MM-ProUI-EX-test.zip](https://github.com/user-attachments/files/15857408/Aquila_GD32_MM-ProUI-EX-test.zip). The 1x3 point of the mesh (the last X point until it begins the next Y set) is at the middle of the bed. The 2x3 is maybe 1 inch or 1/2 inch away from the 1x1 which the 1x1 point is right where the X home is.

Adjusting the X maximum mesh inset to 220 (when it was 210) does not make a difference, the 1x3 point is still at the middle of the bed.

classicrocker883, 2024-06-18:
okay, give this one a try, that should do it.

[Aquila_GD32_MM-ProUI-EX-test2.zip](https://github.com/user-attachments/files/15880394/Aquila_GD32_MM-ProUI-EX-test2.zip)

Arson31, 2024-06-18:
@classicrocker883 Same. The 3rd probe point is at the middle of the bed.

Arson31, 2024-06-19:
@classicrocker883 Neither of the two made a difference.
[Here's](https://github.com/classicrocker883/MRiscoCProUI/assets/5616267/29506e8f-c7cd-49bf-9a83-b55f9a268d1f) a quick video of what I'm talking about. It doesn't probe the entire bed just half of it.

vw72, 2024-06-21:
I don't know if this helps, In the physical settings menu, I set the  max bed size to 230  for both x and y and the bed size to 220 for both of them.  I then set the mesh inset to 10.0 for each.  I'm  using a bltouch, not manual, but maybe these settings will have the head move to the right positions for the mm.

classicrocker883, 2024-06-22:
I'm not sure what's going on, I will have to perform some tests to figure it out. because I had updated the mesh inset code, I tried reverting to what it was originally, so i'll have to do some git bisect I think all the way back to November, so thats gonna take some time, trial and error.

looking back and comparing the code didn't give much clue. it is just odd how the mesh is appearing to be almost exactly half of what it should be.

for now I suppose either use the last build release which worked, or work with what you got. because technically with a good and level bed you should not need any active bed leveling, that is even if there is slight variation +/-0.10, that isn't really significant enough to effect the first layer. I know with mine I can turn active leveling off and there's no noticeable difference.

just hang in there i'll try and figure this out.

Arson31, 2024-06-22:
@classicrocker883 It could've been either the November or October release, they may have the same logic so I doubt it makes a difference.

RCP1990, 2024-07-20:
Im having a similar issue when using the June UBL PRO UI. my Aquila has an ender extender kit so my bed is 400mm x 400mm. Ive set my physical settings and mesh settings to 400mm. When I home my Z axis, it correctly goes to the center of the bed but when I try to make a mesh, it probes based on the default ~220X220 bed size instead.

classicrocker883, 2024-07-21:
@RCP1990 have you used an October or November version? if so, did it work normally then?  
also, do you experience the same kind of result how it is half of the bed, or is it a bit different?

that is interesting since it is UBL, yet works fine on default bed sizes.

RCP1990, 2024-07-21:
I have not. ill install the November and test it out.

classicrocker883, 2024-07-22:
what .bin file do you use? chip and version and such? because I may have found something and I can provide and updated firmware for you to test.

classicrocker883, 2024-07-22:
[GD32-MM-ProUI-EX-July-Test1.zip](https://github.com/user-attachments/files/16334147/GD32-MM-ProUI-EX-July-Test1.zip)
@Arson31 I made another if you'd like to try to see if this works now

RCP1990, 2024-07-23:
@classicrocker883 UBL works as expected on the November release. (im using a G32 Aquila and the UBL Pro UI 7x7 bin, for the record)

Arson31, 2024-07-25:
@classicrocker883 Same problem with 3x3 mesh as [#143](https://github.com/classicrocker883/MRiscoCProUI/issues/143#issuecomment-2179412186)

classicrocker883, 2024-07-26:
@Arson31 @RCP1990 thanks for confirming. well I uploaded updated releases for July [2.1.3f-7](https://github.com/classicrocker883/MRiscoCProUI/releases/2.1.3f-7) here. 

I looked back and reverted the things which had to of caused what started the issue. 
if you try the new build let me know if this has solved it

Arson31, 2024-07-28:
@classicrocker883 No change with [JULY 2024 2.1.3f-7](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3f-7).

classicrocker883, 2024-07-28:
well I hate to say it, but it would be great it you could try like the December, January, Feb... and so on, from the working November (or is it November Rev. 1?) until you get to the one that doesn't work. that would be super helpful.

even after comparing then from now I still cannot find why it would the mesh would be like in half

classicrocker883, 2024-07-28:
I really appreciate the help trying to solve this. I made just one more test file which is really optional, it would be much better going month by month releases and letting me know the point it goes from working to not working.

[GD32-MM-ProUI-EX-July-Test2.zip](https://github.com/user-attachments/files/16403217/GD32-MM-ProUI-EX-July-Test2.zip)

This is a separate test which I took November and copied the current **dwin.cpp** (main UI display file). this should work - but if this *doesn't* work then it will mean its something in this file and should narrow it down.

actually it's not exactly the November release, but close enough.

again at this point there has been lots of changes since November so finding where one release works and the next doesn't should really help solve the issue.

Arson31, 2024-07-28:
@classicrocker883 The [GD32-MM-ProUI-EX-July-Test2.zip](https://github.com/user-attachments/files/16403217/GD32-MM-ProUI-EX-July-Test2.zip) had no change. The [February Rev. 1 build](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3f-2-1) probes the entire bed properly using the 3x3 like it had in the past.

classicrocker883, 2024-07-30:
so youre saying Test2 works as it should, or no?

here is another test which I added the workaround for the mesh inset, which is present in November, and that I had previously removed thinking I solved the issue before.
[GD32-MM-ProUI-EX-July-Test3.zip](https://github.com/user-attachments/files/16422713/GD32-MM-ProUI-EX-July-Test3.zip)

if this works then thats gotta be it.

Arson31, 2024-08-03:
@classicrocker883 Sorry for the wait, [GD32-MM-ProUI-EX-July-Test3.zip](https://github.com/user-attachments/files/16422713/GD32-MM-ProUI-EX-July-Test3.zip) works as it should.

classicrocker883, 2024-08-17:
so I uploaded new releases, this should be fixed!

Arson31, 2024-08-25:
@classicrocker883 I flashed the latest firmware today and the issue is still there. [GD32-MM-ProUI-EX-July-Test3.zip](https://github.com/user-attachments/files/16422713/GD32-MM-ProUI-EX-July-Test3.zip) and the [February Rev. 1 build](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3f-2-1) works.

classicrocker883, 2024-08-30:
@Arson31 so here is a test  
[GD32-MM-ProUI-EX-August-Test1.zip](https://github.com/user-attachments/files/16817373/GD32-MM-ProUI-EX-August-Test1.zip)

so for some reason the original struct doesn't register the values, and I just incorporated the workaround from February which defines the `MESH_MIN/MAX_X/Y` insets to the eeprom (**settings.cpp**) directly.

Arson31, 2024-08-30:
@classicrocker883 same issue with 3x3 mesh with https://github.com/classicrocker883/MRiscoCProUI/issues/143#issuecomment-2179412186.

classicrocker883, 2024-08-31:
So here are a couple more,  
- Test2 has the older **settings.cpp** structure
- Test3 I tweaked the same/recent **settings.cpp** which I noticed a couple things wrong, and bumped the EEPROM version
Hopefully either one does work, otherwise I would have to go back and try adding the old struct back and previous meshinset workaround.

[GD32-MM-ProUI-EX-August-Test2.zip](https://github.com/user-attachments/files/16824450/GD32-MM-ProUI-EX-August-Test2.zip)
[GD32-MM-ProUI-EX-August-Test3.zip](https://github.com/user-attachments/files/16824451/GD32-MM-ProUI-EX-August-Test3.zip)

Arson31, 2024-09-01:
@classicrocker883 I'll test when I come back Wednesday.

Arson31, 2024-09-05:
@classicrocker883 
[GD32-MM-ProUI-EX-August-Test2.zip](https://github.com/user-attachments/files/16824450/GD32-MM-ProUI-EX-August-Test2.zip) did not work.
[GD32-MM-ProUI-EX-August-Test3.zip](https://github.com/user-attachments/files/16824451/GD32-MM-ProUI-EX-August-Test3.zip) works.

classicrocker883, 2024-09-10:
Test3 working is good news. I will be posting a new release soon, working on a couple things - extra firmware compilations.

classicrocker883, 2024-09-19:
@Arson31 
got a new release:
https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3f-9

if you can let me know if that is also working as it should

Arson31, 2024-09-19:
@classicrocker883 the problem's still there.

classicrocker883, 2024-09-21:
ok, so you sure it was Test3 from earlier that worked? because that is what I went with for changes

Arson31, 2024-09-21:
@classicrocker883 Ok, something strange is happening. When I tested the two firmwares 2 weeks ago:
#

> [GD32-MM-ProUI-EX-August-Test2.zip](https://github.com/user-attachments/files/16824450/GD32-MM-ProUI-EX-August-Test2.zip) did not work.
[GD32-MM-ProUI-EX-August-Test3.zip](https://github.com/user-attachments/files/16824451/GD32-MM-ProUI-EX-August-Test3.zip) works.

That was true before, but now I can't get either of them to probe correctly. Here's what I did:

1. Flash the firmware
2. Remove SD card
3. Change X physical settings to 199.0
4. Save setting
5. Level
6. Mesh settings
7. Change mesh points to 3x3
8. Change X Mesh Maximum to 199.0
9. Continue Bed Mesh
10. Doesn't probe correctly
11. Save setting
12. Restart
13. Level
14. Continue Bed Mesh
15. Still doesn't probe correctly

I flashed back to [GD32-MM-ProUI-EX-July-Test3.zip](https://github.com/user-attachments/files/16422713/GD32-MM-ProUI-EX-July-Test3.zip) here's what I did with that:

1. Flash firmware
2. Change X physical settings to 199.0
3. Level
4. Mesh settings
5. Change mesh points to 3x3
6. Change X Mesh Maximum to 199.0
7. Continue Bed Mesh
8. Probes correctly
9. Save settings
10. Remove SD card
11. Restart
12. Level
13. Continue Bed Mesh
14. Probes correctly

If the order I handle these doesn't matter then I don't know how to go about this. Does reformatting the SD card before every flash make a difference? I've never done so with any of these tests but going back to [GD32-MM-ProUI-EX-July-Test3.zip](https://github.com/user-attachments/files/16422713/GD32-MM-ProUI-EX-July-Test3.zip) without reformatting the SD card fixed the probing.

Also the latest release https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3f-9 had violent movements when auto homing. Never had that before.

classicrocker883, 2024-09-22:
here is a new test file. I moved the values to a struct to follow the current head fork of MRiscoCProUI. before it was a simple workaround, this change is better so hopefully it works.
[GD32-MM-ProUI-EX-Sept-Test1.zip](https://github.com/user-attachments/files/17090963/GD32-MM-ProUI-EX-Sept-Test1.zip)
 
>If the order I handle these doesn't matter then I don't know how to go about this. 

it seems like your steps are correct, but it shouldn't matter changing any settings specifically after every flash. you can if you want for consistency's sake, but having to remove the card + restarting wouldn't make a difference.

> Does reformatting the SD card before every flash make a difference? 

The SD card itself doesn't need to be formatted every time. just once ever.

classicrocker883, 2024-09-23:
here is a new test file. I moved the values to a struct to follow the current head fork of MRiscoCProUI. before it was a simple workaround, this change is better so hopefully it works.
[GD32-MM-ProUI-EX-Sept-Test1.zip](https://github.com/user-attachments/files/17090963/GD32-MM-ProUI-EX-Sept-Test1.zip)
 
>If the order I handle these doesn't matter then I don't know how to go about this. 

it seems like your steps are correct, but it shouldn't matter changing any settings specifically after every flash. you can if you want for consistency sake - or 

> Does reformatting the SD card before every flash make a difference? 

The SD card itself doesn't need to be formatted every time. just once ever.

Arson31, 2024-09-23:
@classicrocker883 I tried [GD32-MM-ProUI-EX-Sept-Test1.zip](https://github.com/user-attachments/files/17090963/GD32-MM-ProUI-EX-Sept-Test1.zip) and it did not probe properly. I noticed after restart the steppers react violently, The carriage slams the X switch when auto-homing.

classicrocker883, 2024-09-24:
that is strange. but the previous August Test3 worked the best? no other problems? I might revert back to that, because Marlin should have fixed the issue with the motors shaking. 😔

Mik-S-UMS, 2024-09-25:
I think some of the test code made it into the main release as someone on [Reddit](https://www.reddit.com/r/VoxelabAquila/comments/1fn82lb/aquila_s3_firmware_help/) mentioned the motors moving too fast when homing, and issue #159 appears to be the same thing.

classicrocker883, 2024-10-12:
I've uploaded a new release for October, the issue should be fixed, @Arson31 hopefully it works let me know

Arson31, 2024-10-12:
@classicrocker883 I've tried it and had the same problem as the previous builds. The violent stepper movement went away though. I'm sticking to [GD32-MM-ProUI-EX-July-Test3.zip](https://github.com/user-attachments/files/16422713/GD32-MM-ProUI-EX-July-Test3.zip).

Arson31, 2024-10-18:
@classicrocker883 Same problem for [GD32-MM-ProUI-EX-October-Test1.zip](https://github.com/user-attachments/files/17412745/GD32-MM-ProUI-EX-October-Test1.zip).

classicrocker883, 2024-10-18:
im sorry this doesn't make any sense. how can something work before, and not this time?
thanks for trying out testing.

classicrocker883, 2024-10-18:
[GD32-MM-ProUI-EX-October-Test2.zip](https://github.com/user-attachments/files/17434016/GD32-MM-ProUI-EX-October-Test2.zip)

well I moved where the mesh_inset is defined to a struct like how the recent original MRiscoC is, couldn't find anything else that stood out which would make it not be the correct mesh inset. I know it did work before for UBL so its odd to say why it wouldn't for Manual Mesh, and when it did at one point.

hopefully moving it this way makes a difference otherwise it could be something else happening with Manual Mesh and not with the other bed leveling types.

Arson31, 2024-10-18:
@classicrocker883 [GD32-MM-ProUI-EX-October-Test2.zip](https://github.com/user-attachments/files/17434016/GD32-MM-ProUI-EX-October-Test2.zip) didn't work either.

1. I changed X physical settings to 199.0.
2. Store Settings
3. Leveling
4. Mesh settings
5. Change mesh points to 3x3
6. Change X Mesh Maximum to 199.0
7. Continue Bed Mesh
8. Goes to left most point
9. Next point
10. Moves an inch or 2 to the right
11. Next point
12. Another inch or 2 to the right at the middle of the bed.

I did not remove the SD Card.

None of this may not matter, I'm just listing the steps of what I did exactly.

classicrocker883, 2024-10-19:
[GD32-MM-ProUI-EX-October-Test3.zip](https://github.com/user-attachments/files/17445347/GD32-MM-ProUI-EX-October-Test3.zip)
i think i know what it is. in the proui library theres a function `ManualMesh_Resize`, so this test3 should work

Arson31, 2024-10-19:
@classicrocker883 That did it! It probes correctly now. Thanks!

PR #147 (2024-08-06): Mesh Inset workaround fix #143

### Description
- Fixes #143 

### Requirements

### Benefits

### Configurations

### Related Issues
commit 1467b206 (2024-08-06): Update - Mesh Inset workaround fix

 - Fixes #143
closed 2024-10-19 by Arson31