classicrocker883/MRiscoCProUI #146: [BUG]  Once "Turn off LCD" button is pressed, you will need to press twice to get the ability to scroll.

1020alex, 2024-07-25:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

After pressing the "Turn off LCD" button the screen turns off successfully. However, to get the screen back to normal (and scroll), the button must be pressed twice.

### Printer Model

Voxelab Aquila

### Model Type

OG

### Your Mainboard

Aquila GD32

### ProUI-EX | Extra Features?

Yes

### Bed Leveling

UBL - Unified Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2024-07-26:
I've tried figuring this one out before, well the thing is when the LCD is off, scrolling will turn it back on, but not the ability to scroll / use the display.

You **should** be able to push the button (press encoder knob) when LCD is off and return full functionality, correct?
<-*This* doesn't require pressing twice, right?

classicrocker883, 2024-11-02:
unfortunately however it is now is the best i can do. after LCD goes off, scrolling turns it on but scroll doesn't work unless you press button.

would need to do a lot of trial and error to figure it out

closed 2024-11-02 by classicrocker883