classicrocker883/MRiscoCProUI #149: [BUG] after homing, fast movements with skipped steps

Weresk, 2024-08-25:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

1) Flash new firmware  Version: 2.1.3f-8-1 (on the previous Version: 2.1.3f-7 all OK)
2) Start printing something and after homing movements, maybe they are called travel movements, very fast head jerking with skipping steps (as if huge speeds were set)
[Configuration.zip](https://github.com/user-attachments/files/16741379/Configuration.zip)

### Printer Model

Voxelab Aquila

### Your Mainboard

Aquila GD32

### ProUI-EX | Extra Features?

Yes

### Bed Leveling

MM - Manual Mesh Bed Leveling

### Did you include your own configuration files?

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2024-08-31:
in **settings.cpp**, look for 
```y
      //
      // Motor Current PWM
      //
      #if HAS_MOTOR_CURRENT
```
or anything that has `#if HAS_MOTOR_CURRENT` + it's `#endif` remove those because there is no such `HAS_MOTOR_CURRENT` that is a mistake.

classicrocker883, 2024-11-02:
let me know if you still have an issue with the new release

classicrocker883, 2024-12-20:
id like to say this was related to the same issue, closing

closed 2024-12-20 by classicrocker883