classicrocker883/MRiscoCProUI #150: [FR] Need ability to Raise Acceleration...

inspectionsbybob, 2024-08-30:
### Is your feature request related to a problem? Please describe.

I am trying to "go Fast," and my heavily modified Aquila can easily extrude nicely at 80-100MM/sec, but my acceleration on the last release I have from you is limited to 2000.  Do the newer releases have a higher limit?   Can I request a Higher limit version?

### Are you looking for hardware support?

No

### Describe the feature you want

To be able to raise ACCELERATION on X-Y to 5-10,000 to see where the aquila gives up...

### Additional context - Possible alternatives

None as acceleration is the sticking point...

classicrocker883, 2024-08-30:
from what I can tell these are the max Acceleration values `{ 6000, 6000, 300, 9999 }`, are you sure youre not confused with Steps/MM?

inspectionsbybob, 2024-08-30:
Those are a lot higher than the version I have, I will try it as soon as
the current print finishes...

-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

inspectionsbybob, 2024-08-30:
Wow...
It has changed a LOT and for the Better, much better.  I'm impressed.
I am currently going through calibrations, but it looks really good.

Now I have to figure out how far I can push my Aquila, now that firmware
limits do not constrain it.

A really good Fan (HeroMe) and a 60W CHT heater make a HUGE difference,
along with cleaning and tightening everything.

What Have you gotten your Aquila to do with Tolerable quality?
Bob Sisson

classicrocker883, 2024-08-31:
thank you for the comment. what I have done to my Aquila is add direct drive and linear rails; this really made a difference as I noticed prints have come out more precisely. also added Dual Z axis lead screws, dual cooling fans, upgraded copper plated heating block and CHT nozzle, titanium "all metal" heatbreak, replaced thermistors for hotend+bed w/cartridge style and copper shell for better heat transfer, also improved frame stiffness with brackets and support, placed the machine on top of sound isolation dampers + these vibration dampening blocks and 12" granite tile, and I swapped the X/Y axis motors with bigger motors this improves effectiveness because stock Y motor gets very hot - because the bed has a lot of mass so it only makes sense to have a bigger motor there, and for X axis having direct drive adds weight so these motors run cool now.

yrabbit, 2024-08-31:
A few pictures of this miracle machine?

By the way what kind of motors did you use and did you change the current?

inspectionsbybob, 2024-09-01:
It is a Voxlabs Aquila that I added:
Dual Z - belt (still only single stepper)
Belt tensioners on X & Y
Heat sinks on steppers
60W CHT heater
CHC nozzle
HeroMe Fan housing
500W power supply (leftover from other project)

The CHT heater made the biggest difference. A 40w standard heater was
+/-10degC and sometimes had trouble keeping up at high flows, the CHT
heater is rock solid +/-1degC even before your new (MPC) software. I used
to have random clogs and if I watched the temps, at some point the nozzle
temp would drop, and clog, or blow off a Bowden fitting. Now it needs
K-factor tuning (bulging at turns)

Where I need to tune is acceleration, if I really want to slow things down
I use Gyroid infill on small areas, the hot end can never get any speed and
seems to put down the Gyroid "lazily." Only on long straight sections does
the hotend get any speed... it loves straight sides and triangle infill...

I only occasionally get the "groan" from the steppers on travel moves (high
acceleration and speed), and they are still only warm... resetting the
current up a bit is on the list... somewhere.

-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

yrabbit, 2024-09-01:
> Dual Z - belt (still only single stepper)

Well, I don't see any problems here - they work fine if the thrust plates are installed.

 > Belt tensioners on X & Y

On the Aquila? What's wrong with the standard ones?

> CHC nozzle

CHT? And no difficulty with retraction?

inspectionsbybob, 2024-09-01:
Retraction isn't as good as the Bambu, but rarely more than whisps... most
of that is tuning...

Tuning tuning and more tuning... and sticking with the same filament.  The
bambu
Does something fancy with retractions, but I am close, will likely need
more current to get the acceleration wanted...  at this point it is all
acceleration... my temps, speed, jerk and other things are pretty close...
so now its just acceleration tuning...
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

classicrocker883, 2024-09-11:
> A few pictures of this miracle machine?
> 
> By the way what kind of motors did you use and did you change the current?

sure, sorry for late reply just got back from vacation will post soon, and for motors I used

3-17HS15-1504S-X1
[3pcs Nema 17 Bipolar 45Ncm(63.74oz.in) 1.5A 42x42x39mm 4 Wires](https://www.omc-stepperonline.com/3pcs-nema-17-bipolar-45ncm-63-74oz-in-1-5a-42x42x39mm-4-wires-w-1m-pin-connector-3-17hs15-1504s-x1)

which is basically Nema 17 42-40 aka the Extruder motor.

classicrocker883, 2024-09-24:
here is a picture of my original and my Aries. as you can see I've been doing a lot of changes and stuff to it. still got an X2 that needs to be put together. 
[image]