classicrocker883/MRiscoCProUI #154: [BUG] skr mini e3 v3

Fireball122, 2024-09-11:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

so i have a random question i am using a skr mini e3 v3 on a Voxelab aquila and i have your firmware running on it and my dwin set updated to the right one but whenever i try to click on an icon it freezes and the board restarts. any help would be greatly appreciated

### Printer Model

None

### Your Mainboard

None

### ProUI-EX | Extra Features?

Yes

### Bed Leveling

UBL - Unified Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2024-09-11:
I am not sure, there are different versions for this board, did you grab the right one?

for instance
```
for SKR Mini E3 V3.0.1
Aquila_SKR-Mini-E3-V3-0-1_UBL-ProUI-EX-08-14.bin

for SKR Mini E3 V3.0
Aquila_SKR-Mini-E3-V3_UBL-ProUI-EX-08-14.bin
```

---

otherwise, I do know that to use the DWIN display, it requires a specific cable or a custom one to connect the LCD. double check the pinouts are correct.

```y
  /**
   *        ------                ------                ------
   * (ENT) | 1  2 | (BEEP)       |10  9 |              |10  9 |
   *  (RX) | 3  4 |         (RX) | 8  7 | (TX)      RX | 8  7 | TX
   *  (TX)   5  6 |        (ENT)   6  5 | (BEEP)   ENT | 6  5 | BEEP
   *   (B) | 7  8 | (A)      (B) | 4  3 | (A)        B | 4  3 | A
   *   GND | 9 10 | (VCC)    GND | 2  1 | VCC      GND | 2  1 | VCC
   *        ------                ------                ------
   *         EXP1                  DWIN               DWIN (plug)
   *
   * All pins are labeled as printed on DWIN PCB. Connect TX-TX, A-A and so on.
   */
```

classicrocker883, 2024-09-14:
double check your board version. also make sure connection aren't grounded such as the thermistor. it's possible that a sort of *short* or *grounding* is causing this to happen.

I know the creality 427 boards (at least mine) have this issue when attaching it to under the printer it grounds to the frame and would cause a similar issue. also if a wire grounds or shorts itself it will cause this to reboot.

so definitely check your LCD wire connections.

Fireball122, 2024-09-15:
Thank you for all of your help, and it was my cable, i was using a converting cable that i got with the board and it displayed so i though i had it oriented correctly but i had to flip my cable around (the side that was plugged into the mother board went to the screen) and its working now thank you also for the graph that help me realize my mistake.

Fireball122, 2024-09-15:
:)

closed 2024-09-15 by Fireball122