classicrocker883/MRiscoCProUI #155: [BUG] Printer Blacks Out After Upgrading

The-UnknownHacker, 2024-09-12:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

After installing the firmware the printer screen blacks out fully and is not responding

I have tried everything including upgrading the DWIN lcd firmware and also Trying to revert to the stock firmware but nothing seems to be working

### Printer Model

Voxelab Aquila

### Model Type

Aquilla X2

### Your Mainboard

Aquila HC32

### Add-ons that could be involved

None

### ProUI-EX | Extra Features?

No (formerly NoPro)

### Bed Leveling

Default - No Probe / No Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

The log output command returns nothing

Mik-S-UMS, 2024-09-12:
A black screen means the firmware is not running. Are you sure you installed the right version for your chip? 
Confirm what chip is on your motherboard, re-download the firmware and try again.

The-UnknownHacker, 2024-09-12:
I have done that but now its not taking the firmware  - My chip is HC32 and I downloaded this firmware - 

https://www.mediafire.com/file/jwt7pqkswohuxei/HC32_Default-08-14.bin/file

Mik-S-UMS, 2024-09-13:
Another cause of this is a thermistor problem that prevent the FW from booting up.
Try unplugging the thermistors from the motherboard and it should boot up. If it does then it means you will need to replace that thermistor.

The-UnknownHacker, 2024-09-13:
That still didn't work

Mik-S-UMS, 2024-09-13:
Could you post a picture of your motherboard showing what chip it has?

Is the firmware actually installing? 
It can fail if it has the same filename as what is currently installed so give the .bin file a unique name and make sure it is in the "firmware" folder on the root of the SDcard. If the install is successful the .bin file will be deleted so this is a good way to see if it installed or not. Make sure the SDcard is formatted to FAT32 4096 blocks size too.

The-UnknownHacker, 2024-09-14:
It has an H32 chip and also when I installed it the file got deleted from the sd card but looks like the install is corrupted and it wont boot now

Mik-S-UMS, 2024-09-14:
Try an earlier release just in-case something is wrong with the latest version.

There are 4 lights on the motherboard. Which ones are on and is any flashing? There are labels next to each light.

Just to make sure, Is the screen actually plugged in? You should see the screen light up a bit even when it is black when you turn the printer on.

classicrocker883, 2024-09-14:
try renaming the file to something else. or try a different type like Manual Mesh (*_MM*)

The-UnknownHacker, 2024-09-16:
done that

The-UnknownHacker, 2024-09-16:
still didnt work though

classicrocker883, 2024-09-19:
im working on a new release. I can test the firmware on an HC32 board to make sure its all good. last I checked it did flash fine and no problems.

I would look to see all your connections and wires are good. actually it might help to unplug everything from the board except for the power going in and LCD cable and try flashing it again - with a renamed file or an older version which is known to be good.

Cocoandloco, 2024-12-03:
Hi, I have a voxelab aquila x2 N32, I just install the new december firmware for the printer (this, Aquila_N32_UBL-ProUI-EX-11-30.bin) everything work ok but when I installed the new firmware for the screen (from the file, display assets) and now I have the same problem, the screen is black. I tried to reinstall old firmware and the original from voxelab but nothing works.

closed 2024-12-13 by The-UnknownHacker