classicrocker883/MRiscoCProUI #156: [BUG] SKR 3 E3V3 Hotend Fan not working

danderson255, 2024-09-12:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

When the hotend is heated, the hotend fan never turns on. Fan works and is wired correctly as it runs fine with klipper but not with the pro ui firmware. I've even tried older versions and they have the same issue. It's a v3.0 board with the correct version of the firmware. I've also tried MM and UBL versions to see if it was just one particular build but they all have the same issue with the hot end fan.

### Printer Model

Voxelab Aquila

### Model Type

OG 

### Your Mainboard

BTT SKR MINI E3 v3.0

### ProUI-EX | Extra Features?

Yes

### Bed Leveling

UBL - Unified Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2024-09-14:
forgive me I'm not familiar with the BTT SKR Mini E3 that well. I did not know the Hotend fan had a switch. for an Aquila with a typical board, the hotend fan is always on, and the part cooling fan is adjustable (PWM). this was always assumed. 

so that is a feature then with SKR Mini - the hotend fan can be turned on/off and is speed variable as well??

I will have to look into this to see what pinout is used, perhaps there is an option in Marlin that enables this.

danderson255, 2024-09-15:
No worries, i tried the MRiscoC firmware for this board and its the same
situation and the encoder works in reverse with his build lol.
I know most people likely dont use this board for Marlin due to the display
wiring and usually only use it with Klipper.

Other than the fan issues, i think the rest of the firmware works ok, i'll
have to try bed leveling etc tomorrow.  I'll email you with my results but
yes all fans are able to be turned on and off, but the only one that you
can change the speed of is the part cooling fan.  The motherboard fan also
doesnt seem to kick on, i'll try that again in the morning and see if I'm
remembering correctly or not as to the mb fan working or not.

I appreciate the reply and help in making it work :)
David

danderson255, 2024-09-15:
Hey Andrew,

So i had a chance to check on the fan issues and it seems everything is
good, motherboard fan does work, so i did remember incorrectly, so the only
fan thats not working would be the hotend fan.  I appreciate you helping
fix this as ive only compiled marlin maybe 3 times over the  years and my
memory on how to do so isnt very clear at this point lol.  Im
looking forward to using this firmware with this printer now that the
motherboard has been upgraded/replaced and a few other improvements.

Thank you for your time and help!

danderson255, 2024-09-15:
Oh and the pinout for FAN2 should be PB15.

Thanks again :)

classicrocker883, 2024-09-16:
> Oh and the pinout for FAN2 should be PB15. Thanks again :)

thank you for letting me know, that should help figure things out. so maybe this will help your situation; I'm not sure if this *fix* information is posted somewhere, but with the stock original Aquila and board, the motherboard fan turns on only when the part cooling fan turns on / PWM speed adjustable (perhaps inherited from Ender-3V2 - since being a clone of it), and a simple work around is to splice that fan with the hotend fan since it is always on when the printer is on - to be on the safe side always keeping it cool.

So maybe try this for now, just wire motherboard fan directly to constant power (24v I assume), and hotend fan in place of the motherboard fan. that is, how is that fan controlled/adjustable? like by part cooling fan or something else? otherwise just put both fans to constant power / always on.

that may be a temporary fix while I look for a solution.

classicrocker883, 2024-09-16:
Update
---

I found out how things should be defined in the configuration files. based on a bit of research, this is how the `FAN#` should be:

<!DOCTYPE html>
<html lang="en">
<body>
    <h1>Fan Assignments</h1>
    <table>
        <tr>
            <th>Header name</th>
            <th>Function</th>
        </tr>
        <tr>
            <td>FAN0</td>
            <td>Part cooling</td>
        </tr>
        <tr>
            <td>FAN1</td>
            <td>Hot end cooling</td>
        </tr>
        <tr>
            <td>FAN2</td>
            <td>Electronics enclosure cooling</td>
        </tr>
    </table>
</body>
</html>

[based on this post here](https://www.reddit.com/r/ender3/comments/wc09uz/bigtreetech_skr_mini_e3_v30_fan_connections/)
I supposed "Electronics enclosure cooling" can mean the motherboard cooling fan

<br>

So what I will do is follow these pinouts because that seems to be typical. I'm just thinking it would be easier for other users since this is what they would probably follow, but if you have any changes instead or whatever let me know. I will be posting a new release and this issue should be fixed.
```y
#define FAN0_PIN    PC6   // "FAN0"
#define FAN1_PIN    PC7   // "FAN1"
#define FAN2_PIN    PB15  // "FAN2"
```

danderson255, 2024-09-18:
Im back home now, i'll check this evening when i get to the house for any
new firmware and give it a try and report back my findings.

Thanks for working on this, i really do appreciate it.  This printer is
going to be for my niece and nephew who want to make things for school
projects and stuff like keychains to help them learn about business etc.
Klipper was too confusing for them so rather than removing the board and
putting the stock one back in, i figured this was the best option.

Have a great day

danderson255, 2024-09-19:
> https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3f-9
>
> new release. let me know if the fans work as it should.
> the pinouts should be as follows:
> Fan Assignments
> Header name Function
> FAN0 Part cooling
> FAN1 Hot end cooling
> FAN2 Electronics enclosure cooling

OK, I just got home and loaded it, only changes needed are to swap FAN1 and
FAN2 around.

FAN1=Motherboard/Electronics Enclosure Fan
FAN2=Hotend Cooling Fan
FAN0=Part Cooling Fan

If those are made, i believe it should work perfectly as intended :) I'll
try and swap the wires for now just so i can try and test the firmware and
make sure i dont find any other changes that may need to be made but i dont
think there will be any as i had good luck with the firmware for the GD32
board. I may try and add a BLtouch to this machine if i can find the extra
one i have laying around here somewhere lol.  Anything to help make it
easier on the kids to learn and get started with.

Thank you Andrew, im excited to get this thing finished once the changes
are made.  Hopefully the kids will have fun and make some cool things for
themselves and friends!

David

danderson255, 2024-09-25:
ok well let me know if you're able to make the change, everything seems to
be working find other than the two pinouts needing to be swapped with each
other.

really appreciate all your time and help Andrew!

Have a great week, or whats left of it!

danderson255, 2024-09-30:
Ok, guess you are done replying, sorry to have bothered you, thanks for the
help though.

classicrocker883, 2024-10-01:
?? sorry what change did you need? ive been working on fixing another issue with homing. current firmware just isnt safe to use, but older versions are fine.

danderson255, 2024-10-05:
Sorry. ive been slammed with life this week.. The motherboard fan and the
hotend fan pins need swapped in the firmware to match what BTT uses.  At
the moment its possible to physically swap them but im sure it will be a
issue for anyone else using this SKR board.  I've been so busy ive not been
able to do much with printers this last week or so.  Im hopeful tomorrow
ill get some time to myself to work on this printer again as its almost
done other than the small change in firmware and some  hardware upgrades.

Have a good weekend!

danderson255, 2024-10-14:
Hey! I just wanted to thank you for the update and the changes.  Will be
trying to level and print something tonight and see how the printer is
running.  My sisters kids are gonna be thrilled :)

Thanks again man!

danderson255, 2024-10-14:
Well it appears the homing issue was fixed, but i apparently forgot to
change the wiring back to what it should be so thats why it was working for
my fans lol.  I guess ill just leave it this way for now.  Anyhow, have a
good week.

danderson255, 2024-10-15:
So i checked the source code, you have it set correctly as to what i used
in klipper and what the documentation says to use, but its still backwards,
so idk whats up with this board or maybe its something in marlin somewhere,
but it wouldnt work as intended until i swapped the pins and recompiled,
now itrs working, i dont understand it, its reversed from what klipper and
the documentation says to use in marlin and when installing the board.  So
for what its worth, im sorry for bugging you so much about it, but i got it
working this evening, so im not sure why this board is possessed but it
apparently is LOL.

Have a great week and coming weekend Andrew!

classicrocker883, 2024-10-15:
hey no problem, anytime. so it seems this issue is fixed? as for why it may be backwards, perhaps you have the connectors swapped? with all those wires it can get confusing. well you'll know which fan is which when only one is plugged in. I made the pinouts to what i guess it *should* be. maybe there was something I got wrong in the **Configuration_adv.h** file. if there's anything else I can do on my end let me know.

otherwise i'll close this for now, and happy printing! hope everything else goes well.

closed 2024-10-15 by classicrocker883