classicrocker883/MRiscoCProUI #157: Hot end temperature problem

haw070553, 2024-09-13:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

All works fine until the hot end temperature is reached and the hot end temperature is reached.  As soon as it starts to print, the hot end temperature resets to 0 and the hot end starts to cool down.
If you are quick you can go to tune and reset the tempertaure and then the print continues as normal.

### Printer Model

Voxelab Aquila

### Model Type

X2 

### Your Mainboard

Aquila N32

### Add-ons that could be involved

CR TOUCH 

### ProUI-EX | Extra Features?

Yes

### Bed Leveling

BLT - BL Touch Bilinear mesh

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Mik-S-UMS, 2024-09-14:
Sounds like it could be a slicer problem if the heater is being told to turn off.

Try printing one of the test prints on the SDcard to confirm. 
You should only need to just start to print the first layer to see if there is any difference.
If this starts printing fine then check your slicer settings.

haw070553, 2024-09-14:
Thank you for your suggestion, I will check this out.

May I also take the time to thank you and your team for creating and
improving this software. It is quicker and looks so good on the printer.

Howard.

haw070553, 2024-09-14:
Hi Mik,
I have looked at the Gcode and you are correct.  The problem is in
voxelMaker.  I have changed my slicer and the problem is no more.  Thanks
again for your help

Howard

classicrocker883, 2024-09-14:
this was my guess as well, the Gcode being the issue. closing for now

closed 2024-09-14 by classicrocker883

haw070553, 2024-09-15:
Ok, thanks.