classicrocker883/MRiscoCProUI #162: [BUG] Spontaneous excessive homing feedrate

Ecky-Thump, 2024-10-06:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

After a period of time the homing feed rate will increase dramatically from a cold start.  Only by resetting the settings can the original feed rate be re-acquired, but - again - after a cold start the excessive speeds return.  M210 isn't enabled in the available builds, so I can't ascertain what the final actual excessive feed rates are.

### Printer Model

Voxelab Aquila

### Model Type

OG

### Your Mainboard

Aquila GD32

### ProUI-EX | Extra Features?

Yes

### Bed Leveling

UBL - Unified Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2024-10-10:
I've uploaded a new release for October, the homing issue should be fixed, and some builds do have M210 enabled with a menu option to change these values.

you can find these values here in **Configuration.h**:
`#define HOMING_FEEDRATE_MM_M { (60*60), (60*60), (8*60) }`
X / Y: 3600
Z: 480

have you tried the new build?

Ecky-Thump, 2024-10-10:
Confirmed.  I've been busy putting the the October 2024 build to use and can confirm that the issue is resolved.  Thanks!

closed 2024-10-12 by classicrocker883