classicrocker883/MRiscoCProUI #165: [BUG] NeoPixel enabling with SKR E3V3 shows orror on compile

SrgWhoopsie, 2024-10-27:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

Hi,

trying to compile the latest build source for my X2, 

ive added 5 Neo pixels and wanted to turn on the feature in Config and with doing so i keep running into this error on compile.

Marlin\src\lcd\e3v2\proui\dwin.cpp: In function 'void ApplyLEDColor()':
Marlin\src\lcd\e3v2\proui\dwin.cpp:2558:49: error: cannot convert 'LEDColor' to 'uint32_t' {aka 'long unsigned int'} in assignment
 2558 |     void ApplyLEDColor() { HMI_data.Led_Color = LEDColor({ leds.color.r, leds.color.g, leds.color.b OPTARG(HAS_WHITE_LED, leds.color.w) }); }
      |                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                 |
      |                                                 LEDColor
Compiling .pio\build\STM32G0B1RE_btt\src\src\libs\vector_3.cpp.o
*** [.pio\build\STM32G0B1RE_btt\src\src\lcd\e3v2\proui\dwin.cpp.o] Error 1

any idea why this is happening or how to fix it? 
[Marlin.zip](https://github.com/user-attachments/files/17532784/Marlin.zip)

### Printer Model

Voxelab Aquila

### Model Type

X2

### Your Mainboard

BTT SKR MINI E3 v3.0

### ProUI-EX | Extra Features?

Yes

### Bed Leveling

UBL - Unified Bed Leveling

### Did you include your own configuration files?

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2024-10-28:
I ran into a few issues before seeing this error popup. with your configurations, I had to change a few things:
- add `NO_CONTROLLER_CUSTOM_WIRING_WARNING`
- change `define MULTIPLE_PROBING 2`

but I'm not sure what changed as to why this error came up.
I was able to compile correctly by a few things if you wanted to try it yourself:
- add `  LEDColor Led_Color;` to the `HMI_value_t` struct in **src/lcd/e3v2/proui/dwin.h**

in **src/lcd/e3v2/proui/dwin.cpp**
- change `void ApplyLEDColor() { HMI_data.Led_Color` to `void ApplyLEDColor() { HMI_value.Led_Color`
- remove the following from `void DWIN_CopySettingsFrom`
```y
  #if ALL(LED_CONTROL_MENU, HAS_COLOR_LEDS)
    leds.set_color(
      (HMI_data.Led_Color >> 16) & 0xFF,
      (HMI_data.Led_Color >>  8) & 0xFF,
      (HMI_data.Led_Color >>  0) & 0xFF
      OPTARG(HAS_WHITE_LED, (HMI_data.Led_Color >> 24) & 0xFF)
    );
    leds.update();
  #endif
  ```

SrgWhoopsie, 2024-10-29:
Awesome cheers mate, I'll give it a go this afternoon when I get home from work and update this comment.

Thank you

classicrocker883, 2024-10-29:
forgot to mention add `#include "../../../feature/leds/leds.h"` to the top of **dwin.h**

SrgWhoopsie, 2024-10-29:
alright so i just copied your dwin.cpp and dwin.h from the PR branch into my current repo, 

change add NO_CONTROLLER_CUSTOM_WIRING_WARNING
change define MULTIPLE_PROBING 2 
done 

and im getting a heap of errors now lol 

`In file included from Marlin\src\module/stepper/../../feature/../lcd/e3v2/proui/dwin_popup.h:23,
                 from Marlin\src\module/stepper/../../feature/../lcd/marlinui.h:51,
                 from Marlin\src\module/stepper/../../feature/tmc_util.h:25,
                 from Marlin\src\module/stepper/trinamic.h:35,
                 from Marlin\src\module/stepper/indirection.h:73,
                 from Marlin\src\module/stepper.h:48,
                 from Marlin\src\MarlinCore.cpp:51:
Marlin\src\module/stepper/../../feature/../lcd/e3v2/proui/dwin.h:108:24: error: 'Def_Leds_Color' was not declared in this scope; did you mean 'Def_Text_Color'?     
  108 |   LEDColor Led_Color = Def_Leds_Color; // Moved from HMI_data
      |                        ^~~~~~~~~~~~~~
      |                        Def_Text_Color
In file included from y:\hsoj9\documents\github\marlin_\marlin\src\lcd\e3v2/proui/dwin_popup.h:23,
                 from y:\hsoj9\documents\github\marlin_\marlin\src\lcd\marlinui.h:51,
                 from Marlin\src\feature\../module/stepper/../../feature/tmc_util.h:25,
                 from Marlin\src\feature\../module/stepper/trinamic.h:35,
                 from Marlin\src\feature\../module/stepper/indirection.h:73,
                 from Marlin\src\feature\../module/stepper.h:48,
                 from Marlin\src\feature\babystep.cpp:31:
y:\hsoj9\documents\github\marlin_\marlin\src\lcd\e3v2/proui/dwin.h:108:24: error: 'Def_Leds_Color' was not declared in this scope; did you mean 'Def_Text_Color'?   
  108 |   LEDColor Led_Color = Def_Leds_Color; // Moved from HMI_data
      |                        ^~~~~~~~~~~~~~
      |                        Def_Text_Color
In file included from Marlin\src\feature\bedlevel\ubl\ubl.cpp:43:
Marlin\src\feature\bedlevel\ubl\../../../lcd/e3v2/proui/dwin.h:108:24: error: 'Def_Leds_Color' was not declared in this scope; did you mean 'Def_Text_Color'?
  108 |   LEDColor Led_Color = Def_Leds_Color; // Moved from HMI_data
      |                        ^~~~~~~~~~~~~~
      |                        Def_Text_Color
In file included from y:\hsoj9\documents\github\marlin_\marlin\src\lcd\e3v2/proui/dwin_popup.h:23,
                 from y:\hsoj9\documents\github\marlin_\marlin\src\lcd\marlinui.h:51,
                 from Marlin\src\feature\../module/stepper/../../feature/tmc_util.h:25,
                 from Marlin\src\feature\../module/stepper/trinamic.h:35,
                 from Marlin\src\feature\../module/stepper/indirection.h:73,
                 from Marlin\src\feature\../module/stepper.h:48,
                 from Marlin\src\feature\runout.h:31,
                 from Marlin\src\feature\host_actions.cpp:37:
y:\hsoj9\documents\github\marlin_\marlin\src\lcd\e3v2/proui/dwin.h:108:24: error: 'Def_Leds_Color' was not declared in this scope; did you mean 'Def_Text_Color'?   
  108 |   LEDColor Led_Color = Def_Leds_Color; // Moved from HMI_data
      |                        ^~~~~~~~~~~~~~
      |                        Def_Text_Color
In file included from Marlin\src\feature\bedlevel\ubl\../../../lcd/e3v2/proui/dwin_popup.h:23,
                 from Marlin\src\feature\bedlevel\ubl\../../../lcd/marlinui.h:51,
                 from Marlin\src\feature\bedlevel\ubl\ubl_G29.cpp:33:
Marlin\src\feature\bedlevel\ubl\../../../lcd/e3v2/proui/dwin.h:108:24: error: 'Def_Leds_Color' was not declared in this scope; did you mean 'Def_Text_Color'?       
  108 |   LEDColor Led_Color = Def_Leds_Color; // Moved from HMI_data
      |                        ^~~~~~~~~~~~~~
      |                        Def_Text_Color
*** [.pio\build\STM32G0B1RE_btt\src\src\MarlinCore.cpp.o] Error 1
*** [.pio\build\STM32G0B1RE_btt\src\src\feature\babystep.cpp.o] Error 1
*** [.pio\build\STM32G0B1RE_btt\src\src\feature\bedlevel\ubl\ubl.cpp.o] Error 1
In file included from y:\hsoj9\documents\github\marlin_\marlin\src\lcd\e3v2/proui/dwin_popup.h:23,
                 from y:\hsoj9\documents\github\marlin_\marlin\src\lcd\marlinui.h:51,
                 from Marlin\src\feature\../module/stepper/../../feature/tmc_util.h:25,
                 from Marlin\src\feature\../module/stepper/trinamic.h:35,
                 from Marlin\src\feature\../module/stepper/indirection.h:73,
                 from Marlin\src\feature\../module/stepper.h:48,
                 from Marlin\src\feature\controllerfan.cpp:28:
y:\hsoj9\documents\github\marlin_\marlin\src\lcd\e3v2/proui/dwin.h:108:24: error: 'Def_Leds_Color' was not declared in this scope; did you mean 'Def_Text_Color'?   
  108 |   LEDColor Led_Color = Def_Leds_Color; // Moved from HMI_data
      |                        ^~~~~~~~~~~~~~
      |                        Def_Text_Color
In file included from Marlin\src\feature\../lcd/e3v2/proui/dwin_popup.h:23,
                 from Marlin\src\feature\../lcd/marlinui.h:51,
[... 64 lines cut ...]
  108 |   LEDColor Led_Color = Def_Leds_Color; // Moved from HMI_data
      |                        ^~~~~~~~~~~~~~
      |                        Def_Text_Color
*** [.pio\build\STM32G0B1RE_btt\src\src\feature\powerloss.cpp.o] Error 1
*** [.pio\build\STM32G0B1RE_btt\src\src\gcode\bedlevel\ubl\M421.cpp.o] Error 1
*** [.pio\build\STM32G0B1RE_btt\src\src\feature\runout.cpp.o] Error 1
[... cut ...]
y:\hsoj9\documents\github\marlin_\marlin\src\lcd\e3v2/proui/dwin.h:108:24: error: 'Def_Leds_Color' was not declared in this scope; did you mean 'Def_Text_Color'?   
  108 |   LEDColor Led_Color = Def_Leds_Color; // Moved from HMI_data
      |                        ^~~~~~~~~~~~~~
      |                        Def_Text_Color
*** [.pio\build\STM32G0B1RE_btt\src\src\gcode\calibrate\G28.cpp.o] Error 1`

downloaded the new branch for the PR, pasted my config and config_a to it, commented out the wiring warning and it works 🙌 so i guess theres something in my files thats wonky 

thanks for the bud, greatly appreciate it

classicrocker883, 2024-10-29:
in **dwin_defines.h** change:
```diff
- #if ALL(LED_CONTROL_MENU, HAS_COLOR_LEDS) && DISABLED(LED_COLOR_PRESETS)

+ #if ALL(LED_CONTROL_MENU, HAS_COLOR_LEDS)
```

PR #166 (2024-11-01): Fix for LED

### Description
fixes #165 

### Requirements

### Benefits

### Configurations

### Related Issues
closed 2024-11-01 by classicrocker883