classicrocker883/MRiscoCProUI #167: [BUG] Unable to Home Z-Axis

rannick1982, 2024-11-02:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

When trying to home the Z-Axis withe Biqu MicroProbe, the printer simply climbs the Z rail 10mm at a time.  It then treats the previous position as the 0 point, and I am unable to go below it to get back down to bed.

### Printer Model

Voxelab Aquila

### Model Type

Aquila OG (GD32)

### Your Mainboard

Aquila GD32

### Add-ons that could be involved

Biqu MicroProbe

### ProUI-EX | Extra Features?

Yes

### Bed Leveling

BLT - BL Touch Bilinear mesh

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Mik-S-UMS, 2024-11-02:
This behaviour means the ABL is not connected. 

Check your wiring and make sure the version you installed has _BMP in the filename for the BIQU MicroProbe to work.

rannick1982, 2024-11-02:
Aquila_GD32_BMP_BLT-ProUI-EX-10-27.bin is the firmware I downloaded.  The wiring, from what I can find, is Brown, Red, Yellow, Black, White on to the header.  I have also updated the DWIN firmware, which is awesome btw.  :)

classicrocker883, 2024-11-03:
that is odd behavior. if you can provide a video clip that will help. but when you select to Home, the Z axis is supposed to rise 10mm before anything else - to move the hotend out of way.

you say it rises 10mm at a time? I'm not sure how you mean. it would be a good idea to double check the wiring. because IIRC, the white and black go to the Z-axis connector (where the limit switch went)

rannick1982, 2024-11-03:
> that is odd behavior. if you can provide a video clip that will help. but when you select to Home, the Z axis is supposed to rise 10mm before anything else - to move the hotend out of way.
>
> you say [...]

I can give moving the black and white to the Z-Stop.  I certainly didn't give that a try, as "most" of the documentation I found online told me to wire it up with the BYR.  :)  Thank you for giving that as a suggestion!  And yes, I'll post a video of the behavior, if it persists after moving the wires.  :)

rannick1982, 2024-11-03:
Ok, yes, I was a big dumb, and I should have put the Black and White to the Z-stop.  It's working perfectly now, sorry about that, and thanks for the help!

closed 2024-11-03 by rannick1982