classicrocker883/MRiscoCProUI #168: [BUG]  Marlin EEPROM Editor Octoprint Plugin not connecting since October 2024 release

bdleung, 2024-11-18:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

Octoprint's Marlin EEPROM editor plugin (https://github.com/cp2004/OctoPrint-EEPROM-Marlin) v3.3.0 does not connect to versions of firmware later than October 2024 / 2.1.3f-10 (filename: Aquila_N32_UBL-ProUI-EX-10-09.bin). 

I tried the newer builds (Nov 2024 rev1, Nov 2024, Oct 2024 rev1) and the only one that connects is the October 2024 (2.1.3f-10) one.

The M503 command in the Octoprint terminal did not seem to return any useful data in the firmware versions that did not work with the EEPROM editor plugin.

### Printer Model

Voxelab Aquila

### Model Type

Aquila OG

### Your Mainboard

Aquila N32

### ProUI-EX | Extra Features?

Yes

### Bed Leveling

UBL - Unified Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2024-11-19:
after comparing October and October Rev.1, the following changes stand out:
- `HOST_KEEPALIVE_FEATURE` is Enabled in Rev.1
- - may cause issue with serial - Octoprint
<br>

- Changed toolchain:
```diff
- platform_packages = toolchain-gccarmnoneeabi@1.70201.0
+ platform_packages = toolchain-gccarmnoneeabi@1.90201.191206
```
- - bumped version because it took thousands of bytes on final file size (able to fit much more options)

bdleung, 2024-11-19:
> after comparing October and October Rev.1, the following changes stand out:
...
> i think it could be that, ill upload a test file with changes if you want to try out.

Great - thankyou so much! I'm more than happy to test and report back.

classicrocker883, 2024-11-19:
[N32F103RC_voxelab_Tests.zip](https://github.com/user-attachments/files/17809488/N32F103RC_voxelab_Tests.zip)

here are 2 test files

they both have reverted to `platform_packages = toolchain-gccarmnoneeabi@1.70201.0` like in October 

in **settings.cpp** (the main EEPROM file), i found a change:
```y
bool MarlinSettings::load() {
    #if ANY(EEPROM_AUTO_INIT, EEPROM_INIT_NOW)
      (void)init_eeprom();
```
-Test1:  has this reverted to `(void)save();`
-Test2:  has current `(void)init_eeprom();`

hopefully either or both work.

bdleung, 2024-11-19:
Both test files worked perfectly - can both read the EEPROM and restore a saved backup to the EEPROM.

classicrocker883, 2024-11-19:
thats good news.  but if you dont mind doing 1 more test for me. this time I left everything the same execpt used `platform_packages = toolchain-gccarmnoneeabi@1.90201.191206`

[Aquila_N32_UBL-ProUI-EX-11-19-Test3.zip](https://github.com/user-attachments/files/17813906/Aquila_N32_UBL-ProUI-EX-11-19-Test3.zip)

this will confirm it is that

bdleung, 2024-11-19:
Test3 works perfectly. Was able to both load and save config from/to printer. Thankyou so much for addressing this so quickly! :)

classicrocker883, 2024-11-20:
thats great! thanks for testing. if you don't mind, I have one more test. This should help narrow it down. I would like to confirm the actual issue so I can let either [cp2004/OctoPrint-EEPROM-Marlin](https://github.com/cp2004/OctoPrint-EEPROM-Marlin) or [foosel @OctoPrint](https://github.com/OctoPrint/OctoPrint) know - maybe it can be looked at.

I believe it is `HOST_KEEPALIVE_FEATURE` causing the problem since that is something I add between release versions

the following Tests have everything normal enabled except `HOST_KEEPALIVE_FEATURE` disabled.

[Aquila_N32_UBL-ProUI-EX-11-19-Test4+5.zip](https://github.com/user-attachments/files/17825464/Aquila_N32_UBL-ProUI-EX-11-19-Test4%2B5.zip)

-Test4 is like November Rev.1 except `HOST_KEEPALIVE_FEATURE` disabled.

- Test5 (optional) same as 4 but bumped this:
`platform = ststm32@~15.4.1` to `platform = ststm32@~17.1.0`

if Test4 **works** then its definitely that keepalive feature.
if Test4 **doesn't** work then try Test5, that would mean the toolchain in `platform_packages` is not compatible with `platform` (in the **.ini** - *[env]* variables)

I know all this code stuff may not make sense to you but it helps if I ever need to look back.

classicrocker883, 2024-11-20:
thats a relief, i was afraid i would have to revert more changes. i appreciate the help thank you again

classicrocker883, 2024-11-20:
actually I posted an issue @[cp2004/OctoPrint-EEPROM-Marlin](https://github.com/cp2004/OctoPrint-EEPROM-Marlin)

its is asking for these:
-`M503 Response`
-`API Response`
-`OctoPrint logs`
<sup>Please upload the entire <b>octoprint.log</b> file here.</sup>

which I dont have, but if you can paste them here in a comment and ill put it in the description.
https://github.com/cp2004/OctoPrint-EEPROM-Marlin/issues/81

or you can make a new issue yourself instead if you want

bdleung, 2024-11-20:
which firmware would you like me to generate this data for? The non working Nov rev 1 one?

bdleung, 2024-11-21:
> actually I posted an issue @[cp2004/OctoPrint-EEPROM-Marlin](https://github.com/cp2004/OctoPrint-EEPROM-Marlin)
> 
> its is asking for these: -`M503 Response` -`API Response` -`OctoPrint logs` Please upload the entire octoprint.log file here.
> 
> and OctoPrint version (i put the most recent for now) which I dont have, but if you can paste them here in a comment and ill put it in the description. [cp2004/OctoPrint-EEPROM-Marlin#81](https://github.com/cp2004/OctoPrint-EEPROM-Marlin/issues/81)

The issue does seem to be something to do with the M503 support.

Just redownloaded the relevant firmwares with the firmware selection and reflashed to test M503 functionality. Found that Octoprint plugin didn't like me rolling back firmware versions, restarting Octoprint before each M503 test seem to untangle it.

On Aquila_N32_UBL-ProUI-EX-11-03.bin version:
EEPROM read/write not working
M503 response: 
`Recv: echo:Unknown command: "M503"`

On Aquila_N32_UBL-ProUI-EX-11-20.bin version:
EEPROM read/write not working
M503 response: 
`Recv: echo:Unknown command: "M503"`

On the Aquila_N32_UBL-ProUI-EX-11-20-Test6.bin version:
EEPROM read/write working
M503 response:
```
Send: M503
Recv: echo:; Linear Units:
Recv: echo:  G21 ; (mm)
Recv: echo:; Temperature Units:
Recv: echo:  M149 C ; Units in Celsius
Recv: echo:; Volumetric Extrusion (Disabled):
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E98.00
Recv: echo:; Max feedrates (units/s):
Recv: echo:  M203 X800.00 Y800.00 Z18.00 E120.00
Recv: echo:; Max Acceleration (units/s2):
Recv: echo:  M201 X800.00 Y800.00 Z120.00 E1200.00
Recv: echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
Recv: echo:  M204 P800.00 R800.00 T1000.00
Recv: echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_jerk> Y<max_jerk> Z<max_jerk> E<max_jerk>):
Recv: echo:  M205 B50000.00 S0.00 T0.00 X10.00 Y10.00 Z0.40 E10.00
Recv: echo:; Home Offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Unified Bed Leveling:
Recv: echo:  M420 S0 Z10.00 ; Leveling OFF
Recv: 
Recv: Unified Bed Leveling System v1.01 inactive
Recv: echo:Active Mesh Slot 0
Recv: echo:EEPROM can hold 3 meshes.
Recv: 
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H210.00 B60.00 F0
Recv: echo:  M145 S1 H240.00 B75.00 F0
Recv: echo:; Hotend PID:
Recv: echo:  M301 P13.19 I0.77 D56.05
Recv: echo:; Bed PID:
Recv: echo:  M304 P128.06 I24.95 D438.07
Recv: echo:; Display Sleep:
Recv: echo:  M255 S10 ; (minutes)
Recv: echo:; LCD Brightness:
Recv: echo:  M256 B100
Recv: echo:; Power-Loss Recovery:
Recv: echo:  M413 S0 ; OFF
Recv: echo:; Retract (S<length> F<feedrate> Z<lift>):
Recv: echo:  M207 S5.00 F2400.00 W13.00 Z0.20
Recv: echo:; Recover (S<length> F<feedrate>):
Recv: echo:  M208 S0.00 F2400.00 W0.00 R480.00
Recv: echo:; Homing Feedrate:
Recv: echo:  M210 X3600.00 Y3600.00 Z480.00
Recv: echo:; Probe Offset:
Recv: echo:  M851 X-45.00 Y-7.00 Z0.00 ; (mm)
Recv: echo:; Nonlinear Extrusion:
Recv:   M592 A0.00 B0.00 C1.00
Recv: echo:; Hotend Idle Timeout:
Recv:   M86 B0 E0 S360 T180
Recv: echo:; Linear Advance:
Recv: echo:  M900 K0.00
Recv: echo:; Filament load/unload:
Recv: echo:  M603 L10.00 U120.00 ; (mm)
Recv: echo:; Filament runout sensor:
Recv: echo:  M412 S0 D25.00 ; Sensor OFF
Recv: ok
```

classicrocker883, 2024-11-21:
I'm actually a bit confused by this.

So this was taken from the **November** release? 
`HOST_KEEPALIVE_FEATURE`: *Enabled*
>On Aquila_N32_UBL-ProUI-EX-11-03.bin version:
EEPROM read/write not working
M503 response:
`Recv: echo:Unknown command: "M503"`
---

This taken from the newer **November Rev.1**?
`HOST_KEEPALIVE_FEATURE`: *Disabled*
>On Aquila_N32_UBL-ProUI-EX-11-20.bin version:
EEPROM read/write not working
M503 response:
`Recv: echo:Unknown command: "M503"`
---

This is basically no difference than the previous **November Rev.1** - except for a couple changes which may fix the issue while having:
`HOST_KEEPALIVE_FEATURE`: *Enabled*
>On the Aquila_N32_UBL-ProUI-EX-11-20-Test6.bin version:
EEPROM read/write working
M503 response:
`Recv: echo:; Linear Units:` + ect... (a normal `M503` response`)

classicrocker883, 2024-11-21:
*-Test-6.bin* seems to be working like it should - which is having `HOST_KEEPALIVE_FEATURE` Enabled

so just to be clear, the most recent latest release **November Rev.1** doesn't work like it should??

this is what I am confused about, because as you've said before
>Test4 works fine for reading/writing EEPROM.

*-Test4* has HOST_KEEPALIVE_FEATURE disabled - which seems to solve the issue.
and
>restarting Octoprint before each M503 test seem to untangle it.

does this mean it is the plugin or perhaps Octoprint itself that is the issue?

bdleung, 2024-11-21:
> _-Test-6.bin_ seems to be working like it should - which is having `HOST_KEEPALIVE_FEATURE` Enabled plus those couple changes I made.

Yes, correct.

> so just to be clear, the most recent latest release **November Rev.1** doesn't work like it should??

Yes, correct. I was surprised myself but it looks like the Octoprint plugin GUI that I was using to test read/write to the EEPROM seems to get confused when changing versions, hence me changing my testing strategy to restart Octoprint each time after a change (which clears the GUI and related browser caching).

I will download all your test binaries above again (test 1,2,3... etc) and test for M503 support manually via the terminal and report back as the GUI test is giving confusing reports. 

It takes longer to test this way but I'll aim to do it within the next 24-48 hours so we have a conclusive testing set. Apologies for the confusion, I only discovered the GUI giving conflicting results when downgrading fw today.

bdleung, 2024-11-23:
I just re-ran all tests from the working firmware to the latest and tested in terminal to see if M503 was working. 

I also tested whether the EEPROM editor could connect to the printer.

To do this, I found I had to restart Octoprint after each upgrade and prior to each EEPROM plugin test. Otherwise, the plugin GUI would display as normal (it seems to be cached, based on a previous connection) and allow me to edit values (eg the Z-offset, which I could confirm was saved to the machine by looking at its LCD) when in actual use it would throw the "printer is disconnected" message when first loading the plugin. I'm guessing M503 is called at initial EEPROM plugin startup but it doesn't call it again after the plugin is loaded. 

Octoprint 1.10.3 and EEPROM Editor plugin 3.3.0 was used.

Test procedure:
1) Flash firmware to printer
2) Restart Octoprint
3) Go to Octoprint "Terminal" tab and test for M503 response
4) Go to EEPROM plugin and see if it was able to connect
5) If it can connect, then modify z-offset value and see if it alters on printer's LCD

Here are the results:
Aquila_N32_UBL-ProUI-EX-10-09.bin
M503: OK
EEPROM plugin: OK
Modify value? Yes

Aquila_N32_UBL-ProUI-EX-10-27.bin
M503: Not OK (`Recv: echo:Unknown command: "M503"`)
EEPROM plugin: Cannot connect to printer: (`Looks like the printer is disconnected. Connect to the printer to edit it's configuration.`)
Modify value? No, since cannot access EEPROM plugin

Aquila_N32_UBL-ProUI-EX-11-03.bin
M503: Not OK (`Recv: echo:Unknown command: "M503"`)
EEPROM plugin: Cannot connect to printer: (`Looks like the printer is disconnected. Connect to the printer to edit it's configuration.`)
Modify value? No, since cannot access EEPROM plugin

Aquila_N32_UBL-ProUI-EX-11-18-Test1.bin
M503: OK
EEPROM plugin: OK
Modify value? Yes

Aquila_N32_UBL-ProUI-EX-11-18-Test2.bin
M503: OK
EEPROM plugin: OK
Modify value? Yes

Aquila_N32_UBL-ProUI-EX-11-19-Test3.bin
M503: OK
EEPROM plugin: OK
Modify value? Yes

Aquila_N32_UBL-ProUI-EX-11-19-Test4.bin (Note: this shows Nov 20 on the printer LCD when booting up)
M503: OK
EEPROM plugin: OK
Modify value? Yes

Aquila_N32_UBL-ProUI-EX-11-19-Test5.bin (Note: this shows Nov 20 on the printer LCD when booting up)
M503: OK
EEPROM plugin: OK
Modify value? Yes

Aquila_N32_UBL-ProUI-EX-11-20-Test6.bin
M503: OK
EEPROM plugin: OK
Modify value? Yes

Aquila_N32_UBL-ProUI-EX-11-20.bin
M503: Not OK (`Recv: echo:Unknown command: "M503"`)
EEPROM plugin: Cannot connect to printer: (`Looks like the printer is disconnected. Connect to the printer to edit it's configuration.`)
Modify value? No, since cannot access EEPROM plugin

Please don't hesitate to let me know if you want me to test anything else! :)

Crazy-Charles, 2024-11-23:
Just to add a small note, when M503 reports an "unknown command", so does M115, so i don't think the issue is with the code as such, more like a compiling and linking issue?

classicrocker883, 2024-11-24:
> Just to add a small note, when M503 reports an "unknown command", so does M115, so i don't think the issue is with the code as such, more like a compiling and linking issue?

that may be a factor since the *Test* firmware was compiled directly using VSCode, for all releases I do Github Actions to compile them.

classicrocker883, 2024-11-24:
this one from latest release [Aquila_N32_UBL-ProUI-EX-11-20.bin](https://github.com/classicrocker883/MRiscoCProUI/releases/download/2.1.3f-11-1/Aquila_N32_UBL-ProUI-EX-11-20.bin) has `HOST_KEEPALIVE_FEATURE` disabled and shows that result? 

that is odd because both these have `HOST_KEEPALIVE_FEATURE` enabled:
- *Aquila_N32_UBL-ProUI-EX-10-27.bin*
- *Aquila_N32_UBL-ProUI-EX-11-03.bin*

Well let me confirm something here..
[N32_UBL-HFK-OnOff-Test.zip](https://github.com/user-attachments/files/17891631/N32_UBL-HFK-OnOff-Test.zip)
Here I have two firmwares - one with `HOST_KEEPALIVE_FEATURE` (*-HFK*) enabled (*-On*) and one disabled (*-Off*)

My hypothesis:
- firmware having it disabled/off should work with everything "OK"
- firmware having it enabled/on should give "Not OK"

@bdleung would you mind Testing these two and show results? and thank you for doing that.  
ps. could you tell me what steps are needed if I wanted to test on my own?
so I got an N32 board, raspberry pi with Octopi / Octoprint installed. I just install that plugin and then what?

Side note: 
*Aquila_N32_UBL-ProUI-EX-11-20.bin* has *HKF* disabled and was compiled using Github Actions, the *Test* files using VSCode.

bdleung, 2024-11-24:
You're welcome re: testing, it's the least I can do to help :)

Results:
Aquila_N32_UBL-ProUI-EX-11-20.bin (re-downloaded using your link just now, rather than the one downloaded from the firmware selector - however verified SHA256 hash same for both files)
M503: Not OK (`Recv: echo:Unknown command: "M503"`)
M115: Not OK (`Recv: echo:Unknown command: "M115"`)
EEPROM plugin: Cannot connect to printer: (`Looks like the printer is disconnected. Connect to the printer to edit it's configuration.`)
Modify value? No, since cannot access EEPROM plugin

N32_UBL-HKF-On
M503: OK
M115: OK
EEPROM plugin: OK
Modify value? Yes

N32_UBL-HKF-Off
M503: OK
M115: OK
EEPROM plugin: OK
Modify value? Yes

Steps to test:
1. Flash firmware to board (and verify it applied by checking boot screen and that all settings went to default on LCD)
2. Install the EEPROM plugin on your octoprint install
3. Click on/off icon (top banner of Octoprint GUI) and "Restart OctoPrint" (to force the plugin reload and make sure nothing is cached - this is what was confusing us before)
4. After Octoprint restarts, open new tab in browser and open GUI (again, to ensure nothing is cached)
5. Ensure that printer is connected via "Connection" (top left of Octoprint GUI)
6. On right hand side of Octoprint, click "Terminal" tab and check for M503 or M115 functionality (eg: type 'M503', enter, and see what is output)
7. On right hand side of Octoprint, click hamburger dropdown menu (top right) and select "EEPROM Editor". You'll see all the relevant data of the firmware if it is connecting properly, otherwise an icon with a plug crossed out and an error message
8. If it is connecting properly, I also go to "Offsets" in the plugin and change the "Z probe Z offset" to a random number then click "Save" and verify that the value changes on the printer's LCD.
9. Rinse and repeat for any other firmware version 
10. Have a beer when finished :)

Crazy-Charles, 2024-11-24:
> that may be a factor since the _Test_ firmware was compiled directly using VSCode, for all releases I do Github Actions to compile them.

Which brings me to the second issue I encountered, after downloading your source into VSCode I could not compile any working firmware, each try just produced a blank screen and a non running MCU, I have a spare V1.0.1 board with the NC32 i use for testing and enabled the LED_PIN as PC0 to monitor the MCU heartbeat.  That was until I changed the [STM32F1_maple] environment to match the official Marlin version, i.e. platform and platform_packages, now each build variation works and I have not seen any issues with M503 or M115.
All you need for testing is a board and display, if M503 etc works, so will Octoprint.

classicrocker883, 2024-11-24:
seeing as both **N32_UBL-HKF-On/Off** have same results makes me believe the issue can be related to compiling - Github Actions

- ok got some changes to the **workflows/compile-configs.yml** 
<sup>(no real changes since **November Rev.1**)</sup>
1a. First is just a recompile/reupload of the download assets 
-- [2.1.3f-11-1a](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3f-11-1a)
1b. Second is reverted Python version from 3.11 to 3.10 
-- [2.1.3f-11-1b](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3f-11-1b)
1c. Third is kept Python version 3.11, but reverted other versions (ex. `actions/cache@main`) 
-- [ 2.1.3f-11-1c](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3f-11-1c)

Hypothesis:
I suspect **1a** to not work just like the latest releases - *Aquila_N32_UBL-ProUI-EX-11-20.bin*
I hope either 1b or 1c actually works, because that will confirm some things, but if both work that may not pinpoint the issue.

classicrocker883, 2024-11-24:
> I could not compile any working firmware, each try just produced a blank screen and a non running MCU. That was until I changed the [STM32F1_maple] environment to match the official Marlin version, i.e. platform and platform_packages, now each build variation works and I have not seen any issues with M503 or M115.

that is strange I wonder why it would make things not work when things do work here. the only changes I made to [STM32F1_maple] environment was change `toolchain-gccarmnoneeabi@1.90301.200702` (because it uses MUCH less bytes in the end) and linked to my own `framework-arduinoststm32-maple` - which is to the updated [rogerclarkmelbourne/Arduino_STM32](https://github.com/rogerclarkmelbourne/Arduino_STM32) instead of an older v1.0.0 i think.

are these the changes you reverted back from?

this source code uses `PROUI_EX` - a special library containing code for the ProUI extra features.
could you show me the Configuration files you used?

Crazy-Charles, 2024-11-24:
As expected file Aquila_N32_UBL-ProUI-EX-11-24 from 1a doesn't work, still waiting for the others to appear.

Yes they were the changes to the environment i reverted back. I also used your config files without any changes to first establish a baseline.
After i had a successful build on VSCode and a bootable firmware, i also changed the STSTM32 versions and toolchain to different versions and apart from additional "warnings" appearing, they all built and worked as expected, i have not retried your maple environment but i will do that now and report back.

Crazy-Charles, 2024-11-24:
Well, builds and boots up fine!!!

`PS D:\Github Repositories\MRiscoCProUI-2024-November> platformio run -e N32F103RC_voxelab_maple
Processing N32F103RC_voxelab_maple (board: genericSTM32F103RC; platform: ststm32@~17.1.0; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 (17.1.0) > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32-maple @ 4.10000.240825 (1.0.0) 
 - tool-stm32duino @ 1.0.2 
 - toolchain-gccarmnoneeabi @ 1.90301.200702 (9.3.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 28 compatible libraries
Scanning dependencies...
Dependency Graph
|-- SoftwareSerialM @ 1.0.0
|-- STM32ADC @ 1.0
|-- EEPROM
|-- Wire @ 1.0
|-- USBComposite for STM32F1 @ 1.0.8
|-- Servo(STM32F1) @ 1.1.2
Building in release mode`

Enabled ProUi and that is ok too

Crazy-Charles, 2024-11-24:
Aquila_N32_UBL-ProUI-EX-11-24 from 1b DOESN'T work

Crazy-Charles, 2024-11-24:
Aquila_N32_Default-MPC-11-24 from 1c DOES work (ProUI-EX version wasn't ready and i'm currently cooking my lunch!)

classicrocker883, 2024-11-24:
- New firmware to test
1d. Kept Python version 3.11, but updated other versions (ex. `actions/cache@main` to `actions/cache@4.1.2`
basically like *1c* but bumped versions to latest (`actions/cache@4.0.2` => `actions/cache@4.1.2`)
-- [2.1.3f-11-1d](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3f-11-1d)

just would like to see this work like *1c*, perhaps an issue with arbitrary naming version @main?? rather than the actual release version.

Crazy-Charles, 2024-11-24:
Aquila_N32_Default-MPC-11-24 from 1d DOES work

classicrocker883, 2025-02-06:
hi checking in because issue #196 seems to be the exact same thing that went on here, and I thought we had this fixed. can someone confirm whether or not this issue is back with the latest release?

bdleung, 2025-02-06:
> hi checking in because issue [#196](https://github.com/classicrocker883/MRiscoCProUI/issues/196) seems to be the exact same thing that went on here, and I thought we had this fixed. can someone confirm whether or not this issue is back with the latest release?

I'm currently experiencing a multi day power outage at home, with power not set to be reconnected until mid next week. Will test and update this thread as soon as I can!

bdleung, 2025-02-12:
results from test

fw: Aquila_N32_UBL-ProUI-EX-01-30.bin
M503: Not OK (Recv: echo:Unknown command: "M503")
M115: Not OK (Recv: echo:Unknown command: "M115")
EEPROM plugin: Cannot connect to printer: (Looks like the printer is disconnected. Connect to the printer to edit it's configuration.)
Modify value? No, since cannot access EEPROM plugin

the most recent one I have that's working is Aquila_N32_UBL-ProUI-EX-10-09.bin

classicrocker883, 2025-02-14:
I made up some tests using the same Actions workflow versions which apparently was the cause, been playing around with that if youd like to check out the tests its here https://github.com/classicrocker883/MRiscoCProUI/issues/196

otherwise here they are, check the other issue to see which ones may or may have not worked for the other user
- [Test1](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3g-2-t1)
- [Test2](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3g-2-t2)
- [Test3](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3g-2-t3)
- [Test4](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3g-2-t4)
- [Test5](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3g-2-t5)

bdleung, 2025-02-15:
I just tried Tests1-5 for the Aquila_N32_UBL-ProUI-EX versions of the firmware - none of them worked, with M503 and M115 returning "unknown command"

I did try Test5's Aquila_N32_Default-02-14.bin (MD5: 8a545304382f631ee37df94002400644) - and can confirm M503 and M115 worked for that firmware. I couldn't change the Z probe Z offset setting (I'm guessing because it's a non-BL touch fw) however was able to change other settings using the EEPROM plugin.

bdleung, 2025-02-20:
Looking at the other thread, I can confirm Test7 (Aquila_N32_UBL-ProUI-EX-02-18.bin - MD5 0b037adda7c4d6e64967dbfdb4864bd7)  works for my N32 - M503/M115/EEPROM editor all working!

Test8 does not work, however

classicrocker883, 2025-02-20:
yes, except... we cannot celebrate just yet. this is very odd, odd indeed, because Test7 is **Exactly** as October was  - without any changes or updates. so what I did for Test8 and Test9 was take the same exact `actions/**` versions (except for Test9, I reverted back from _Upload Release Asset_ by **tanyagray/action-upload-release-asset** to my repo copy of hers which I updated - basically hoping perhaps it was my copy of _Upload Release Asset_ (renamed _Upload A Release Asset_).

so I was hoping at LEAST Test 8 would work..... so I have one more idea because during that time they were changing `ubuntu-latest` from 22.04 to 24.04. but that doesn't make sense now since it uses `ubuntu-latest` it is at 24.04 - not the *latest* at the time of release in October.

so the issue cannot be the *compile-configs.yml* in question. there must be some kind of other update which has effected it.

I have an idea about a couple more tests, so if you dont mind stay tuned. I really really do appreciate the help

bdleung, 2025-02-20:
Just waiting for the Aquila_N32_UBL-ProUI-EX-*.bin version of test 10 to compile then I'll do a test!

classicrocker883, 2025-02-21:
it didnt compile because with Input Shaping enabled it went over in bytes so it cancelled out the rest of the builds

[you can see it here](https://github.com/classicrocker883/MRiscoCProUI/actions/runs/13445433977/job/37569670084)
>.pio/build/GD32F103RC_voxelab_maple/firmware-20250220-221703.elf section `.rodata' will not fit in region `rom'
/home/runner/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: region `rom' overflowed by 308 bytes
collect2: error: ld returned 1 exit status

so what I did was delete the old [**Test10**](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3g-2-t10) (tag/2.1.3g-2-t10) and had to disable EXTENDED_CAPABILITIES_REPORT with Input Shaping. once it is done compiling it *should* upload the rest - that is... hopefully there are no other firmware files that do not go over their limited size like before, in which i would just have to find something small/insignificant to disable.

but i got the message over at the other issue and it seems like its working! which is great news. what I would like to do for **Test11** is update all the `actions/**` versions JUST to be sure there are no more issues.

I suspect there aren't going to be anymore and we can finally celebrate lol

bdleung, 2025-02-21:
Test10 is working fine for M503, M115 and EEPROM editor (Aquila_N32_UBL-ProUI-EX-02-21.bin), superb work!

Waiting for Test11 to compile and will confirm after testing.

bdleung, 2025-02-21:
Test11 is GOOD 🥇