classicrocker883/MRiscoCProUI #170: [BUG] the hotend  dose not heat up

dewey2008, 2024-12-10:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

When using your build it dose not stay at 200 C but with the stock build from voxelab it stays at 200 just fine when printing 

### Printer Model

Voxelab Aquila

### Model Type

x3 plus

### Your Mainboard

Aquila N32

### Other Mainboard Type

n/a

### ProUI-EX | Extra Features?

Yes

### Bed Leveling

UBL - Unified Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Mik-S-UMS, 2024-12-10:
Have you done a PID tune yet?

dewey2008, 2024-12-10:
this fixed it also how do i edit the bed size to 300mm  it currently only auto leveling a 235mm x 235mm area

closed 2024-12-11 by dewey2008