classicrocker883/MRiscoCProUI #171: [FR] can you add it so the voxelab x3 plus bed size will print i can only print in a ender 3 size with the firmware flashed onto the printer

dewey2008, 2024-12-11:
### Is your feature request related to a problem? Please describe.

yes i can only print in a 220 x 220  size bed not 300×300×300mm

### Are you looking for hardware support?

N32 Voxelab x3 plus

### Describe the feature you want

working 300×300×300mm area

dewey2008, 2024-12-11:
ok i used the source code and changed the print bed size and built the firmware but i cant get the z axis to go down only up

dewey2008, 2024-12-11:
only thing i changed was the build size and the printer name

Mik-S-UMS, 2024-12-11:
I'm not sure but I think you can change the bed size in the settings.

The Z axis only moving up when homing will be due to the wrong configuration of the inductive sensor as it works differently to a BLtouch.

closed 2024-12-11 by dewey2008