classicrocker883/MRiscoCProUI #173: [BUG] Printer Blacks Out After Upgrading

classicrocker883, 2024-12-20:
Hi, I have a voxelab aquila X2 N32, I just install the new december firmware for the printer, Aquila_N32_UBL-ProUI-EX-11-30.bin, everything work ok (the file was deleted from the SD) but when I installed the new firmware for the screen from the file, display assets DWIN_SET and now I have the same problem, the screen is black. I tried to reinstall old firmware and the original from voxelab web but nothing works. My cpu chip is nation N32G455. the cpu board is FFP0173 and the screen is FFP0173_TFT4.3_V1.0.
The leds on the board: 5V is on, 3V is on, CPU is blinking always and TX led blink one time when I turn on the printer, the BL probe is a 3d Touch and I have unplugged the Z connector (J11 on the board). I see the screen light up just a moment after turn on the printer.

_Originally posted by @Cocoandloco in https://github.com/classicrocker883/MRiscoCProUI/issues/155#issuecomment-2513699585_

classicrocker883, 2024-12-20:
@Cocoandloco is this still an issue even with the latest release?
also, what `kb` size is the file size

Mik-S-UMS, 2024-12-20:
I think the problem is a broken screen as they posted this on [reddit](https://www.reddit.com/r/VoxelabAquila/comments/1h7vhyh/aquila_x2_n32_black_screen_after_update_firmware/) and the board is working correctly when communicating with pronterface and has the same issue when going back to stock FW.

I wonder if there was a bad solder joint on the socket for the ribbon cable on the screen that came loose during the screen update. Everything else checks out.

Cocoandloco, 2024-12-26:
Hola, después de la actualización como comenté, la pantalla dejo de funcionar. Ya no la uso actualmente, estoy haciendo lo básico con pronterface aunque despues de actualizar no he conseguido imprimir correctamente absolutamente nada, pero entiendo que debe ser cosa de ajustar o calibrar bien la impresora (con el firmware original de voxelab me imprimia perfectamente). otra cosa que queria saber es por que no funcionan los comandos en pronterface como por ejemplo M115 para ver la info del firmware.

classicrocker883, 2024-12-28:
BTT SKR Mini E3 V3 needs special modification for the LCD display to work
[see here about it](https://www.smith3d.com/skr-mini-e3-v2-v3-on-ender-3-v2-ender-3-s1-lcd-dwin-knob-screen/)

The file size for the firmware shouldn't be more than ~228KB I believe. - for N32 / GD32 chip boards. anything more than that are for 427 or 422 or something else.

No firmware file for N32 boards should be more than 228KB so I do not know how you got 284KB from

closed 2025-03-02 by github-actions[bot]