classicrocker883/MRiscoCProUI #196: Command M503 does not work on builds compiled from GitHub Actions

natey416, 2025-01-28:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

Attempting to get EEPROM Editor working through Octoprint.
When using the firmware compiled from GitHub actions (Aquila_GD32_UBL-ProUI-EX-12-12.bin), the M503 command sends response ```Recv: echo:Unknown command: "M503"```

When compiling through platformio, copying the configurations found in [configurations/Voxelab Aquila/UBL/](https://github.com/classicrocker883/MRiscoCProUI/tree/2025-January/configurations/Voxelab%20Aquila/UBL), I was able to get the M503 command working.

Here's my marlin configs: [Marlin.zip](https://github.com/user-attachments/files/18566717/Marlin.zip)
Didn't change anything besides the input shaping settings.

Hoping this is just something with the actions bot.

### Printer Model

Voxelab Aquila

### Model Type

Aquila OG

### Your Mainboard

Aquila GD32

### Add-ons that could be involved

CR-Touch, Rasp-pi

### ProUI-EX | Extra Features?

Yes

### Bed Leveling

UBL - Unified Bed Leveling

### Did you include your own configuration files?

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2025-01-28:
some things had to be disabled in order for all the options to fit.
there is one called `MARLIN_SMALL_BUILD` in **Configuration_adv.h** in which disables M503, but it wouldnt be enabled for Aquila_GD32_UBL-ProUI-EX. maybe if it had MPC & INPUT_SHAPING

classicrocker883, 2025-01-28:
I made a [test release here](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3g-1-t). let me know if it still does not work.

do you recall a time [month] that this did start working ?

classicrocker883, 2025-01-28:
https://github.com/classicrocker883/MRiscoCProUI/issues/168

does this issue sound familiar? maybe related?

natey416, 2025-01-28:
I did see #168 when looking up my issue, this could be related, which I why I ended up trying my own config. My configurations are attached in the original post, Everything else is based on 2024-December release.

Using [Aquila_GD32_UBL-ProUI-EX-01-26.bin](https://github.com/classicrocker883/MRiscoCProUI/releases/download/2.1.3g-1-t/Aquila_GD32_UBL-ProUI-EX-01-26.bin), still results in ```Recv: echo:Unknown command: "M503"```

I can do more testing after work today, but that's all I had time for this morning.

classicrocker883, 2025-01-29:
instead of `ubuntu-latest`, ill go the way Marlin went which was to keep 22.04 since they were having issues with the latest (24.04)

Im going to post a January release and hopefully everything else should be fixed

classicrocker883, 2025-01-30:
I changed the workflow so hopefully this fixes the issue   
Posted the [new release here](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3g-1)

natey416, 2025-01-31:
Just downloaded [Aquila_GD32_UBL-ProUI-EX-01-30.bin](https://github.com/classicrocker883/MRiscoCProUI/releases/download/2.1.3g-1/Aquila_GD32_UBL-ProUI-EX-01-30.bin). 
M503 gives the same ```Unknown command: "M503"```
Compiling it works just fine with default UBL aquila config.

If I'm the only one with this issue, it might just be my hardware. There doesn't seem to be a lot of complaints about this.

classicrocker883, 2025-01-31:
how do you get this error? what steps do you take

natey416, 2025-01-31:
I'm switching from the older [Alex/Jyers Firmware](https://github.com/classicrocker883/Marlin-Alex). I use the firmware selector to select the Aquila G32 mainboard, ProUI, Cr/Bl Touch, and UBL. I take this firmware and place it in the ```SDROOT/firmware/``` folder on the printer's SD card. I unplug octoprint from the printer, insert the SD, start up the printer. Firmware automatically updates successfully and the splash screen runs just fine.

Before I touch any settings, I connect octoprint back up, establish connection, then go to the terminal and run the command ```M503```. The error will show then. Running it through pronterface shows the same error. 

This process is the same whether I download from firmware selector or manually pick out the firmware in the release section. Compiling it with the [defaults](https://github.com/classicrocker883/MRiscoCProUI/tree/2.1.3g-1/configurations/Voxelab%20Aquila/UBL) the actions bot should be using works just fine.

classicrocker883, 2025-01-31:
have you tried any of the older firmware files, like [2024-April release](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3f-4) I would suggest.

natey416, 2025-01-31:
I can try those ones out this afternoon when I get off work. I'll let you know which month stops working.

classicrocker883, 2025-02-01:
actually a quick look at the other issue I see this:
>I tried the newer builds (Nov 2024 rev1, Nov 2024, Oct 2024 rev1) and the only one that connects is the October 2024 (2.1.3f-10) one.

but have you tried **October**?
which you can[ find here](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3f-10)

natey416, 2025-02-02:
Apologies for the wait. Got wrapped up in a bunch of stuff yesterday.
Just tried out 2.1.3f-10 October 2024. M503 and EEPROM Editor work as expected!
Attempting 2.1.3f-10-1 through latest results in the same ```Unknown commmand: "M503"```
Everything prior (including 2.1.3f-7 through 2.1.3f-9) seem to work as expected.

classicrocker883, 2025-02-06:
thanks for checking. yes that seems like the same exact issue that was going on. what I had done was change the Actions versions to the `@main` version - meaning it would always be the "latest" - however apparently not a good idea because obviously there seems to be some bugs, so I'm thinking I may have bumped a couple versions (possibly w/bug) since I thought I had fixed it since by reverting the versions. let me check. and in the next release hopefully that should be fixed.

classicrocker883, 2025-02-06:
I made a couple tests if youd like to try out

here is [Test1](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3g-2-t1) which reverts to one change

and here is [Test2](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3g-2-t2) which reverts way back to October

I suspect there may have been something.... I left it at `ubuntu-22.04`. one of these tests *should* work

natey416, 2025-02-06:
Both tests don't seem to work. Same Unknown Command error on both.
I downloaded Aquila_GD32_UBL-PROUI-EX-02-06.bin from both releases

classicrocker883, 2025-02-09:
that is so odd! well I can try using `ubuntu-latest` and one other thing which i wouldnt think affect it but its possible.

here is [Test3](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3g-2-t3) you can try, this fully reverts back to October and should narrow down what the issue is

classicrocker883, 2025-02-10:
I just seen there was an update in one of the versions so I bumped that up maybe that can help fix any bug

here is [Test4](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3g-2-t4) if you'd like to give it a go.

classicrocker883, 2025-02-13:
and here I bumped the versions to the latest, couple other changes  
[Test5](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3g-2-t5)

please let me know if anything works

classicrocker883, 2025-02-17:
There is 
- [Test4](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3g-2-t4)
<p>

and 
- [Test5](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3g-2-t5)

as well. have you tried those?

natey416, 2025-02-17:
Sorry, should have clarified. The 3 I tried were tests 3-5

classicrocker883, 2025-02-18:
made a couple more tests

- [Test 6](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3g-2-t6) downgrade versions maybe it has something to do with that.
- [Test 7](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3f-10-t7) exactly as October 2024 release which worked, just want to confirm that it is a code thing or github actions thing.

natey416, 2025-02-20:
Test 6 doesn't work, but Test 7 gets me a normal M503 result! I'm able to use the EEPROM plugin as intended.

classicrocker883, 2025-02-20:
@natey416 so I spoke with @bdleung who has been following this issue and also helping test the solutions to this issue.

[Test10](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3g-2-t10)

please try this instead of any previous if you haven't done so already, because I believe I have solved the issue.
- **Test10** *SHOULD WORK*!!
please let me know!

natey416, 2025-02-21:
I think we can celebrate now! M503 and EEPROM editor work as intended!
Thank you so much for your help and patience!

natey416, 2025-02-21:
[I did notice the same thing](https://github.com/classicrocker883/MRiscoCProUI/issues/168#issuecomment-2672894044) that @bdleung noticed with some versions not ending up in the compiled list. Aquila_GD32_UBL-ProUI-EX-*.bin with Input shaping and/or MPC are missing.

classicrocker883, 2025-02-21:
I made a [**Test11**](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3g-2-t11) with latest updated actions versions, hopefully this works without issues and we can move on and finally close out the issue

natey416, 2025-02-21:
Test 11 working well! M503 and M115 both output expected results. I tried GD32_UBL_ProUI-EX and ***-EX-IS and they both worked!
I feel confident in closing this.

classicrocker883, 2025-02-21:
thank you so much, for your help and patience. 
# ⭐️⭐️⭐️⭐️⭐️ !!!

PR #198 (2025-02-21): Fix EEPROM Editor Octoprint Plugin - M503

### Description

Fix #196 and #168

#### Marlin EEPROM Editor Octoprint Plugin not connecting and command `M503` does not work on builds compiled from GitHub Actions

### Requirements

### Benefits

### Configurations

### Related Issues
closed 2025-02-21 by classicrocker883