classicrocker883/MRiscoCProUI #197: [BUG] X2 and X3 stops extruding on first layer on Jan build  both are N32 boards

motleycrue76, 2025-02-07:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

hi, the speed seems ok, but it stops extruding on the first layer and keeps going.  the extruder motor stops completely, but continues to try and print.  help

### Printer Model

Voxelab Aquila

### Model Type

X2 X3

### Your Mainboard

Aquila N32

### Add-ons that could be involved

CR touch for the X2,  Inductive for X3

### ProUI-EX | Extra Features?

No (formerly NoPro)

### Bed Leveling

UBL - Unified Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

motleycrue76, 2025-02-07:
im somewhat new to all this,  but with "Alex's" firmare on my X2 i have 4000 hours of prints all worked just fine.  i tried to upgrade to to the Jan 2025 firmware and it all went wrong.  i dont know how to edit or compile firmware myself, so your help would be great.  the X2 has the added CR touch, while the X3 is all stock out of the box (inductive sensor)

classicrocker883, 2025-02-09:
I suggest try using **PROUI_EX** version. otherwise, an older release should work like `2.1.3f-10` (October).

If it is the non-ProUI-EX that is the issue that will help narrow it down and I'll look into it.

motleycrue76, 2025-02-11:
Hi, thanks for responding.   I failed to mention the machine is an X3 Plus.  300x300x300.  I did flash with the Oct. PROUI_EX version and the same happened.  Prints the perimeter fine, when it goes to fill in, it stops extruding but keeps going as if it was.

Sent from AOL on Android

classicrocker883, 2025-02-12:
having it stop extruding after first layer *is* an odd issue. is there anything in your Gcode file that has `M111 S8`? it seems as if the printer is doing a **Dry-Run** mode.

search into your Gcode print file for `M111`, if you do see one or not add `M111 S0`

if you like attach/send the Gcode file you are using and I can take a quick look.

motleycrue76, 2025-02-12:
hi, yes i did use the ProUI_EX (Oct 2024, build) and changed the bed size to X=300 and Y=300. Also set the max X an Y travel to 300 did not make a difference when i set it to do bed mesh leveling. only scanned 230x230.  i flashed back to the stock firmware and printed the same file 4 times fine.  the g-code(pasted below) unless theres new code in Marlin, it does not load the mesh leveling data for the bed.  trying to remember think its M420 /S or something like that.  i will have to add the correct code into Voxelmaker slicer.  i really appreciate all your help!!!!  if you need the whole file let me know.

;generated by ffslicer 1.30.3 02/08/25 01:11:49
;right_extruder_material: PETG
;layer_height: 0.2
;perimeter_shells: 2
;top_solid_layers: 5
;bottom_solid_layers: 3
;fill_density: 15%
;fill_pattern: hexagon
;base_print_speed: 70
;travel_speed: 100
;right_extruder_temperature: 240
;left_extruder_temperature: 0
;platform_temperature: 75
;right_extruder_temperature_raft0: 0
;left_extruder_temperature_raft0: 0
;right_extruder_temperature_reset: 
;left_extruder_temperature_reset: 
;start gcode
M140 S75 T0
M104 S240 T0
M104 S0 T1
G90
G28
G1 Z50.000 F420
G161 X Y F3300
M190 S75 T0
M109 S240 T0
M109 S0 T1
G92 E.0000
;layer_count: 32
M108 T0
G1 X122.16 Y155.25 F6000
;preExtrude:0.20
;preExtruder0
G1 Z.200 F420
G1 X122.16 Y155.25 F6000
G1 X122.16 Y35.25 E15.1666 F1200
;raft
;layer:0.31
;shell
M106 S255
G1 E10.1666 F2700
G1 Z.310 F420
G1 X130.76 Y43.85 F3000
G1 E15.1666 F2700
;extrude_ratio:1.03158
G1 X130.76 Y43.85 F3000
G1 X140.79 Y43.85 E15.6734 F1200
G1 X140.79 Y53.85 E16.1786
G1 X140.79 Y63.85 E16.6838
G1 X140.79 Y73.85 E17.1890
G1 X140.79 Y83.85 E17.6942
G1 X140.79 Y192.65 E23.1911
G1 X130.76 Y192.65 E23.6978
G1 X130.76 Y43.85 E31.2155
G1 X130.36 Y43.45 F3000
G1 X141.19 Y43.45 E31.7626 F1200
G1 X141.19 Y193.05 E39.3208
G1 X130.36 Y193.05 E39.8679
G1 X130.36 Y43.45 E47.4260
;infill
;extrude_ratio:1
G1 E42.4260 F2700
G1 X140.08 Y44.19 F3000
G1 E47.4260 F2700
;extrude_ratio:1.03158
G1 X140.08 Y44.19 F3000
G1 X140.45 Y44.56 E47.4525 F1200
G1 X140.45 Y45.12 E47.4808
G1 X139.51 Y44.19 E47.5476
G1 X138.95 Y44.19 E47.5759
G1 X140.45 Y45.69 E47.6830
G1 X140.45 Y46.26 E47.7118
G1 X138.38 Y44.19 E47.8597
G1 X137.81 Y44.19 E47.8885
G1 X140.45 Y46.82 E48.0768
G1 X140.45 Y47.39 E48.1056
G1 X137.24 Y44.19 E48.3346
G1 X136.68 Y44.19 E48.3629
G1 X140.45 Y47.96 E48.6322
G1 X140.45 Y48.53 E48.6610
G1 X136.11 Y44.19 E48.9711
G1 X135.54 Y44.19 E48.9999
G1 X140.45 Y49.09 E49.3504
G1 X140.45 Y49.66 E49.3792
G1 X134.98 Y44.19 E49.7700
G1 X134.41 Y44.19 E49.7988
G1 X140.45 Y50.23 E50.2304
G1 X140.45 Y50.79 E50.2586

classicrocker883, 2025-02-13:
you did mention you tried the recent latest release? because someone else who uses 400x400x400 Aquila Max says they have no issue now.

ill take a look at your code and even try printing and see if I get the same issue.

motleycrue76, 2025-02-14:
hey, i flashed it to tonite...  Aquila_N32_IND_UBL-ProUI-EX-01-30.bin  if that is not the correct one, please link it for me.  i did take the same G-code file and put it in my X2 and it prints fine.  on the X3 plus it still prints the perimeter and then stops on the infill, and at the same time i noticed the "hot end" heater shuts off.  but the bed stays heating.
 i did find the bed size issue,  after resizing the bed, you have to go resize the mesh inset settings.  i did not know that, now i do.

motleycrue76, 2025-02-14:
ive tried a few versions, so i could not remember.  none worked though, i did just notice the heater shutting off when it goes to infill tonite.

classicrocker883, 2025-02-14:
that file should be correct one with Induction Probe.
so everything is working fine on the X2 now, but still issue with X3? could it be possible there is something with the wiring connection?

motleycrue76, 2025-02-14:
The X2 is back on the old Alex firmware.  I tried last night flashing the X3 and it did the same as before.  I just noticed that when it's printing the perimeter and switches to do the infill on the first layer, it stops extruding and shuts the hot end heater off, but keeps trying to print.  If I then shut it off and flash it back to stock, and set it to print the same file, it prints fine.  I was just testing to make sure it wasn't something with the slicer.  I'm gunna take the cover off the motherboard and get a couple photos of it.  The X3 plus is brand new.  Wondering if they changed something on it.

Sent from AOL on Android

motleycrue76, 2025-02-15:
I think I found the problem.  Running the 1.30.bin firmware as soon as it starts to print it shuts the hot end off.  If I go into the tune menu and turn the hot end right back up, it's printing fine.  Hope this helps. 

Sent from AOL on Android

motleycrue76, 2025-02-16:
hey,  been messing with things today.  and think i nailed it down.  i copied the start of the G file below.  i think its a slicer issue.  3 lines down after the "start gcode"  M104 S0 T1  seems to be the problem.  if i edit the code and after it add "M104 S240 T0"  it prints fine.  was this something changed in marlin 2.0?  also since ive got it mostly working, the updates even to the old "alex" firmware are amazing!!  THANKYOU!!! 

;generated by ffslicer 1.30.3 02/15/25 23:43:40
;right_extruder_material: PETG
;layer_height: 0.2
;perimeter_shells: 2
;top_solid_layers: 5
;bottom_solid_layers: 3
;fill_density: 15%
;fill_pattern: hexagon
;base_print_speed: 60
;travel_speed: 100
;right_extruder_temperature: 240
;left_extruder_temperature: 0
;platform_temperature: 75
;right_extruder_temperature_raft0: 0
;left_extruder_temperature_raft0: 0
;right_extruder_temperature_reset: 
;left_extruder_temperature_reset: 
;start gcode
M140 S75 T0
M104 S240 T0
M104 S0 T1
G90
G28
G1 Z50.000 F420
G161 X Y F3300
M190 S75 T0
M109 S240 T0
M109 S0 T1
G92 E.0000
;layer_count: 105
M108 T0
G1 X176.98 Y167.73 F6000
;preExtrude:0.20
;preExtruder0
G1 Z.200 F420
G1 X176.98 Y167.73 F6000
G1 X170.55 Y186.02 E2.4503 F1200
G1 X169.67 Y188.19 E2.7463
G1 X167.45 Y190.76 E3.1755
G1 X164.48 Y192.52 E3.6119
G1 X161.73 Y193.11 E3.9673
G1 X127.47 Y194.15 E8.2994
G1 X124.43 Y193.40 E8.6951
G1 X121.75 Y191.83 E9.0877
G1 X119.58 Y189.54 E9.4864
G1 X118.16 Y186.79 E9.8776
G1 X117.72 Y184.54 E10.1674
G1 X117.13 Y179.80 E10.7711
G1 X113.24 Y145.24 E15.1666
;raft
;layer:0.30
;shell
M106 S255
G1 E10.1666 F2700
G1 Z.300 F420
G1 X121.91 Y145.37 F3000
G1 E15.1666 F2700
;extrude_ratio:1.03158
G1 X121.91 Y145.37 F3000
G1 X122.00 Y144.73 E15.1982 F1200
G1 X122.30 Y144.09 E15.2328
G1 X122.71 Y143.61 E15.2637
G1 X123.30 Y143.22 E15.2982
G1 X123.92 Y143.03 E15.3299
G1 X124.63 Y143.02 E15.3647
G1 X125.25 Y143.19 E15.3961
G1 X125.85 Y143.56 E15.4306
G1 X126.28 Y144.03 E15.4617
G1 X126.59 Y144.67 E15.4965
G1 X126.70 Y145.30 E15.5277
G1 X126.58 Y146.36 E15.5799
G1 X127.60 Y147.53 E15.6558
G1 X128.15 Y148.15 E15.6963
G1 X129.68 Y149.82 E15.8071
G1 X131.58 Y151.74 E15.9391
G1 X131.58 Y155.95 E16.1450
G1 X131.58 Y165.95 E16.6339
G1 X131.58 Y175.95 E17.1228
G1 X131.58 Y178.66 E17.2553
G1 X131.47 Y179.45 E17.2943
G1 X131.18 Y180.16 E17.3318
G1 X130.56 Y180.96 E17.3813
G1 X130.49 Y181.56 E17.4108
G1 X130.91 Y182.43 E17.4581
G1 X131.02 Y183.06 E17.4893
G1 X130.94 Y183.76 E17.5238
G1 X130.68 Y184.36 E17.5557
G1 X130.24 Y184.90 E17.5898
G1 X129.70 Y185.28 E17.6221
G1 X129.04 Y185.50 E17.6561
G1 X128.39 Y185.52 E17.6879
G1 X127.71 Y185.35 E17.7222
G1 X127.15 Y185.03 E17.7537
G1 X126.66 Y184.51 E17.7886
G1 X126.37 Y183.94 E17.8199
G1 X126.23 Y183.25 E17.8543
G1 X126.29 Y182.60 E17.8862
G1 X126.56 Y181.92 E17.9220
G1 X126.97 Y181.26 E17.9600
G1 X126.15 Y180.28 E18.0225
G1 X125.82 Y179.58 E18.0603
G1 X125.68 Y178.83 E18.0976
G1 X125.75 Y178.06 E18.1354
G1 X126.01 Y177.34 E18.1728
G1 X126.44 Y176.71 E18.2101
G1 X127.04 Y176.21 E18.2483
G1 X127.72 Y175.89 E18.2851
G1 X128.48 Y175.75 E18.3228
G1 X129.27 Y175.81 E18.3616
G1 X130.27 Y176.08 E18.4122
G1 X130.27 Y158.54 E19.2698
G1 X130.02 Y157.64 E19.3155
G1 X129.34 Y156.92 E19.3639
G1 X128.40 Y156.58 E19.4128
G1 X127.41 Y156.71 E19.4616

Mik-S-UMS, 2025-02-16:
This reminds me of a post from someone that was having problems with Voxelab CS and started to complain on [Reddit](https://www.reddit.com/r/VoxelabAquila/comments/11hna0p/did_anyone_contact_voxelab_support_whats_your/). One of the things he was adamant on was the error message `M104 Invalid extruder 1` showing up in the terminal and assumed this was the source of all their problems, only for the CS to say that was not the problem.

I assume this was here for those printers that have 2 extruders and makes sure the second one is turned off. Stock FW just gives that message and ignores it. I thought it was just in the test Gcode files on the SDcard but this may be generated by Voxelmaker by default. I know Cura does not do this, don't know about other slicers.

I don't know if Mriscoc (or Alex's FW) supports more than one extruder so they may just ignore the T flag for M104 and this would explain the hotend turning off.

classicrocker883, 2025-02-18:
I think i found the issue, *possibly*.

so basically when you enter `M104` in the Gcode, this is the code I think which causes the issue

```y
void M104_M109(// other code ...) {
// other code ...
    const int8_t target_extruder = get_target_extruder_from_command();
    if (target_extruder < 0) return;

...

// get_target_extruder_from_command(); =
int8_t GcodeSuite::get_target_extruder_from_command() {
  #if HAS_TOOLCHANGE
    if (parser.seenval('T')) {
      const int8_t e = parser.value_byte();
      if (e < EXTRUDERS) return e;
      SERIAL_ECHO_START();
      SERIAL_CHAR('M'); SERIAL_ECHO(parser.codenum);
      SERIAL_ECHOLNPGM(" " STR_INVALID_EXTRUDER " ", e);
      return -1;
    }
  #endif
  return active_extruder;
}
```
since it does not have `HAS_TOOLCHANGE` - meaning multi extruder or whatnot all does it this
`return active_extruder;`

and.....

```y
// The active extruder (tool). Set with T<extruder> command.
#if HAS_MULTI_EXTRUDER
  extern uint8_t active_extruder;
#else
  constexpr uint8_t active_extruder = 0;
#endif
```

so since it **also** isn't `HAS_MULTI_EXTRUDER` (more than 1), `active_extruder` should always be `0`.

seeing the Gcode having T1 in it and causes the issues then that is it!

when it returns `active_extruder` it exits the functions (as `return`).

so to fix this it probably should be..

```y
```y
void M104_M109(// other code ...) {
// other code ...
    const int8_t target_extruder = get_target_extruder_from_command();
    if (target_extruder < 0) {
      return;
    } // else if !HAS_MULTI_EXTRUDER
    else if {target_extruder == 0) continue;
```
or something to that nature....

classicrocker883, 2025-02-21:
hey @motleycrue76 , just checking in... do you still need help with any issues? are you all set and everything good and working for you??

motleycrue76, 2025-02-27:
hey, sorry ive been down a few days.  i fell ill, but getting better. dam covid!  so an update, if i edit the G-code and take out the 2nd command of "M109 S0 T1" everything works fine.  ill post the code at the end. if i take out the first time it makes that command it heats both the bed and nozzle together, but if i leave it in and only take out the 2nd, it completely waits till the bed is up to temp before it heats the nozzle.   greatly reducing nozzle poop!!  i updated the X2 to "N32-BLT-ProUI-EX-IS-MPC-1-30" and its working fine as long as i take out that 2nd  M109 S0 T1 command.

motleycrue76, 2025-02-27:
i know thats a slicer issue, but is there anyway that the heating could be controlled like that in the firmware?  wait till the bed gets to temp then heat the nozzle?

motleycrue76, 2025-02-27:
crap i forgot to post he code........original is first, modified is 2nd. 

;generated by ffslicer 1.30.3 01/10/25 17:17:28
;right_extruder_material: PETG
;layer_height: 0.2
;perimeter_shells: 2
;top_solid_layers: 4
;bottom_solid_layers: 3
;fill_density: 15%
;fill_pattern: hexagon
;base_print_speed: 60
;travel_speed: 100
;right_extruder_temperature: 240
;left_extruder_temperature: 0
;platform_temperature: 75
;right_extruder_temperature_raft0: 0
;left_extruder_temperature_raft0: 0
;right_extruder_temperature_reset: 
;left_extruder_temperature_reset: 
;start gcode
M140 S75 T0
M104 S240 T0
M104 S0 T1
G90
G28
G1 Z50.000 F420
G161 X Y F3300
M190 S75 T0
M109 S240 T0
M109 S0 T1
G92 E.0000
;layer_count: 19
M108 T0
G1 X82.45 Y141.45 F6000

;generated by ffslicer 1.30.3 01/10/25 17:17:28
;right_extruder_material: PETG
;layer_height: 0.2
;perimeter_shells: 2
;top_solid_layers: 4
;bottom_solid_layers: 3
;fill_density: 15%
;fill_pattern: hexagon
;base_print_speed: 60
;travel_speed: 100
;right_extruder_temperature: 240
;left_extruder_temperature: 0
;platform_temperature: 75
;right_extruder_temperature_raft0: 0
;left_extruder_temperature_raft0: 0
;right_extruder_temperature_reset: 
;left_extruder_temperature_reset: 
;start gcode
M140 S75 T0
M104 S240 T0
M104 S0 T1
G90
G28
G1 Z50.000 F420
G161 X Y F3300
M190 S75 T0
M109 S240 T0
G92 E.0000
;layer_count: 19
M108 T0
G1 X82.45 Y141.45 F6000

classicrocker883, 2025-02-27:
everyone's been getting sick with something its that time of year, ugh i dont even want to remember last week that was just the worst.

what slicer are you using anyway? VoxelMaker? because I know that will add `T0` or `T1` for some odd reason. at least it does when for the Aries, I mostly use that slicer when on the Aries because of the wifi it has to it.

as far as I know heating is controlled by the Gcode command. it should not be a problem with the firmware, but like I said I think it has to do with it `T1`. it thinks there is another extruder, `T0` mean the first extruder, so technically using that shouldnt cause any issue with printers having only 1 extruder. but I think thats the issue... in the Gcode with the extra `T1`.

have you tried manually removing `T1`? - and optionally `T0` - we can test either or to see.

just open the Gcode in Notepad or similar and test it and see how that goes.

motleycrue76, 2025-02-28:
the slicer is voxelmaker cause it prints PETG so good.  the X2 is printing fine with the latest firmware( minus that one rouge "M104 S0 T1")  the X3 is  under extruding.  its running the Oct release of fimware.  i did have to turn the voltage down on the board to the E stepper.  it was at 1.6v  and getting red hot!  i need to check calibration, but am having trouble trying to find it in the menu to advance the extruder "100mm" and measure it.

classicrocker883, 2025-02-28:
youre using UBL firmware right? to adjust the Steps/mm, simply go from the Main Menu to Advanced Settings (should be bottom left square), then scroll to and select *Motion* > *Steps/mm* and you should find it for the E axis.

I would update to the newest firmware, which would be February this year, it has updates and changes since Oct.

motleycrue76, 2025-03-03:
so both the X2 and X3+ are running the latest update.  and the X2 is solid working great.  X3+ is running a test print at the moment of 5hrs.  i was chasing an extruding problem on the X3.  turned out to be a bad nozzle.  replaced and problem went away.  was trying to switch to harden steel nozzles.  ive tried a few from amazon and all have been bad,  any suggestions on good ones?

classicrocker883, 2025-03-03:
the only real reason to use a Hardened steel nozzle is if youre using abrasive filaments. even glow in the dark or filament with sparkles are abrasive. you *can* use brass nozzles because theyre cheap like you can get a bunch and replace it every whenever, or spend a bit more and get something good.

and you can also get a CHT nozzle and/or volcano nozzle if youre looking to print fast. I suggest a MicroSwiss, I assume youre looking for MK8 version?
- I purchased this [Plated A2 Tool Steel](https://www.amazon.com/dp/B07DJY52CC) Wear Resistant Nozzle
it is great I think.
- Looking just now, I noticed something new [CM2](https://www.amazon.com/dp/B0CC6RBT26) M2 Hardened Steel Tip
These two here should be what youre looking for

<br>

I want to add this anecdote while we're on the subject: 
- I also purchased this [Bondtech CHT](https://www.amazon.com/dp/B0CMPTKYW6) BiMetal Coated Nozzle - for Abrasive material (except for 0.4mm) and it worked well for a while until the tip of the nozzle fell out!!
I was lucky to respond back to them and they sent me another one, but still.... I think it will happen again.

motleycrue76, 2025-03-05:
yea, i print mostly in PETG,  but mom wanted me to try the wood filament for a thing she wanted and a friend at work gave me 2 spools of carbon filament and was gunna try using it.  thats why i asked.  so anyway, both the X2 and X3+ are working great!!  the "gridfinity" box has some ringing but over the stock firmware, its 100% better!!!.  ive not tried any of the "tuning" prints.

motleycrue76, 2025-03-08:
i have not seen that, but i will check it out.  so whats the stock limits of the aquila's as far as speed? i know that every machine is different, but if you know as a guide?  i did do a test prob about 6 months ago on the X2 at the max the voxelmaker slicer would do as far as print speed. and that was a base print speed of 180mm/s  from the stock speed of 60mm/s.  while it did make a 20h print in just under 5hrs,  it was very under extruded, but not terrible  and that was back then X2 was running the Alex firmware.  so i welcome your input.

motleycrue76, 2025-03-08:
just sent you some "coffee" funds for all your help.  THANKYOU!!!!

classicrocker883, 2025-03-09:
I appreciate the ko-fi thanks for that
as for the speed you're right it does depend on the printer. it also depends on the type of filament. as far as I know PLA is the only kind that you can print fast at.

I haven't tried PETG or ABS any faster than recommended but that gives me an idea of trying, why not?

anyway, I think there's a Cura plug-in with all kinds of test prints if you want to know how fast you can print.

but basically there are trade-offs if you want to print 200mm/s, your hardware (hotend, extruder) may be limited by how much volume can flow. that is why we all try upgrading the parts, like to a volcano hotend, or CHT nozzle or both. adding braces to the frame for stiffness, bigger motors with more torque that can handle the speed because the small stock motors would get way too hot - which is not good for them long term. probably even make your belt tensioner tighter so you don't get skipped steps (this also is not good long term).

so for that 20hr to 5hr print, you can just try increasing the flow rate and see if that helps - of course try it on a test print first.

CNC Kitchen has a few of videos about this subject. where he tests volume flow and where the extruder maxes out. so say you want to push more filament through since it's moving faster, the extruder can only push so much before it loses grip because the filament isn't melting fast enough.

motleycrue76, 2025-04-12:
hey andrew, hope your well.  things been good here.  did have a funny problem where (on the X3+) randomly the nozzle would crash on the bed at the start of a print.  but if you turn it off and turn it back on, it started fine,  but it has not happened in a while.  it seems that after i did 2 almost 24hr prints on it that came out perfect, its not done it yet.  the X2 is a champ!  running fine!!  4500hrs of print time.  i wish someone would modifiy the "Satsana" hotend fan cover for the X2  to fit the X3.  cause im getting "heat creap" on the X3.  i tried using freeCAD,  but its been since the early 90's since i used any of that software.  = lost!!  :)

classicrocker883, 2025-04-12:
the X3, that is with the inductive sensor? it may have to do with something in Marlin code.
I suppose what I can do is make a firmware.bin file using Marlin Source code you can test out to see if that issue is there. that would help narrow down what it can be if its specific to this firmware or something on Marlin's end.

is that something you want to try?

motleycrue76, 2025-04-16:
yes its an inductive sensor.  it has not happened on the X2, only the X3.  but its very random, and not really and issue, i think its with the power supply.  its very "unsmoothed"  or clean.  i tapped into the fan supply for the hot end, for a LED lite i made for the gantry top to shine down on the bed. the X2 has the same lite with little flutter. It flutters like crazy on the X3 when its heating the bed.  the lite is only taking off 200ma  on the fan supply not the hot end or bed heaters.  im gunna ask a friend thats way more technical than i am about adding a capacitor to smooth the supplies out.  and ask him about the issue if that could be causing the inductive sensor mishaps.

Mik-S-UMS, 2025-04-16:
The LED light is the problem here. The inductive sensor also takes its power supply from the hotend fan port on the motherboard so if there is any disruption to its supply it will not work properly. As you said it flutters on the X2 then this will have a greater affect on the X3.

How have you wired up this LED as they not need 24v to work and it may be drawing a lot more current then you think and it is causing it to turn itself off and on. Is it a LED module with its driver or are you stepping down the voltage to drive a bare LED?

On my X2 I have a 12v LED strip along the top bar of the frame, some for a web cam, and used a LED chip I salvaged from the backlight of a scrap TV to go inside the carriage to light the nozzle and all of these are stable. I used a buck regulator connected directly to a spare output of the PSU to drop the voltage down to 12v for the strips and a big resistor to limit the current for the hotend LED. 

[image]

classicrocker883, 2025-04-16:
for the inductive sensor, I would try adding a ferrite core around the wires see if that helps, a lot of others do the same for their BL touch.

and for any power I would tap into the power supply directly. but if it's a small light like on the hotend, then that is fine.

for me, I have one of those powered USB hubs, basically if you'd hook up a webcam and raspberry pi and light, it can over draw, but with the powered hub, its under its own power.

[here's](https://a.co/d/avRi02w) what I'm talking about

[image]

the blue is for data stuff, the red is just power, so like for the LED.

classicrocker883, 2025-04-16:
apparently someone else is having the stopping on first layer issue #205 using UBL, this doesn't happen with BLT. so when using `M420 S1`, it will stop/freeze. anyone else confirm this?

motleycrue76, 2025-04-18:
ive not had that problem.  but im running the Jan 30 build on both machines.  as to Mik's comment, thanks for concern,  its 8 white LED's in series rated at 3v 20ma.  i made a strip from a rail cover i printed, drilled, epoxyed the LED's in and soldered them.  there are resistors on both sides of the + and - supply.  i dont remember the values for them.  but they are equal.  they are not at full brightness.  i dint wanna load the supply.  and they are in the supply side of the board, not after.

motleycrue76, 2025-04-18:
i have the the same LED strip on my X2 and it does not hardly flutter at all.  connected the same way to the 24v supply to the board.  though the X2 is BLT and not an inductive sensor.  ive not had one bed crash with it.  i should take a look and see if i can find the ratings for the power supply output.  though i dont think im adding that much draw for the LED's.

Mik-S-UMS, 2025-04-18:
If the LEDs are fluttering then it will have an effect on the power supply. This will be magnified on the X3 as the inductive sensor is powered off the hotend fan port and will prevent a good signal from the inductive sensor so it may get false triggers or not trigger at all. 

I am not an electrical engineer so I can't say exactly what current it should draw or the correct setup, but drawing 200mA seems to be a lot for 8 LEDs, I don't know if this including the power used by the fan. This may be causing the power to be unstable as otherwise it should be constant and the LEDs should be stable.

I had a look at a LED calculator and using [8 x 3v LEDS @20ma](https://ledcalculator.net/#p=24&v=3&c=20&n=8&o=s) in series with you need a 1ohm resistor and it should only draw 20mA, total power draw is 480mW. This power will need to be dissipated so if encased in epoxy they could be overheating and this could be causing the fluttering.

If there are any more or if the LEDs use a higher voltage (white can be 3.3v) then they may need splitting into smaller groups of series connected in parallel, eg 4S-2P
[This calculator](https://www.amplifiedparts.com/tech-articles/led-parallel-series-calculator) gives more configurations but does not allow the result in a link, if you manually enter the values it will show you more info.

Also don't tap into the fan wires for the 24v as it is better to connect straight to the PSU terminals with suitable sized wire, this way the current is coming straight from the PSU and not being split to the LEDs, fan, and inductive sensor as well.

To confirm it is the LEDs causing the problems then you could just disconnect them and see if it stops the nozzle from occasionally crashing into the bed.

classicrocker883, 2025-04-18:
for 1ohm resistor you mean one of these?
https://a.co/d/gXvH5vj
[image]

yrabbit, 2025-04-18:
That's an overkill - you need to dissipate less than a milliwatt there, right? :)

Mik-S-UMS, 2025-04-18:
1/8W resistor is what the calculator suggests but to allow breathing room to dissipate heat.

motleycrue76, 2025-04-22:
just to refresh memory...... 8 - 3mm LED's at 20ma each in series on a 24 rail with 2 - 330ohm resistors, one on the + side, and one on the - side of the supply.

Mik-S-UMS, 2025-06-26:
> I think i found the issue, _possibly_.
> 
> so basically when you enter `M104` in the Gcode, this is the code I think which causes the issue
> 
> void M104_M109(// other code ...) {
> // other code ...
>     const int8_t target_extruder = get_target_extruder_from_command();
>     if (target_extruder < 0) return;
> 
> 
> ...
> 
> // get_target_extruder_from_command(); =
> int8_t GcodeSuite::get_target_extruder_from_command() {
>   #if HAS_TOOLCHANGE
>     if (parser.seenval('T')) {
>       const int8_t e = parser.value_byte();
>       if (e < EXTRUDERS) return e;
>       SERIAL_ECHO_START();
>       SERIAL_CHAR('M'); SERIAL_ECHO(parser.codenum);
>       SERIAL_ECHOLNPGM(" " STR_INVALID_EXTRUDER " ", e);
>       return -1;
>     }
>   #endif
>   return active_extruder;
> }
> 
> since it does not have `HAS_TOOLCHANGE` - meaning multi extruder or whatnot all does it this `return active_extruder;`
> 
> and.....
> 
> // The active extruder (tool). Set with T<extruder> command.
> #if HAS_MULTI_EXTRUDER
>   extern uint8_t active_extruder;
> #else
>   constexpr uint8_t active_extruder = 0;
> #endif
> 
> so since it **also** isn't `HAS_MULTI_EXTRUDER` (more than 1), `active_extruder` should always be `0`.
> 
> seeing the Gcode having T1 in it and causes the issues then that is it!
> 
> when it returns `active_extruder` it exits the functions (as `return`).
> 
> so to fix this it probably should be..
> 
> ```y
> void M104_M109(// other code ...) {
> // other code ...
>     const int8_t target_extruder = get_target_extruder_from_command();
>     if (target_extruder < 0) {
>       return;
>     } // #elseif !HAS_MULTI_EXTRUDER
>     else if {target_extruder == 0) continue;
> 
> or something to that nature....

@classicrocker883 Did this fix make it into the current release as someone on [Reddit](https://www.reddit.com/r/VoxelabAquila/comments/1lkaaka/voxelmaker_adding_t1_lines/) is having this issue with Voxelmaker.

motleycrue76, 2025-06-27:
Hi Mik, thanks for the comment!  I haven't updated since we kinda found the issue a few months back.  I just edit the g-code after slicing it to take out that one bad line that has to do with M109.. "T1"   kinda haven't been printing much lately,  it's outside weather now. :)  but thanks for the info.  I'll try the new update when I get a chance soon.

Sent from AOL on Android 

// The active extruder (tool). Set with T command.

classicrocker883, 2025-06-27:
@Mik-S-UMS I don't think it has been added to the code yet. but I can post a test .bin and see how that goes 🤷🏻‍♂️🤔

Mik-S-UMS, 2025-06-27:
If you are able to give it a try and let 21Comfort on Reddit test it. They do currently edit this out of the Gcode and does use other slicers so are aware of the problem but other people may only use Voxelmaker and not know about this.

EL-S, 2025-06-28:
I'm having this exact same issue with the Aquila X2 on HC32_MM-ProUI-EX-06-21.bin with Octoprint and VoxelMaker 1.2.8. When I remove all lines from the Gcode for T1, it prevents the target hot end temperature from being set to 0.

classicrocker883, 2025-06-30:
[T0-T1-Tests.zip](https://github.com/user-attachments/files/20972637/T0-T1-Tests.zip)

I uploaded a couple tests in the zip here:
- HC32_MM-ProUI-EX-06-30.bin
- Aquila_N32_IND_UBL-ProUI-EX-06-30.bin

let me know how it goes, if using T1 in the Gcode allows it to continue to print or not

EL-S, 2025-06-30:
It seems that HC32_MM-ProUI-EX-06-30.bin has not worked.

Send: M115
Recv: FIRMWARE_NAME:Marlin 2.1.3g-6Aquila-MM (Jun 30 2025 00:24:59) SOURCE_CODE_URL:github.com/classicrocker883/MRiscoCProUI PROTOCOL_VERSION:1.0 MACHINE_TYPE:Aquila KINEMATICS:Cartesian EXTRUDER_COUNT:1 UUID:342b0449-c119-4bed-8329-80d6343ca97a

[image]

EL-S, 2025-07-01:
I don't have an environment set up to build and compile firmware but I had a look over the source code briefly

Seems like M104_M109.cpp is fine.

I had a look at the implementation for get_target_extruder_from_command(), seems like if HAS_TOOLCHANGE is defined, the T parameter is checked and then selects the specified extruder, else it ignores the T parameter completely and returns the active extruder (T0) which looks like the behavior we are observing

```cpp
/**
 * Get the target extruder from the T parameter or the active_extruder
 * Return -1 if the T parameter is out of range
 */
int8_t GcodeSuite::get_target_extruder_from_command() {
  #if HAS_TOOLCHANGE
    if (parser.seenval('T')) {
      const int8_t e = parser.value_byte();
      if (e < EXTRUDERS) return e;
      SERIAL_ECHO_START();
      SERIAL_CHAR('M'); SERIAL_ECHO(parser.codenum);
      SERIAL_ECHOLNPGM(" " STR_INVALID_EXTRUDER " ", e);
      return -1;
    }
  #endif
  return active_extruder;
}
```

classicrocker883, 2025-07-02:
> ```y
> /**
>  * Get the target extruder from the T parameter or the active_extruder
>  * Return -1 if the T parameter is out of range
>  */
> int8_t GcodeSuite::get_target_extruder_from_command() {
>   #if HAS_TOOLCHANGE
>     if (parser.seenval('T')) {
>       const int8_t e = parser.value_byte();
>       if (e < EXTRUDERS) return e;
>       SERIAL_ECHO_START();
>       SERIAL_CHAR('M'); SERIAL_ECHO(parser.codenum);
>       SERIAL_ECHOLNPGM(" " STR_INVALID_EXTRUDER " ", e);
>       return -1;
>     }
>   #endif
>   return active_extruder;
> }
>```

This is the code which I used to tweak. I had it return 0 when extruders are = 1.
so it seems active_extruder is defined by the T command, I tried searching how that might effect it and target_extruder gets set in `M104_M109()` by `get_target_extruder_from_command()`

EL-S, 2025-07-02:
> This is the code which I used to tweak. I had it return 0 when extruders are = 1. so it seems active_extruder is defined by the T command, I tried searching how that might effect it and target_extruder gets set in `M104_M109()` by `get_target_extruder_from_command()`

I think if you return 0, it will apply any Tn to T0 always which is causing the issue.

Maybe the best solution is to leave the code as is, but compile with HAS_TOOLCHANGE defined (if that doesn't affect anything else)

Otherwise you could modify it to always check like this (just remove the HAS_TOOLCHANGE check)

```cpp
int8_t GcodeSuite::get_target_extruder_from_command() {
  if (parser.seenval('T')) {
    const int8_t e = parser.value_byte();
    if (e < EXTRUDERS) return e;
    SERIAL_ECHO_START();
    SERIAL_CHAR('M'); SERIAL_ECHO(parser.codenum);
    SERIAL_ECHOLNPGM(" " STR_INVALID_EXTRUDER " ", e);
    return -1;
  }
  return active_extruder;
}
```

```
check for T argument
  checks if requested extruder exists
    return that extruder
  else
    return -1 as doesn't exist
else
  return the active
```

I don't see any problems with this if we can't define HAS_TOOLCHANGE due to side effects

classicrocker883, 2025-07-07:
I added `#if HAS_TOOLCHANGE` to **gcode.cpp**/**gcode.h**

```y
    #if HAS_TOOLCHANGE
      case 'T': T(parser.codenum); break;                         // Tn: Tool Change
    #endif

  #if HAS_TOOLCHANGE
    static void T(const int8_t tool_index) IF_DISABLED(HAS_TOOLCHANGE, { UNUSED(tool_index); });
  #endif
```

this should stop reading the `T` from Gcode altogether.

here are some tests to try let me know how that goes

[Aquila_N32_HC32-Tests-07-07.zip](https://github.com/user-attachments/files/21097586/Aquila_N32_HC32-Tests-07-07.zip)

motleycrue76, 2025-08-30:
wow, alot has happened while i took a break from printing.  ive tried to read some of the work an comments over the months, but with the summer months, outdoor activities took over.  seems this little bug, has been hard to squash!!  can someone just quickly summarize what has happened over the past few months with it?

classicrocker883, 2025-08-30:
i believe my last comment by adding `#if HAS_TOOLCHANGE` may have solved the issue. if someone else would like to confirm that or add anything too

motleycrue76, 2025-09-02:
and the answer is....... unfortunately no.  the moment it starts to print, it turns off the hot end heater off.  :(   is there not a "ignore" command for T1?

EL-S, 2025-09-02:
> I added `#if HAS_TOOLCHANGE` to **gcode.cpp**/**gcode.h**
> 
>     #if HAS_TOOLCHANGE
>       case 'T': T(parser.codenum); break;                         // Tn: Tool Change
>     #endif
> 
> 
>   #if HAS_TOOLCHANGE
>     static void T(const int8_t tool_index) IF_DISABLED(HAS_TOOLCHANGE, { UNUSED(tool_index); });
>   #endif
> this should stop reading the `T` from Gcode altogether.
> 
> here are some tests to try let me know how that goes
> 
> [Aquila_N32_HC32-Tests-07-07.zip](https://github.com/user-attachments/files/21097586/Aquila_N32_HC32-Tests-07-07.zip)

I'm not actually sure if any code change is needed. Can you try only defining HAS_TOOLCHANGE for a test build instead of the previous code changes?

```
/**
 * Get the target extruder from the T parameter or the active_extruder
 * Return -1 if the T parameter is out of range
 */
int8_t GcodeSuite::get_target_extruder_from_command() {
  #if HAS_TOOLCHANGE
    if (parser.seenval('T')) {
      const int8_t e = parser.value_byte();
      if (e < EXTRUDERS) return e;
      SERIAL_ECHO_START();
      SERIAL_CHAR('M'); SERIAL_ECHO(parser.codenum);
      SERIAL_ECHOLNPGM(" " STR_INVALID_EXTRUDER " ", e);
      return -1;
    }
  #endif
  return active_extruder;
}
```

the following code snippet should already stop any Tn outside of the defined extruders amount (1), and that matches the error the stock firmware logs so I am confident they have the HAS_TOOLCHANGE preprocessor enabled. I believe all we need to do is define HAS_TOOLCHANGE in Configuration_adv.h or wherever make the most sense.

```
if (e < EXTRUDERS) return e;
      SERIAL_ECHO_START();
      SERIAL_CHAR('M'); SERIAL_ECHO(parser.codenum);
      SERIAL_ECHOLNPGM(" " STR_INVALID_EXTRUDER " ", e);
      return -1;
```

Apologies I didn't get back about trying the firmware, I have a 3D touch installed now and swapped to prusaslicer so the firmware wasn't compatible. I am using HC32_BLT-ProUI-EX-07-17.bin now but when I run old voxelmaker gcode it is funny to come back and realise it immediately failed due to cold extrusion.. haha

classicrocker883, 2025-09-03:
well `HAS_TOOLCHANGE` is only enabled if there are multi extruders. this doesn't process having it enabled, so that code doesn't run. by having it here: `case 'T': T(parser.codenum);`, is like disabling the code, meaning it *should* not see a `T` in gcode.

on top of the other changes, we can always try adding:

```y

--gcode.h (very end of file)--

  #if HAS_TOOLCHANGE
    static void T(const int8_t tool_index) IF_DISABLED(HAS_TOOLCHANGE, { UNUSED(tool_index); });
  #endif

--parser.h (line 124)--

    case 'G': case 'M': TERN_(HAS_TOOLCHANGE, case 'T':) TERN_(MARLIN_DEV_MODE, case 'D':) TERN_(HAS_CGCODE, case 'C':) {

```

unfortunately, I do not know if this will affect other `T` gcode things like `M86 T180`

So, technically, the Gcode doesn't even need `T0` when setting temp, it can do without.  
its possible that something can be added to M104_M109 or `get_target_extruder_from_command()`  
like `if (parser.seenval('T')) { target_extruder = 0 }`

such as
```y
  #if ENABLED(MIXING_EXTRUDER) && MIXING_VIRTUAL_TOOLS > 1
    constexpr int8_t target_extruder = 0;
  #else
    const int8_t target_extruder = get_target_extruder_from_command();
    if (target_extruder < 0) return;
  #endif
```
we can just add
`#if ENABLED(MIXING_EXTRUDER) && MIXING_VIRTUAL_TOOLS > 1 || HAS_TOOLCHANGE`

motleycrue76, 2025-09-03:
on a side note, i will be on vacation this friday for 2 weeks, so any changes would have to wait till after then.

motleycrue76, 2025-09-24:
I'm back home 😞  but ya back to reality.   I've downloaded both.  Gunna flash tomorrow and see.  
Sent from AOL on Android 

Aquila_N32_BLT-test.zip

classicrocker883, 2025-09-27:
I do see that it does give an error. ill have to recompile and reupload

motleycrue76, 2025-09-28:
got it loaded into the printer.........................SUCCESS!!!!!!   Great work!!  i know this has been a hard problem to nail down!!  next question is when will this update go into the firmware for the X3?

[4 more replies]