classicrocker883/MRiscoCProUI #199: [BUG] esteps not saving

dmotorhead, 2025-02-22:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

I uploaded the newest release Feb 21 2025 went to make all my adjustments and did my esteps for my microswiss ng direct drive extruder which needs esteps at 400.00 ok, store settings check and restart the printer esteps didn't save did this multiple times.

### Printer Model

Creality Ender-3

### Model Type

v2

### Your Mainboard

Creality v4.2.7/4.2.2 F103RET6

### Add-ons that could be involved

Cr touch

### ProUI-EX | Extra Features?

Yes

### Bed Leveling

UBL - Unified Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2025-02-22:
which firmware.bin file did you use to download?

dmotorhead, 2025-02-22:
I believe I used Ender_422_UBL-ProUI-EX-MPC-02-21.bin

classicrocker883, 2025-02-22:
I can tell you youre not wrong, on my test rig it is having the same issue, none of the settings are saving

dmotorhead, 2025-02-22:
I'm glad its not just me, is it just the esteps that are not saving? or mesh and other things? sucky to have this bug but I'm glad it was found to be an issue. I'm gonna wait out until you get it fixed before I reflash my machine. I did end up going back to Mriscoc just to keep printing but I appreciate your hard work on this firmware as I use yours for my voxelab aquila and plan to reflash the newest release to that as well, currently waiting for a btt skr mini v3.0 mainboard for that which has a microswiss extruder as well.

classicrocker883, 2025-02-22:
so what happened was I tried stopping the message "Mesh 0 Loaded" from appearing everytime at startup. and well while that succeeded.... this caused the issue. so I simply reverted that change and rebuilt the firmware files.

I literally JUST started the Github Actions workfllow, so if you dont see the firmware youre looking for just give it a while, it will show eventually.

closed 2025-02-22 by classicrocker883

dmotorhead, 2025-02-22:
Thank you so much for your hard work! Seriously you rock! You are a genius and I appreciate all your hard work and time!