classicrocker883/MRiscoCProUI #200: save settings DEFAULT_EJERK

yrabbit, 2025-02-23:
https://github.com/classicrocker883/MRiscoCProUI/blob/fef8476dbb747cd502c651a487ae1d63a21ffb95/Marlin/src/module/settings.cpp#L944

Hi, a little problem with DEFAULT_EJERK - the thing is that it is defined just in case of ENABLED(CLASSIC_JERK), and otherwise it just doesn't exist and the above line gives a compilation error.

https://github.com/classicrocker883/MRiscoCProUI/blob/fef8476dbb747cd502c651a487ae1d63a21ffb95/configurations/Voxelab%20Aquila%20427/UBL/Configuration.h#L1367-L1386

classicrocker883, 2025-02-23:
### Sorry I dont understand what the issue is. 
- Can you explain it a bit better and what compilation error? 
- What does **settings.cpp** have to do with this?
- Can you show me your configuration files and how you came to this conclusion?

Just for instance, the menu gets its max/min values from **limits.h**.

<br>

https://github.com/classicrocker883/MRiscoCProUI/blob/fef8476dbb747cd502c651a487ae1d63a21ffb95/Marlin/src/lcd/e3v2/common/limits.h#L72-L97

---

basically it comes down to these values

```y
  constexpr xyze_float_t min_jerk_edit_values = LOGICAL_AXIS_ARRAY_1(MIN_JERK_EDIT_VALUE); // { 0.1, 0.1, 0.1, 0.1 }
  constexpr xyze_float_t max_jerk_edit_values = { DEFAULT_XJERK * 2, DEFAULT_YJERK * 2, DEFAULT_ZJERK * 4, DEFAULT_EJERK * 4 };
```

yrabbit, 2025-02-23:
I wasn't actually guessing, I'm just getting my printer up to date with the current version for the first time since October and the compilation has failed somewhat.

[image]

Then I just put the value 10. instead of the name (well since it's ‘dummy’ there anyway) and I have to report that the February version works fine :)

I just wanted to let you know about this problem.

Here are the configuration files, but honestly I don't think there's anything criminal there - they're the same as they were in October when we last chatted.

classicrocker883, 2025-02-25:
ah i see. I have made changes in that file since then and notice a couple odd things. 

```y
// Original Marlin code

      #if ENABLED(CLASSIC_JERK)
        EEPROM_WRITE(planner.max_jerk);
        #if HAS_LINEAR_E_JERK
          dummyf = float(DEFAULT_EJERK);
          EEPROM_WRITE(dummyf);
        #endif
      #else
        const xyze_pos_t planner_max_jerk = LOGICAL_AXIS_ARRAY(5, 10, 10, 0.4, 0.4, 0.4, 0.4, 0.4, 0.4, 0.4);
        EEPROM_WRITE(planner_max_jerk);
      #endif

// My changed code line ~938
      #if ENABLED(CLASSIC_JERK)
        EEPROM_WRITE(planner.max_jerk);
      #else
        const xyze_pos_t planner_max_jerk = LOGICAL_AXIS_ARRAY(10, 10, 10, 0.4, 0.4, 0.4, 0.4, 0.4, 0.4, 0.4);
        EEPROM_WRITE(planner_max_jerk);
        #if HAS_LINEAR_E_JERK
          dummyf = float(DEFAULT_EJERK);
          EEPROM_WRITE(dummyf);
        #endif
      #endif

```

basically there is an impossibility for `HAS_LINEAR_E_JERK` while `CLASSIC_JERK` is disabled (i.e. Junction Deviation).

so I assume that is what you are using then, JD - `HAS_JUNCTION_DEVIATION`?

I can come up with a better fix, but for now just use the Marlin original code I guess for now. it isnt just this line but there are a couple lines like this I made changes to.

```y
// Marlin Original code

        #if ENABLED(CLASSIC_JERK)
          EEPROM_READ(planner.max_jerk);
          #if HAS_LINEAR_E_JERK
            EEPROM_READ(dummyf);
          #endif
        #else
          for (uint8_t q = LOGICAL_AXES; q--;) EEPROM_READ(dummyf);
        #endif

// My changed code line ~2100

        #if ENABLED(CLASSIC_JERK)
          EEPROM_READ(planner.max_jerk);
        #else
          for (uint8_t q = LOGICAL_AXES; q--;) EEPROM_READ(dummyf);
          #if HAS_LINEAR_E_JERK
            EEPROM_READ(dummyf);
          #endif
        #endif
```

yrabbit, 2025-02-25:
> so I assume that is what you are using then, JD - `HAS_JUNCTION_DEVIATION`?

Yeah, that's right.

Of course it will be a pity if https://github.com/MarlinFirmware/Marlin/pull/27710 is not compatible with JD, but we'll see that later :)

closed 2025-02-25 by yrabbit

yrabbit, 2025-02-25:
> Im pretty sure that itis compatible, I just never used it. there are menu options for JD so if you end up enabling - im curious how it differs from classic jerk - for better or worse?

Printers, especially our printers, are different mechanically, so for me JD gives the best results. If you're interested in the theory behind it, a little video:

https://www.youtube.com/watch?v=Mnvj6xCzikM

Well JD has indeed been the default in the original repository for a long time 😉 https://github.com/MarlinFirmware/Marlin/blob/efba06c4ce4253a42318da62dc696e8d64e21206/Marlin/Configuration.h#L1345-L1350

classicrocker883, 2025-02-25:
nice. thanks for the info. btw. I updated the code so it would stop that error (default branch has those changes), I couldn't figure out another workaround so I hope just omitting it makes it work out.

if you could let me know it still works or if theres an issue id appreciate it (hoping I did the right thing regarding **settings.cpp**) because that file is touchy as hell. a small change can make bigger problems.

yrabbit, 2025-02-25:
Yes, at commit https://github.com/classicrocker883/MRiscoCProUI/commit/4d7f7e61d8770c23b54a71222cc263c2e1988982 compiles and prints without any issues 👍 

[image]

yrabbit, 2025-02-26:
Okay, I jumped to conclusions. Every time I switch on (without SDcard inserted) it says ‘Default settings restored’.

This is not a problem for me since my settings are default since I set them directly in the source code, but I can imagine it is inconvenient for those who expect the settings to be saved between powerups.

I wonder what condition is the trigger for ‘Default settings restored’? Some kind of checksum?

https://github.com/user-attachments/assets/7629bbeb-f773-4a9b-8890-19a335485e09

classicrocker883, 2025-02-26:
yes I have noticed that too. same with having "Mesh # Loaded" at startup, if you look at https://github.com/classicrocker883/MRiscoCProUI/commit/fef8476dbb747cd502c651a487ae1d63a21ffb95 , it was successful in stopping that message from loading, however for some reason it would not save settings after changing them. so that is why it was reverted.

anyway, all that it is saying is EEPROM is initialized.  
if you look at **Marlin\src\module\settings.cpp** - line 3328  
`MSG_EEPROM_INITIALIZED = "Default Settings Restored"`

```y
  bool MarlinSettings::load() {
    // If the EEPROM data is valid load it
    if (validate()) {
      const EEPROM_Error err = _load();
      const bool success = (err == ERR_EEPROM_NOERR);
      TERN_(EXTENSIBLE_UI, ExtUI::onSettingsLoaded(success));
      return success;
    }

    // Otherwise reset settings to default "factory settings"
    reset();

    // Options to overwrite the EEPROM on error
    #if ANY(EEPROM_AUTO_INIT, EEPROM_INIT_NOW)
      (void)init_eeprom();
      LCD_MESSAGE(MSG_EEPROM_INITIALIZED);
      SERIAL_ECHO_MSG(STR_EEPROM_INITIALIZED);
    #endif

    return false;
  }
```

`EEPROM_AUTO_INIT` / `EEPROM_INIT_NOW` are kind of optional, I just never bothered disabling.

so I totally get what you're thinking, instead of saying "Default Settings Restored" at startup, it should say what is actually happening... "EEPROM Initialized"

yrabbit, 2025-03-01:
> so I totally get what you're thinking, instead of saying "Default Settings Restored" at startup, it should say what is actually happening... "EEPROM Initialized"

To be honest, I meant that the M501 should not swear at the EEPROM and I was wondering what kind of event serves as a flag for it - size mismatch, checksum mismatch or something else.

If your M501 passes normally, then probably my EEPROM is just dead.

[image]

classicrocker883, 2025-03-01:
oh i see the "`Warning: EEPROM Corrupt`" you mean?
have you compiled your own firmware or is this from one of the releases?

let me take a look at what my terminal has to say.
this could be because of a couple things...

### Option 1:

```y
if (!bedlevel.sanity_check()) {
     // some code
}
else {
   eeprom_error = ERR_EEPROM_CORRUPT;
}

...

bool unified_bed_leveling::sanity_check() {
  uint8_t error_flag = 0;

  if (settings.calc_num_meshes() < 1) {
    SERIAL_ECHOLNPGM("?Mesh too big for EEPROM.");
    error_flag++;
  }

  return !!error_flag;
}

return EEPROM warning if settings.calc_num_meshes() == true // basically just takes the sizeof bedlevel.z_values to see if it fits with the rest of EEPROM
```

<br>

I think in other words, your mesh values may be too much for the EEPROM. try zero-ing out (reset) the values.

---

### Option 2:

I may need to change (bump) the version, because it says:

```y
/**
 * IMPORTANT:  Whenever there are changes made to the variables stored in EEPROM
 * in the functions below, also increment the version number. This makes sure that
 * the default values are used whenever there is a change to the data, to prevent
 * wrong data being written to the variables.
 */

#define EEPROM_VERSION "V90"
#define EEPROM_OFFSET 100
```

yrabbit, 2025-03-01:
Maybe though there's enough room for two slots and I'm only using one. I can't use the releases at least because I have to specify a 1047 thermistor.

It looks like I have to do a binary search for the version after which the EEPROM starts to generate an error. Uh-oh, I should have compiled periodically - I wouldn't have to start from October now. But I just didn't have anything to print.

``` text
M503
echo:; Linear Units:
echo:  G21 ; (mm)
echo:; Temperature Units:
echo:  M149 C ; Units in Celsius
echo:; Volumetric Extrusion (Disabled):
echo:  M200 S0 D1.75
echo:; Steps per unit:
echo:  M92 X80.00 Y80.00 Z400.00 E565.42
echo:; Max feedrates (units/s):
echo:  M203 X750.00 Y750.00 Z10.00 E80.00
echo:; Max Acceleration (units/s2):
echo:  M201 X750.00 Y750.00 Z120.00 E250.00
echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
echo:  M204 P750.00 R250.00 T750.00
echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>):
echo:  M205 B50000.00 S0.00 T0.00 J0.04
echo:; Home Offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Unified Bed Leveling:
echo:  M420 S0 Z10.00 ; Leveling OFF

Unified Bed Leveling System v1.01 inactive
echo:Active Mesh Slot 0
echo:EEPROM can hold 2 meshes.

echo:; Material heatup parameters:
echo:  M145 S0 H210.00 B60.00 F0
echo:  M145 S1 H240.00 B75.00 F0
echo:  M145 S2 H200.00 B50.00 F0
echo:; Bed PID:
echo:  M304 P128.06 I24.95 D438.07
echo:; Display Sleep:
echo:  M255 S10 ; (minutes)
echo:; LCD Brightness:
echo:  M256 B100
echo:; Power-Loss Recovery:
echo:  M413 S0 ; OFF
echo:; Retract (S<length> F<feedrate> Z<lift>):
echo:  M207 S0.20 F2400.00 W13.00 Z0.20
echo:; Recover (S<length> F<feedrate>):
echo:  M208 S0.00 F2400.00 W0.00 R480.00
echo:; Homing Feedrate:
echo:  M210 X3600.00 Y3600.00 Z480.00
echo:; Probe Offset:
echo:  M851 X-28.00 Y-29.00 Z-1.91 ; (mm)
echo:; Fixed-Time Motion:
echo:  M493 S0 A25.00 B30.00 D0 F0.00 H0.00 P0 K0.00
echo:; Nonlinear Extrusion:
  M592 A0.00 B0.00 C1.00
echo:; Input Shaping:
echo:  M593 X F25.00 D0.30
echo:  M593 Y F30.00 D0.25
echo:; Hotend Idle Timeout:
  M86 B0 E0 S360 T180
echo:; Linear Advance:
echo:  M900 K0.00
echo:; Filament load/unload:
echo:  M603 L10.00 U120.00 ; (mm)
echo:; Filament runout sensor:
echo:  M412 S0 D25.00 ; Sensor OFF
echo:; Model Predictive Control:
echo:  M306 E0 P40.00 C16.70 R0.2200 A0.0680 F0.0970 H0.0056
echo:; Mesh Insets and leveling settings:
echo:  C29 L10.00 R202.00 F10.00 B191.00 N9 T256 ; 9x9  T=256°C
echo:; Physical minimums:
echo:  C100 X0 Y0
echo:; Physical maximums:
echo:  C101 X230 Y220 Z200
echo:; Bed size:
echo:  C102 X220 Y220
echo:; Max Extruder temperature:
echo:  C104 T315
echo:; Park Head:
echo:  C125 X0 Y200 Z50
echo:; Filament runout sensor:
echo:  C412 M1 ; Active mode: HIGH
echo:; Invert Extruder:
echo:  C562 E1 ; inverted
echo:; Toolbar configuration:
echo:  C810 A1 B7 C6 D2 E4
echo:; Probe Z Speed and Multiple Probing:
echo:  C851 S150 M2
ok P63 B31
```

classicrocker883, 2025-03-01:
I may have made a mistake, youre using JD right?
remember how I got rid of `#if HAS_LINEAR_E_JERK` from **settings.cpp**? well there was one more which should have been removed.

#### So in **settings.cpp** line ~2100 make the following changes
```diff
        #if ENABLED(CLASSIC_JERK)
          EEPROM_READ(planner.max_jerk);
        #else
          for (uint8_t q = LOGICAL_AXES; q--;) EEPROM_READ(dummyf);
-         #if HAS_LINEAR_E_JERK
-           EEPROM_READ(dummyf);
-         #endif
        #endif

        #if ENABLED(CLASSIC_JERK)
          EEPROM_READ(planner.max_jerk);
        #else
          for (uint8_t q = LOGICAL_AXES; q--;) EEPROM_READ(dummyf);
        #endif
```

<br>

let me know if that has done anything.

I just dont get why they have `HAS_LINEAR_E_JERK` when `CLASSIC_JERK` is enabled. this is the conditions for it:

```y
#if ALL(HAS_JUNCTION_DEVIATION, LIN_ADVANCE)
  #define HAS_LINEAR_E_JERK 1
#endif
```

classicrocker883, 2025-03-01:
[settingscpp.zip](https://github.com/user-attachments/files/19038329/settingscpp.zip)

I edited this file with a couple changes ALLOWING `HAS_LINEAR_E_JERK`

if youre using JD, can you copy paste this file and let me know if anything changes?

yrabbit, 2025-03-01:
I copied the file you provided with no changes, copied, installed the new firmware and the printer booted up, said the settings were reset, which was expected, but the second time there was no more swearing.

Also the output after the sequence `M500` `M501` has changed: now when reading it prints the checksum which is the same as when writing the settings.

[image]

So thank you very much, you found what was wrong and fixed it very quickly. 👍 🎆

classicrocker883, 2025-03-01:
just to clarify real quick, did you make these changes from here at all?

https://github.com/classicrocker883/MRiscoCProUI/issues/200#issuecomment-2692102263

classicrocker883, 2025-03-14:
hey @yrabbit, [Smooth Linear Advance with extruder synchronisation #27710](https://github.com/MarlinFirmware/Marlin/pull/27710#issuecomment-2717380402) is looking great and close to done!

yrabbit, 2025-03-14:
Yeah, I'm watching this thing with anticipation (even noted to myself that I need to ask you to do a merge as soon as this PR is accepted) 😄

classicrocker883, 2025-03-14:
you know it lol

dbuezas, 2025-03-16:
> Of course it will be a pity if [MarlinFirmware/Marlin#27710](https://github.com/MarlinFirmware/Marlin/pull/27710) is not compatible with JD, but we'll see that later :)

It's compatible with JD, may or may not work with multiple extruders, should be compatible with s-curve but untested and not optimised for it (classic linear advance wasn't either so it should still be an improvement)

yrabbit, 2025-03-16:
That's great news! I don't have multiple extruders, and S-Curve is disabled.

So I can't wait for this thing to get into the main repository.

yrabbit, 2025-04-07:
I think it's time for me to update the firmware, although https://github.com/MarlinFirmware/Marlin/pull/27710 for some reason delayed past experience shows that periodically I need to synchronize FW:)

dbuezas, 2025-04-07:
Have you tried it? Feedback would be appreciated. Some movement in the pr may also encourage the maintainers to prioritize it (maybe)

yrabbit, 2025-04-07:
> Have you tried it? Feedback would be appreciated. Some movement in the pr may also encourage the maintainers to prioritize it (maybe)

No I haven't tried it because this PR is for a different repository. Do you think it will be applied without conflicts on say version 2025-April of this project?

dbuezas, 2025-04-08:
You'll have to try it, I don't know the details of this repository. I'd advise against directly merging it to your repository to avoid future conflicts though

yrabbit, 2025-04-08:
Nope. Some of the fixes pass, some don't, and some are already applied and resent being reapplied. 

I will have to wait until it is accepted in upstream:(

[image]

classicrocker883, 2025-04-09:
I can upload a test repository/PR merging with the current if you wanna give it a try

yrabbit, 2025-04-09:
> I can upload a test repository/PR merging with the current if you wanna give it a try

That would be wonderful :)

classicrocker883, 2025-04-09:
done. you can find the **PR** here https://github.com/classicrocker883/MRiscoCProUI/pull/204

or the branch is 2025-April-LA-Test

yrabbit, 2025-04-10:
At least there was no catastrophe, nothing burned or squealed. The surface quality is at least not worse than on 2025-April (this is with PCBs installed).

[image]
[image]

Further it is necessary to experiment with settings, which is somewhat hampered by the broken (again :))  system of writing settings to EEPROM :)

[image]

yrabbit, 2025-04-10:
> I think before that issue had to do with JD instead of classic jerk, but then we fixed it?

yes we fixed. It was a different error - EEPROM Corrupt . https://github.com/classicrocker883/MRiscoCProUI/issues/200#issuecomment-2691897874

dbuezas, 2025-04-10:
Smooth LA adds an extra parameter to configure (tau) which is similar to the smoothing time in klipper, that's why the eeprom errors. 
Have you noticed the extruder making less noises and measured the printing time?

yrabbit, 2025-04-10:
My extruder doesn't make a sound. But I'm curious about your max ejerk and e steps - you've tested this thing in practice, haven't you?

dbuezas, 2025-04-10:
Me? I'm using it, yes. For a long while. And a previous version for even longer.
I have the print times of a bambu labs or voron printer at quality in my ultimaker 2 (with direct drive, remote cooling and an skr3 board).
My ejerk is 5, acc=3k mm/s², v=120mm/s in outer walls and top surfaces and acc=15k mm/s² v=200mm/s in the rest. 
K≈0.05 and tau=0.01 (for pla)

classicrocker883, 2025-04-10:
ok I made a couple changes to the settings in 2025-April-LA-Test. also made a couple tweaks to prevent division by zero - thats what was suggested to improve.

yrabbit, 2025-04-10:
You're too fast! Version 2025-April without save/restore problems: 

https://github.com/classicrocker883/MRiscoCProUI/commit/5fd8d8248a695edfc11c0cdaf78df205ab810601

I'll try the latest LA version now.

dbuezas, 2025-04-10:
fyi: smooth linear advance will make the machine work differently in seams and sharp curves, everywhere else it will behave similarly to the classic LA. Should also improve the borders of solid infill in top layers (at equal resulting acceleration)

yrabbit, 2025-04-10:
Nope.
https://github.com/classicrocker883/MRiscoCProUI/commit/4403ca36296c1853faeef7d5440f5cd3834ca96d

907 bytes are written, 903 are read back.

[image]

classicrocker883, 2025-04-10:
same as before. only 4 bytes, which shouldn't cause issues. so that appears to be with only `SMOOTH_LIN_ADV` enabled? This can probably be tested using the Marlin Simulator instead of reflashing. but its probably to do with how something is defined, like an integer or float, or how its saved in settings/EEPROM

dbuezas, 2025-04-10:
SMOOTH_LIN_ADV adds another param called tau

yrabbit, 2025-04-11:
> SMOOTH_LIN_ADV adds another param called tau

It's good that it writes and reads tau, but the size of the data in the EEPROM is calculated from the size of the SettingsData structure . Does it mention this new parameter?

https://github.com/classicrocker883/MRiscoCProUI/blob/bb23a0b54edbb4c998cb984814599852c38a1e11/Marlin/src/module/settings.cpp#L746

classicrocker883, 2025-04-11:
> > SMOOTH_LIN_ADV adds another param called tau
> 
> It's good that it writes and reads tau, but the size of the data in the EEPROM is calculated from the size of the SettingsData structure . Does it mention this new parameter?
> 

it may have to do with `Stepper::set_advance_tau()` and `Stepper::det_advance_tau()` in **settings.cpp**.

```
      static void set_advance_tau(float tau) {
        // Validate tau to avoid division by zero or negative values
        if (tau <= 0) {
          SERIAL_ECHOLN("Error: tau must be greater than 0.");
          return;
        }

        extruder_advance_TAU = tau;
        extruder_advance_TAU_TICKS = tau * STEPPER_TIMER_RATE;

        // Calculate α = 1 − exp(−dt / τ)
        const float dt = SMOOTH_LIN_ADV_INTERVAL * SMOOTH_LIN_ADV_EXP_ORDER;
        const float exponent = -dt / extruder_advance_TAU_TICKS;
        extruder_advance_ALPHA = 1 - expf(exponent);
      }

      /**
       * Get the current advance tau value.
       * @return The advance time constant.
       */
      static float get_advance_tau() {
        return extruder_advance_TAU;
      }
```

as you can see `get_advance_tau()` is simple and returns 1 value. `set_advance_tau()` changes a few more.

so I updated 2025-April-LA-Test by adding `stepper_extruder_advance_TAU_TICKS` + `stepper_extruder_advance_ALPHA` to the `SettingsData` struct. hopefully this fixes that EEPROM warning.

classicrocker883, 2025-04-11:
wait a second, @dbuezas shouldn't `EEPROM_WRITE(Stepper::get_advance_tau());` be `EEPROM_WRITE(Stepper::set_advance_tau(tau));`?

yrabbit, 2025-04-11:
:) this trick won't work - how many bytes the fields in the structure occupy is how many write/read commands there should be. We had a difference of 4 bytes, and you added three new float fields. Now it breaks already at the writing stage:)

[image]

yrabbit, 2025-04-11:
Let me explain - write operations in EEPROM remained as it was at 907 bytes, but the size of the structure grew disproportionately - that's why the error when saving:)

classicrocker883, 2025-04-11:
> Let me explain - write operations in EEPROM remained as it was at 907 bytes, but the size of the structure grew disproportionately - that's why the error when saving:)

okay I think I figured it out. I removed the float in SettingsData - `stepper_extruder_advance_TAU`. so this float takes up 4 bytes. its gotta be it. I uploaded the change now.

yrabbit, 2025-04-11:
EEROM is good now :)

``` gcode
Terminal ready
M500
echo: Settings Stored (903 bytes; crc 9198)
Mesh saved in slot 0
//action: notification Settings Stored
ok P63 B31
M501
echo: V90 stored settings retrieved (903 bytes; crc 9198)
//action: notification Stored settings retrieved
Unified Bed Leveling System v1.01 inactive
Mesh loaded from slot 0
Mesh 0 loaded from storage.
echo: ; Linear Units:
echo:   G21 ; (mm)
echo: ; Temperature Units:
echo:   M149 C ; Units in Celsius
echo: ; Volumetric Extrusion (Disabled):
echo:   M200 S0 D1.75
echo: ; Steps per unit:
echo:   M92 X80.00 Y80.00 Z400.00 E565.42
echo: ; Max feedrates (units/s):
echo:   M203 X750.00 Y750.00 Z10.00 E80.00
echo: ; Max Acceleration (units/s2):
echo:   M201 X750.00 Y750.00 Z120.00 E250.00
echo: ; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
echo:   M204 P750.00 R250.00 T750.00
echo: ; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>):
echo:   M205 B50000.00 S0.00 T0.00 J-1000.00
echo: ; Home Offset:
echo:   M206 X0.00 Y0.00 Z0.00
echo: ; Unified Bed Leveling:
echo:   M420 S0 Z10.00 ; Leveling OFF

Unified Bed Leveling System v1.01 inactive
echo: Active Mesh Slot 0
echo: EEPROM can hold 2 meshes.

echo: ; Material heatup parameters:
echo:   M145 S0 H210.00 B60.00 F0
echo:   M145 S1 H240.00 B75.00 F0
echo:   M145 S2 H200.00 B50.00 F0
echo: ; Bed PID:
echo:   M304 P128.06 I24.95 D438.07
echo: ; Display Sleep:
echo:   M255 S10 ; (minutes)
echo: ; LCD Brightness:
echo:   M256 B100
echo: ; Power-Loss Recovery:
echo:   M413 S0 ; OFF
echo: ; Retract (S<length> F<feedrate> Z<lift>):
echo:   M207 S0.20 F2400.00 W13.00 Z0.20
echo: ; Recover (S<length> F<feedrate>):
echo:   M208 S0.00 F2400.00 W0.00 R480.00
echo: ; Homing Feedrate:
echo:   M210 X3600.00 Y3600.00 Z480.00
echo: ; Probe Offset:
echo:   M851 X-28.00 Y-29.00 Z-2.19 ; (mm)
echo: ; Skew Factor:
echo:   M852 I-0.002558 J0.002557 K0.002557 ; XY, XZ, YZ
echo: ; Fixed-Time Motion:
echo:   M493 S0 A25.00 B30.00 D0 F0.00 H0.00 P0 K0.00
echo: ; Nonlinear Extrusion:
  M592 A0.00 B0.00 C1.00
echo: ; Input Shaping:
echo:   M593 X F25.00 D0.30
echo:   M593 Y F30.00 D0.25
echo: ; Hotend Idle Timeout:
  M86 B0 E0 S360 T180
echo: ; Linear Advance:
echo:   M900 K0.00
echo: ; Filament load/unload:
echo:   M603 L10.00 U120.00 ; (mm)
echo: ; Filament runout sensor:
echo:   M412 S0 D25.00 ; Sensor OFF
echo: ; Model Predictive Control:
echo:   M306 E0 P40.00 C16.70 R0.2200 A0.0680 F0.0970 H0.0056
echo: ; Mesh Insets and leveling settings:
echo:   C29 L10.00 R202.00 F10.00 B191.00 N9 T256 ; 9x9  T=256°C
echo: ; Physical minimums:
echo:   C100 X0 Y0
echo: ; Physical maximums:
echo:   C101 X230 Y220 Z200
echo: ; Bed size:
echo:   C102 X220 Y220
echo: ; Max Extruder temperature:
echo:   C104 T315
echo: ; Park Head:
echo:   C125 X0 Y200 Z50
echo: ; Filament runout sensor:
echo:   C412 M1 ; Active mode: HIGH
echo: ; Invert Extruder:
echo:   C562 E1 ; inverted
echo: ; Toolbar configuration:
echo:   C810 A1 B7 C6 D2 E4
echo: ; Probe Z Speed and Multiple Probing:
echo:   C851 S150 M2
```

dbuezas, 2025-04-11:
> wait a second, @dbuezas shouldn't EEPROM_WRITE(Stepper::get_advance_tau()); be EEPROM_WRITE(Stepper::set_advance_tau(tau));?
edit: maybe not, but for some reason it seems backwards.

get_advance_tau just returns the tau value, that's the only thing that needs storing.
All other (alpha, tau_ticks) are computed from tau, they just exist to avoid recomputing them all the time from tau.

dbuezas, 2025-04-11:
I think the issue is that i forgot to guard the   `float stepper_extruder_advance_TAU; // Stepper::get_advance_tau()` with an `#ifdef SMOOTH_LIN_ADV` inside SettingsDataStruct.
So if it is not enabled, the settings are expected to have 4 bytes more than it has (a float)

yrabbit, 2025-04-11:
You mean do something like this? Well, yeah, that's a good option.

settings.cpp 

``` cpp
//
// Linear Advance
//
  #if ENABLED(LIN_ADVANCE)
    float planner_extruder_advance_K[DISTINCT_E];       // M900 K  planner.extruder_advance_K
     #if ENABLED(SMOOTH_LIN_ADV)
           float extruder_advance_TAU;
      #endif
   #endif
```

yrabbit, 2025-04-11:
TAU value is not displayed correctly in the interface 0.00 vs 0.01

[image]

``` text
M503
echo: ; Linear Units:
echo:   G21 ; (mm)
echo: ; Temperature Units:
echo:   M149 C ; Units in Celsius
echo: ; Volumetric Extrusion (Disabled):
echo:   M200 S0 D1.75
echo: ; Steps per unit:
echo:   M92 X80.00 Y80.00 Z400.00 E565.42
echo: ; Max feedrates (units/s):
echo:   M203 X750.00 Y750.00 Z10.00 E80.00
echo: ; Max Acceleration (units/s2):
echo:   M201 X750.00 Y750.00 Z120.00 E250.00
echo: ; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
echo:   M204 P750.00 R250.00 T750.00
echo: ; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>):
echo:   M205 B50000.00 S0.00 T0.00 J0.04
echo: ; Home Offset:
echo:   M206 X0.00 Y-2.00 Z0.00
echo: ; Unified Bed Leveling:
echo:   M420 S0 Z10.00 ; Leveling OFF

Unified Bed Leveling System v1.01 inactive
echo: Active Mesh Slot -1
echo: EEPROM can hold 2 meshes.

echo: ; Material heatup parameters:
echo:   M145 S0 H210.00 B60.00 F0
echo:   M145 S1 H240.00 B75.00 F0
echo:   M145 S2 H200.00 B50.00 F0
echo: ; Bed PID:
echo:   M304 P128.06 I24.95 D438.07
echo: ; Display Sleep:
echo:   M255 S0 ; (minutes)
echo: ; LCD Brightness:
echo:   M256 B100
echo: ; Power-Loss Recovery:
echo:   M413 S0 ; OFF
echo: ; Retract (S<length> F<feedrate> Z<lift>):
echo:   M207 S0.20 F1500.00 W13.00 Z0.20
echo: ; Recover (S<length> F<feedrate>):
echo:   M208 S0.00 F2400.00 W0.00 R480.00
echo: ; Homing Feedrate:
echo:   M210 X3600.00 Y3600.00 Z480.00
echo: ; Probe Offset:
echo:   M851 X-28.00 Y-29.00 Z-2.12 ; (mm)
echo: ; Skew Factor:
echo:   M852 I-0.002558 J0.002557 K0.002557 ; XY, XZ, YZ
echo: ; Fixed-Time Motion:
echo:   M493 S0 A25.00 B30.00 D0 F0.00 H0.00 P0 K0.00
echo: ; Nonlinear Extrusion:
  M592 A0.00 B0.00 C1.00
echo: ; Input Shaping:
echo:   M593 X F25.00 D0.30
echo:   M593 Y F30.00 D0.25
echo: ; Hotend Idle Timeout:
  M86 B0 E0 S360 T180
echo: ; Linear Advance:
echo:   M900 K0.06
  M900 U0.01
echo: ; Filament load/unload:
echo:   M603 L10.00 U120.00 ; (mm)
echo: ; Filament runout sensor:
echo:   M412 S0 D25.00 ; Sensor OFF
echo: ; Model Predictive Control:
echo:   M306 E0 P40.00 C15.53 R0.1111 A0.0850 F0.1233 H0.0056
echo: ; Mesh Insets and leveling settings:
echo:   C29 L10.00 R190.00 F10.00 B191.00 N9 T316 ; 9x9  T=316°C
echo: ; Physical minimums:
echo:   C100 X0 Y0
echo: ; Physical maximums:
echo:   C101 X230 Y220 Z200
echo: ; Bed size:
echo:   C102 X220 Y220
echo: ; Max Extruder temperature:
echo:   C104 T315
echo: ; Park Head:
echo:   C125 X0 Y200 Z50
echo: ; Filament runout sensor:
echo:   C412 M1 ; Active mode: HIGH
echo: ; Invert Extruder:
echo:   C562 E1 ; inverted
echo: ; Toolbar configuration:
echo:   C810 A1 B7 C6 D2 E4
echo: ; Probe Z Speed and Multiple Probing:
echo:   C851 S150 M2
ok P63 B31

```

yrabbit, 2025-04-11:
When homing (G28), the stepper motors make a squeaking/whirring noise. Quite nasty by the way. The only difference between these two videos is whether #define SMOOTH_LIN_ADV is commented out or not. 
I thought only the extruder mode changes, but not the XY stepper motors.

https://yrabbit.cyou/extra/marlin/smooth.mp4

https://yrabbit.cyou/extra/marlin/disable-smooth.mp4

dbuezas, 2025-04-11:
Indeed it only changes extruder behaviour. I'd suggest undoing all the code tweaks made to the PR, maybe one of them engaged the extruder where it shouldn't have been.
I'm sick so i can't help much looking at code at the moment

dbuezas, 2025-04-11:
It will be a nightmare for you to merge further marlin changes in the future if you may many stylistic changes. It's also harder for me to use them to back port any functional improvements into marlin too

yrabbit, 2025-04-11:
> Indeed it only changes extruder behaviour. I'd suggest undoing all the code tweaks made to the PR, maybe one of them engaged the extruder where it shouldn't have been. I'm sick so i can't help much looking at code at the moment

Get well soon. I'm in no hurry to get anywhere and I'm not writing code or porting anything - I'm just a guinea pig for testing all sorts of interesting things.

classicrocker883, 2025-04-11:
well that's good te eeprom thing worked,  however that was because i removed this entirely. 
```y
     #if ENABLED(SMOOTH_LIN_ADV)
           float extruder_advance_TAU;
      #endifNE
```
even with the `#if` guard before that didn't fix it.

and for any other tweaks they are mainly cosmetic. I think the biggest change was refactoring the two functions I mentioned in the **PR** which is only used using Input Shaping.
but I'll look back and compare to see if anything is amiss.

as for the other issues, if you're using an Aquila board, try disabling `OLD_ADAPTIVE_MULTISTEPPING` that might fix the motor noise.
and for the LCD menu option I figured that might not work right away, but Ill get it fixed, for now `M900 U` can be used to change Tau

classicrocker883, 2025-04-11:
okay thats good, that doesnt use `OLD_ADAPTIVE_MULTISTEPPING` just make sure its not enabled at all (see **configurations/Voxelab Aquila 427** - use this as template/guide)

I uploaded another revision to 2025-April-LA-Test, being able to change TAU through the LCD *should* work now. as for Homing its possible that the other things you have enabled could impact that, like Input Shaping, FT Motion, ect.

I would try disabling all those things which aren't enabled in the default configuration, that would at least narrow what it could be.

and was the homing noise always there from the beginning? or when did you notice it?

yrabbit, 2025-04-11:
So both the very first (4485e36bce03399691fd8fefc72f5a4bfc9b22bd) and the very last (b674c42af11cb9b2606dffcef7c24779d4923439) work quietly with minimally modified config:

``` diff
--- a/configurations/Voxelab Aquila 427/UBL/Configuration_adv.h
+++ b/configurations/Voxelab Aquila 427/UBL/Configuration_adv.h
@@ -2367,8 +2367,22 @@
   //#define ADVANCE_K_EXTRA       // Add a second linear advance constant, configurable with M900 L.
   //#define LA_DEBUG              // Print debug information to serial during operation. Disable for production use.
   //#define EXPERIMENTAL_I2S_LA   // Allow I2S_STEPPER_STREAM to be used with LA. Performance degrades as the LA step rate reaches ~20kHz.
+  #define SMOOTH_LIN_ADV            // Removes motion acceleration limitation by allowing to gradual ramp up of nozzle pressure
+  #if ENABLED(SMOOTH_LIN_ADV)
+    #define ADVANCE_TAU 0.01        // (secs) Smoothing time to reduce extruder acceleration due to linear advance.
+                                    // It is also the time ahead the smoother needs to look in the planner
+                                    // so the planner needs to have enough blocks loaded.
+                                    // for k=0.04 at 10k acceleration and an orbiter 2, it can be as low as 0.0075.
+                                    // The lower the better until stepper skipps
+                                    // Higher k and higher print accelerations may require larger tau to avoid skipping steps
+    #define SMOOTH_LIN_ADV_HZ 5000  // How often to update extruder speed
+    #if ANY(INPUT_SHAPING_X, INPUT_SHAPING_Y, INPUT_SHAPING_Z)
+      #define INPUT_SHAPING_E_SYNCH // Synchronize the extruder shaped xy axes (increrses precision)
+    #endif
+  #endif
 #endif

+
 /**
  * Nonlinear Extrusion Control
  *
```

So it's my normal settings which are the cause.

The source of the heartbreaking sounds in my version is the stepper motors because I physically disconnected the extruder motor and it can't make any sounds. 

[image]

 Finding a “bad” setting will take some time and I'm not even sure I will find it - if as stated XY motors should not be affected by the new algorithm, then the matter may be of a timing nature - well, by way of nonsense the timed interrupt handling function has become long and no longer has time to fit between interrupts, there may be a loop that sometimes takes a long time to execute and other things, and as a result I may get a good effect with one setting then another, but in fact the point is in the total number of them, not one by one.

But I'll try to find:)

dbuezas, 2025-04-11:
> Get well soon
Thanks!

Oh the Creality 4.2.7 has a STM32F103 mcu, which I think has no floating processing unit. 
This makes the new algorithm very slow since it is not optimised to use fixed point arithmetic (yet), and I have a monster 550MHZ board, so _"may be a loop that sometimes takes a long time to execute"_ is a likely reason for timing issues.

You could try reducing the linear advance interrupt frequency (e.g `#define SMOOTH_LIN_ADV_HZ 2500`), quality may or may not be affected tho.

yrabbit, 2025-04-11:
I'll give it a try.

Yes, MCU is weak, but there is a lot of free memory, both flash and RAM, which is literally wasted. Some tablets of pre-calculated pieces can go there;)

[25 more replies]