classicrocker883/MRiscoCProUI #203: Aquila G32 M29 Auto Save Mesh no longer closes Mesh screen

maggit13, 2025-03-17:
### Did you test the latest release build?

No, but I will try it!

### Bug Description

I use G29 in my Cura gcode to run a new mesh for every print, then M500 to save settings. Here is a sample from a print:

G92 E0 ; Reset Extruder
G28 ; Home all axes
G29 ; Run mesh leveling on every print
M500 ; Save settings to EEPROM
C108; Close the mesh viewer, optional

Up until 2.1.3f-11-1, Nov. 2024 build, it would run the mesh then save the settings and I would hear the beep and it would return to the print progress screen on the printer automatically. Since installing the Nov. build, PROUI with BLT, same options as before, it no longer beeps or closes the mesh window. It will start printing but I have to go to the printer and click continue manually. If I click Save, the screen flashes a few times and the print aborts.

I have reproduced this since installing the newer firmware for every print, I even refreshed all of recommended gcode options in Octoprint and in Cura. 

### Printer Model

Voxelab Aquila

### Model Type

OG

### Your Mainboard

Aquila GD32

### Add-ons that could be involved

BL Touch, Octoprint, Cura

### ProUI-EX | Extra Features?

Yes

### Bed Leveling

BLT - BL Touch Bilinear mesh

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

yrabbit, 2025-03-17:
Just out of curiosity, if you build mesh every time you print, why do you save it to EEPROM?

maggit13, 2025-03-17:
That's a great question, I guess in case I start another print that doesn't have that code in it. Just want to get something saved in case.

yrabbit, 2025-03-17:
> That's a great question, I guess in case I start another print that doesn't have that code in it. Just want to get something saved in case.

Ok, but somehow I doubt the M500 records mesh, shouldn't something like the G29 S be used?

maggit13, 2025-03-17:
I can certainly try it. My concern is that this worked perfectly before I updated the firmware a few months ago, so not sure what changed. What command is sent when clicking on save in the mesh screen?

yrabbit, 2025-03-17:
Well this is definitely not my area of expertise - I'm just a simple user here. It just seems to me that given that there are multiple slots for meshes, there would be a whole paragraph on the M500 page describing which slot the current mesh falls into or something like that. Since there isn't, the mesh probably just isn't saved by M500 😄 

https://marlinfw.org/docs/gcode/M500.html

maggit13, 2025-03-17:
Makes sense, I'll look into it. Thanks for the suggestion!

https://marlinfw.org/docs/gcode/M500.html

https://marlinfw.org/docs/gcode/M500.html

maggit13, 2025-03-17:
I found another issue that is pretty much what I'm experiencing, I have the C108 in my gcode and it used to close the menu after saving
[https://github.com/mriscoc/Ender3V2S1/issues/344](url)

classicrocker883, 2025-03-17:
> I found another issue that is pretty much what I'm experiencing, I have the C108 in my gcode and it used to close the menu after saving [https://github.com/mriscoc/Ender3V2S1/issues/344](url) 

some firmware files dont have the custom C gcodes. especially if its UBL and ProUI-EX because it takes up a lot of room. what firmware .bin have you used? also M420 is usually the right gcode when it comes to the mesh leveling. i mean i believe G29 might only work for *_BLT* versions.

but there have been some issues after the first October release so please try the most recent. but if you are using or need to use custom configuration let me know

maggit13, 2025-03-17:
Thanks. I use the BLT and PROUI EX bin file, not UBL. That makes sense about the October release since I'm having this problem with the November release. I will update to March and see what happens.

Thanks again!

maggit13, 2025-03-18:
I updated to March 2025 with BLT and ProUI EX, removed the C108 command and the first print still did not close the mesh menu. I then tried a few other files that I used in the past and suddenly it saved, beeped and closed the menu. These files had the C108 command but who knows. So seems to be working as expected now. Thanks for the suggestions!

On a side note, the progress bar is no longer working in the printer display. I have M73 Progress and Detailed Progress plugins in Octoprint but the progress stays at 0%. Not a big deal but curious as to why it's not working anymore.

Thanks again!

maggit13, 2025-03-21:
Awesome, thank you!

I'm not sure, it should work haven't touched that in a while, maybe something in an update changed it. ill get back to you.
ps. you can run a dry print run with a certain command M111 8
disable with M111 0
https://marlinfw.org/docs/gcode/M111.html

I'm not sure, it should work haven't touched that in a while, maybe something in an update changed it. ill get back to you.
ps. you can run a dry print run with a certain command M111 8
disable with M111 0
https://marlinfw.org/docs/gcode/M111.html

classicrocker883, 2025-03-31:
so this is what I've noticed regarding the print % bar. what happens after the screen times out and turns off, it basically freezes the image until you press the knob. turning the encoder knob may bring the screen back on, but you have to press the knob in to update the screen. if you do this after the print is finished it will select Done and bring you to the mainmenu so you wont be able to see the % bar changed. before the print ends if the screen if off just press the knob in and you should see it updated to the correct %, or disable LCD display timeout entirely (LCD timeout `0`).

let me know if that solves that issue. otherwise is there anything else?

maggit13, 2025-03-31:
Thanks for the update. My LCD timeout is always 0 so it's on all the time. Everything else updates normally during a print, the printer messages, positions, everything else, except the progress stays at 0% even though octoprint and my app move normally. I've tried reinstalling the m73 progress and detailed progress plugins but nothing is working.

classicrocker883, 2025-03-31:
oh I think when starting a print with Octoprint results in this behavior, because it is normal when starting a print through the LCD. did you mention that this only happened in this firmware, and works how to should like in Alex's or something?

I think it may have to do with printing from SD card vs through octoprint. so when you go to print, it goes through the usb rather directly from SD. if im able to figure out how to make the LCD update to octoprint that would be great.

maggit13, 2025-03-31:
Yes, I always print through octoprint and this behavior is new with this recent firmware version. In the past, your firmware has always worked on showing the progress on the lcd screen with the m73 progress plugin.

Thanks!

classicrocker883, 2025-05-11:
upon further review, @maggit13 have you checked your Gcode for `M73` commands?
the slicer should put these in like in [Gcode this test file](https://github.com/classicrocker883/MRiscoCProUI/blob/2025-May/slicer%20scripts/cura/SimpleCuraTest.gcode)

this should set the LCD display to show time remaining. if you do not see it update, try going into Tune menu and back, let me know if this works or if this is what you have to do to see it.

maggit13, 2025-07-09:
Hi, I did add the M73 commands in the gcode and the progress bar still showed 0% during a print. Not sure what's going on but the printer messages show the percentage and everywhere else, just not on the printer lcd display itself. Again, it did work up to some point then stopped without me changing any settings.

Thanks again

classicrocker883, 2025-07-10:
when checking the LCD, did you make sure it was not "frozen", that is it has a timeout and after some time it will turn off. try setting this in Control > Advanced Settings to `0` that will set it off, or even to something like `60` minutes / 1 hour. this will make sure the LCD stays on and updated

ill have to check for myself since its been some time since using OctoPrint

classicrocker883, 2025-07-13:
I think I found what is causing it.
originally it was `_percent_done = ui.get_progress_percent();`, and now it is `_percent_done = card.percentDone();`, so i can change it back and make a test firmware.

what was the .bin file you used ill make another one with the change if you want to try it out

maggit13, 2025-07-13:
Ok great, thanks. I'll try it out, this is the last one I installed Aquila_GD32_BLT-ProUI-EX-03-07.bin

Is there a way to install a new firmware without losing my settings?

Thanks again.
Harris

classicrocker883, 2025-07-14:
well the only thing the might save is the auto bed level settings. everything else youd have to re-enter, like your PID settings and such, you should enter M503 that'll show all the settings and save the output to a text file. i know some of those settings you can just enter in the terminal and it will reload them, off hand I dont know if the Gcode that it corresponds to is the same one you use to enter it back in.

classicrocker883, 2025-07-14:
[ Aquila_GD32_BLT-ProUI-EX-07-14.zip](https://github.com/user-attachments/files/21211942/Aquila_GD32_BLT-ProUI-EX-07-14.zip) test file

classicrocker883, 2025-08-01:
I tested this out and it seems to be working great. starting a print with Octoprint shows the progress % on LCD. so i will close this for now unless there is anything else.

closed 2025-08-01 by classicrocker883

maggit13, 2025-08-01:
Sounds good, I haven’t tried it out yet but I’ll take your word.

Thanks again!