classicrocker883/MRiscoCProUI #206: [BUG] wiki contain incorrect link to py scripts

elfl0rd, 2025-04-23:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

https://github.com/classicrocker883/MRiscoCProUI/wiki/How-to-generate-a-gcode-preview

First download the correct script/plugin:
    Orca/Prusa/Super Slicer
Use right-click and select "save link as...".

404

### Printer Model

None

### Your Mainboard

None

### ProUI-EX | Extra Features?

Yes

### Bed Leveling

UBL - Unified Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

elfl0rd, 2025-04-23:
side question... does ubl build has enabled g-code preview? Looks like no :(

classicrocker883, 2025-04-24:
yes they all should have gcode preview enabled, unless you're using HC32/H32 board, there is some issue with that enabled.
what is the issue with that?

and thanks for the heads up. link fixed.

elfl0rd, 2025-04-25:
Yes, HC32... Just saw that the warning about previews being disabled was gone and thought it was fixed

classicrocker883, 2025-04-25:
it's an odd bug. there are some things that didn't work and do work now. everything else works fine like in G32 and others, but that chip does keep getting updated through Marlin, and I do see pull requests regarding it so hopefully some day in the future that it can work.