classicrocker883/MRiscoCProUI #208: [FR] autodisable steppers at cancel print

elfl0rd, 2025-04-23:
### Describe the feature you want

Maybe you should disable the motors when manually canceling printing? Because if something goes wrong and printing is interrupted, the table will be cleared anyway, but this requires manually turning off the hold via the menu.
Or is there some reason for the hold to be on in such a case?

classicrocker883, 2025-04-24:
I think the way it is now it disables all but the Z axis, because say something happens halfway during a long print, instead of starting all over, what you can do is reslice the model, except cut away upto the point @ the height it stopped. so if the LCD says Z:5.50, you cut away everything below that and when it starts printing again it continues where it left off.

so that has actually happened to me and saved a long print. 

but I believe there *is* an option to disable this in Advanced Settings "Auto Abort Gcodes". perhaps I can change this to accommodate disabling all steppers instead.

thoughts?

elfl0rd, 2025-04-25:
Checked, canceled the print (I'm talking about manually canceling the print via the “stop print” button on the printer screen while printing) all stepper motors remained locked, including the X-Y axes. Can you tell me what “auto abort g-codes” does? Maybe I'm just missing functionality.
As an option, add a “disable hold” button to the “printing has been interrupted, confirm” info screen.

classicrocker883, 2025-04-25:
yes I'll look into that. let me try to remember exactly what this does.. it might be easier to just try enabling/disabling and do a test print to stop and see. I think it's like how it was original in Marlin when "auto abort gcodes" disabled. I'll get back to you

elfl0rd, 2025-04-25:
Hmm. I checked. Regardless of whether the "auto abort g-codes" setting is enabled when manually aborting a print, the printer raises the print head (z) slightly and shows a "printing was aborted" window with a single "confirm" button. All 3 axes remain locked.

classicrocker883, 2025-04-26:
OK I will look into fixing that. the X Y and E axis should disable at stop print. an update could have changed things to not work as it should. again thank you for letting me know.

classicrocker883, 2025-05-03:
problem doesn't appear to be ProUI related, so I opened an issue at Martlin: https://github.com/MarlinFirmware/Marlin/issues/27831

it appears that using `M84 XYE` (disable XYE) does nothing because it is "Shared with Z"...

ellensp, 2025-05-03:
Correct, it is all axis or nothing.   
 
The board has 1 enable io pin for all 4 stepper drivers.

It is physically impossible to disable just 1,2 or 3 drivers.

classicrocker883, 2025-05-03:
thats a bummer. well thanks for the update

classicrocker883, 2025-05-03:
I  posted a new release for [HC32 here](https://github.com/classicrocker883/MRiscoCProUI/releases/tag/2.1.3g-5-HC32)

let me know how you think about the abort print now. to do what you want it to do which is disable the motors when you abort, you have to disable _Auto Abort Gcodes_  
if you dont already know, navigate from the main menu to **Control**->**Advanced Settings**->**Auto Abort Gcodes** and uncheck the box.

elfl0rd, 2025-05-03:
Um.
Flashed HC32_UBL-ProUI-EX-05-03.bin unchecked Auto Abort Gcodes, aborted print - and all axis still locked. Also no Z movement up after aborting.
With checked Auto Abort Gcodes it moves Z up after aborting, but all axis still locked.

classicrocker883, 2025-05-17:
I updated the auto abort codes, so now it's named "Disable Motors on Abort" in advanced settings. checking this box should be able to do what you want to happen.

before it could not disable motors separately, that is why the old code did not disable any.

let me know if now that is what you're looking for.

classicrocker883, 2025-05-18:
ok thanks, I'll need some more testing.

classicrocker883, 2025-05-19:
so this should be fixed with the firmware I posted on the other issue

elfl0rd, 2025-05-19:
> ok thanks, I'll need some more testing.

Disabling steppers on abort work fine.