classicrocker883/MRiscoCProUI #211: [BUG] unable to store JUNCTION_DEVIATION_MM value

iomax, 2025-04-29:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

I'm upgrading my Aquila from the April 2024 release to the latest one (April 2025). However, no matter what I try (base feature: Junction Deviation and MM-ProUI-EX-IS-MCP  ), change to JUNCTION_DEVIATION_MM, whether made via the GUI or using the M205 command, does not persist after a power-off and restart.

Additionally, there's unexpected behavior with the theme files. The loaded colors do not match the C11 values specified in the theme configuration. For example, if the file specifies C11 E4 R0 G0 B14, after loading you might get R14 G63 B14 instead.

All of the above works correctly on the April 2024 release and I've also to add that a quick check on all Aquila_N32_MM-ProUI-EX-IS-MCP-xx-xx binaries released after April 2024 confirms that the "theme config file" issue has been present from the May 2024 release up to the latest one.

### Printer Model

Voxelab Aquila

### Model Type

Aquila ( original )

### Your Mainboard

Aquila N32

### Add-ons that could be involved

n/a

### ProUI-EX | Extra Features?

Yes

### Bed Leveling

MM - Manual Mesh Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2025-04-30:
MCP, you mean MPC. but what is the "theme config file" issue? are you using a precompile firmware? Junction Deviation isn't enabled unless you're using your own compiled firmware. i can revert whatever changes and it all can be fixed. just need to know what exactly.

iomax, 2025-04-30:
Sorry, you're right – MPC :-)
Junction Deviation is in my own compiled version, of course but regarding the theme "_gcode file_" ( [wiki](https://github.com/classicrocker883/MRiscoCProUI/wiki/Color-Themes ) ), I tried both my build and the precompiled ones (_Aquila_N32_MM-ProUI-EX-IS-MPC-xx.xx_), and it looks like the April 24 version is the last one that worked.

classicrocker883, 2025-04-30:
okay so the `C11` changing theme color and also Junction Deviation (`M205` to report value / `M205 J` to set value) worked on 2024-April? Taking a look back nothing code wise has jumped out - like no real changes I can see, but i can keep looking.

what if you tried `C11 E4 R1 B1 G14`? does that change correctly?

classicrocker883, 2025-04-30:
So I think the color/ `C11` should be fixed, try out the junction_deviation, and if you dont already know `PROUI_ITEM_JD` has to be enabled in **Configuration.h** - with `CLASSIC_JERK` disabled. I'm not sure how this wasn't working, someone else said they used JD and didnt report any issues. but let me know. and personally I use `CLASSIC_JERK`, but if you could help me understand more of the issue I can probably fix it. so far I looked back and compared to the current Marlin, I made a couple changes in the settings which *may* have caused the issue in the first place.

please try the new branch I created for testing:
- here is the [**branch**](https://github.com/classicrocker883/MRiscoCProUI/tree/2025-April-JD-Test)
- here is the [**PR**](https://github.com/classicrocker883/MRiscoCProUI/pull/212)

iomax, 2025-04-30:
I'm happy to help! Please give me a bit of time to try it out, and I'll get back to you as soon as I can. (Yes, I'm aware of the PROUI_ITEM_JD requirement 🙂)

iomax, 2025-04-30:
Quickly : JUNCTION_DEVIATION_MM issue .. solved ! Great !

The theme file / C11 issue is still here although the behavior is now a bit different. The theme file (the same one that worked in the April 2024 release) is still not applied. However, I now briefly see a quick flash with the expected colors before it switches back to the stored ones. Also, I tried commenting out the 'M117' line in the theme file, which resulted in an 'Invalid color element' message.

classicrocker883, 2025-05-01:
OK that's good news. so it seems even though you enter a supposedly valid number  it showed invalid so it restored or showed default colors.. I'll try going back to the actual original code and see if that works like it did.

classicrocker883, 2025-05-01:
I did add a short delay, 300ms. Just to be safe, but I can try removing it. either that it can be the `switch case` causing that. we're close so I'll get back to you when change is made.

classicrocker883, 2025-05-01:
OK I made a couple changes. Hopefully that works.

iomax, 2025-05-02:
Yes, it did the trick!  ... Well done ! :-)  Thanks.

classicrocker883, 2025-05-02:
very cool 👍🏻 , closing this. im open to any suggestions or if there is anything else i can do

commit 87afda9f (2025-05-02): Update - Fix Junction Deviation / C-code C11 (#212)

fixes #211
closed 2025-05-02 by classicrocker883