classicrocker883/MRiscoCProUI #213: [BUG] 1-5s Freeze at layer change

elfl0rd, 2025-05-13:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

I found that during layer change, the printer freezes for 1-5 seconds.
It hangs completely, even the screen stops responding to the encoder.
The problem is that it is leaking material, so the printer makes an overlap at the seam.
As you can see at the end of the video, the printer changed the layer (height from 1.4 to 1.6) while hanging and made a blot in the corner. The photo shows the blot on the substrate, also during the layer change.
The problem does NOT occur on BLT, only on UBL.
Firmware - HC32_UBL-ProUI-EX-05-12.bin

Can this problem be caused by a lack of memory, or is it something more specific?

### Printer Model

Voxelab Aquila

### Model Type

Aquila X2

### Your Mainboard

Aquila HC32

### Add-ons that could be involved

BL-Touch

### ProUI-EX | Extra Features?

Yes

### Bed Leveling

UBL - Unified Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[image]

[image]

[video_2025-05-13_20-51-55.zip](https://github.com/user-attachments/files/20193407/video_2025-05-13_20-51-55.zip)

classicrocker883, 2025-05-15:
very strange. did this happen on another release like April? also, have you tried with or without active leveling enabled (`M420` / `G29` not in the gcode, or `M420 S0` will disable it)

elfl0rd, 2025-05-15:
>very strange.
Agree

>did this happen on another release like April?
Yes, on different releases and on different 3d models "lag" time can be different, idk what affects.

>have you tried with or without active leveling enabled
yes, same behavior (active leveling really disabled - no flashing background on screen under Z)

[image]

classicrocker883, 2025-05-15:
could this be something with the slicer or its settings?
probably not if it doesnt happen using the BLT version... which makes me suspect its the actual HC32 chip.  
I could try some changes if you'd like to test the firmware. I might have a couple ideas but to start we need a baseline of something that *does* work. such as using Marlin Firmware specifically, without ProUI. that way I could narrow it down.

elfl0rd, 2025-05-15:
>could this be something with the slicer or its settings?

Definitely not, I tried with the native Voxelab slicer (VoxelMaker), which I stopped using (having switched to Orca) long before switching from the native firmware, same problem.

It has another problem by the way, the starting code (which cannot be edited).

M140 S80 T0
M104 S235 T0
M104 S0 T1
...
M190 S80 T0
M109 S235 T0
M109 S0 T1

Which results in heating up, starting printing and turning off the heat completely (while still printing) As I understand it, the firmware applies both T0 and T1 to the single heaters on the table and the hotend. (I edited the g-code file by removing the lines with T1 to check the freeze when changing the layer - it is present).

>probably not if it doesnt happen using the BLT version... which makes me suspect its the actual HC32 chip.

Chip HC32F460 KCTA Board V1.0.3.20210609

>I could try some changes if you'd like to test the firmware. I might have a couple ideas but to start we need a baseline of something that does work. such as using Marlin Firmware specifically, without ProUI. that way I could narrow it down.

Yes I would like to test I have time and a printer at hand. But I've never built firmware myself, so I'd need help with that, and ideally a ready file =^_^=.

classicrocker883, 2025-05-17:
you should be able to edit the .gcode file in something like Notepad.
if you could remove all the "T1" or "T0". find and replace all with *nothing*.

I think there was an issue from before involving this exactly.  
let me know if that has made any difference. 

but ill make a firmware file you can test so you don't have to worry about building it. it'll be vanilla Marlin, which will a bit different layout for UI, I will try to keep the settings about the same.

you may have to adjust them. but you can also make a .gcode file using the commands to 'automatically' save the settings for you so you don't have to enter each one through the LCD. 

what you do is if you have a raspberry pi w/Octoprint, or a computer/laptop w/Pronterface, or some Slicers can do this, connect to the printer with a USB cable, go to the terminal and enter `M503`

there you will see a whole bunch of settings. copy these, paste into a new blank document (.gcode)
Example :
```gcode
; M503 Sample Output
G21    ; Units in mm
M149 C ; Units in Celsius
Filament settings: Disabled
M200 D1.75
M200 D0
Steps per unit:
M92 X100.00 Y100.00 Z398.70 E100.00
Maximum feedrates (units/s):
M203 X400.00 Y400.00 Z8.00 E50.00
Maximum Acceleration (units/s2):
M201 X1000 Y1000 Z100 E10000
Acceleration (units/s2): P R T
M204 P400.00 R1000.00 T1000.00
Advanced: S T B X Z E
M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.30 E5.00
Home offset:
M206 X0.00 Y0.00 Z0.00
Auto Bed Leveling:
M420 S1 Z0.00
Material heatup parameters:
M145 S0 H195 B55 F0
M145 S1 H205 B60 F0
PID settings:
M301 P52.25 I5.60 D122.00
Z-Probe Offset (mm):
M851 Z-2.41 
```

now remove the lines that don't begin with a code, like "Steps per unit:".  
then at the very end put `M500` to save.

these are all your saved settings, you can use this every time you update your firmware.

to use, put the file name it like "settings.gcode" and go like you're gonna print the file

elfl0rd, 2025-05-18:
>you should be able to edit the .gcode file in something like Notepad.
if you could remove all the "T1" or "T0". find and replace all with nothing.

Yes, as I wrote, I already did that (otherwise I wouldn't have been able to check the blot at layer transition, since without editing, the plastic solidifies in the nozzle long before the first layer is complete).

>I think there was an issue from before involving this exactly.
let me know if that has made any difference.

Unfortunately no :) it is only in voxelmaker, which I hardly ever use.
I mentioned it because this behavior looks like a bug in the firmware (well, to the best of my understanding).
But there is freezing when changing a layer in voxelmaker and orca.

>these are all your saved settings, you can use this every time you update your firmware.

Done

elfl0rd, 2025-05-18:
Um.
Just for test I tried load settings from gcode file (As you tell me to create), but printer does endless homing circle untill I power it off.

classicrocker883, 2025-05-18:
can you show me your Gcode file? if it doesnt let you drop in the comment box, you can either make a zip archive, or copy the contents in a box like
````
```
; Gcode

your code here

```
````
otherwise I suppose you could just enter those settings via terminal, or through LCD

but I managed to make some test firmware.

[HC32_Marlin-Tests.zip](https://github.com/user-attachments/files/20271904/HC32_Marlin-Tests.zip)

- HC32_Marlin_UBL-Test1.bin
- - the one we need to know about most, if the issue of layer change freezing happens here
- HC32_Marlin_BLT-Test2.bin
- - this one just to test it is working fine.

so here are two files for testing the firmware.  
basically if **UBL-Test1** still doesn't work, then its for sure an HC32 issue.  
optionally, there is **BLT-Test2**, I would install/flash this 2nd if the first one has an issue.

elfl0rd, 2025-05-18:
```
; Gcode

G21 ; (mm)
M149 C ; Units in Celsius
M200 S0 D1.75
M92 X80.00 Y80.00 Z400.00 E98.00
M203 X800.00 Y800.00 Z18.00 E120.00
M201 X800.00 Y800.00 Z120.00 E1200.00
M204 P800.00 R800.00 T1000.00
M205 B50000.00 S0.00 T0.00 X10.00 Y10.00 Z0.40 E10.00
M206 X0.00 Y0.00 Z0.00
M420 S0 Z0.00 ; Leveling OFF
G29 W I0 J0 Z0.07550
G29 W I1 J0 Z-0.00800
G29 W I2 J0 Z-0.06100
G29 W I3 J0 Z-0.03850
G29 W I4 J0 Z-0.00450
G29 W I5 J0 Z0.00000
G29 W I6 J0 Z0.00000
G29 W I7 J0 Z0.00000
G29 W I8 J0 Z0.00000
G29 W I0 J1 Z0.09600
G29 W I1 J1 Z0.00600
G29 W I2 J1 Z-0.01700
G29 W I3 J1 Z-0.00100
G29 W I4 J1 Z0.01850
G29 W I5 J1 Z0.00000
G29 W I6 J1 Z0.00000
G29 W I7 J1 Z0.00000
G29 W I8 J1 Z0.00000
G29 W I0 J2 Z0.06900
G29 W I1 J2 Z0.01100
G29 W I2 J2 Z-0.03050
G29 W I3 J2 Z-0.02700
G29 W I4 J2 Z0.01900
G29 W I5 J2 Z0.00000
G29 W I6 J2 Z0.00000
G29 W I7 J2 Z0.00000
G29 W I8 J2 Z0.00000
G29 W I0 J3 Z0.04550
G29 W I1 J3 Z0.03250
G29 W I2 J3 Z-0.00400
G29 W I3 J3 Z0.01550
G29 W I4 J3 Z0.02300
G29 W I5 J3 Z0.00000
G29 W I6 J3 Z0.00000
G29 W I7 J3 Z0.00000
G29 W I8 J3 Z0.00000
G29 W I0 J4 Z0.00000
G29 W I1 J4 Z-0.01700
G29 W I2 J4 Z-0.04950
G29 W I3 J4 Z-0.04500
G29 W I4 J4 Z-0.02600
G29 W I5 J4 Z0.00000
G29 W I6 J4 Z0.00000
G29 W I7 J4 Z0.00000
G29 W I8 J4 Z0.00000
G29 W I0 J5 Z0.00000
G29 W I1 J5 Z0.00000
G29 W I2 J5 Z0.00000
G29 W I3 J5 Z0.00000
G29 W I4 J5 Z0.00000
G29 W I5 J5 Z0.00000
G29 W I6 J5 Z0.00000
G29 W I7 J5 Z0.00000
G29 W I8 J5 Z0.00000
G29 W I0 J6 Z0.00000
G29 W I1 J6 Z0.00000
G29 W I2 J6 Z0.00000
G29 W I3 J6 Z0.00000
G29 W I4 J6 Z0.00000
G29 W I5 J6 Z0.00000
G29 W I6 J6 Z0.00000
G29 W I7 J6 Z0.00000
G29 W I8 J6 Z0.00000
G29 W I0 J7 Z0.00000
G29 W I1 J7 Z0.00000
G29 W I2 J7 Z0.00000
G29 W I3 J7 Z0.00000
G29 W I4 J7 Z0.00000
G29 W I5 J7 Z0.00000
G29 W I6 J7 Z0.00000
G29 W I7 J7 Z0.00000
G29 W I8 J7 Z0.00000
G29 W I0 J8 Z0.00000
G29 W I1 J8 Z0.00000
G29 W I2 J8 Z0.00000
G29 W I3 J8 Z0.00000
G29 W I4 J8 Z0.00000
G29 W I5 J8 Z0.00000
G29 W I6 J8 Z0.00000
G29 W I7 J8 Z0.00000
G29 W I8 J8 Z0.00000
M145 S0 H210.00 B60.00 F0
M145 S1 H240.00 B75.00 F0
M301 P13.19 I0.77 D56.05
M304 P128.06 I24.95 D438.07
M255 S10 ; (minutes)
M256 B40
M207 S7.00 F2400.00 W13.00 Z0.20
M208 S0.00 F2400.00 W0.00 R480.00
M210 X3600.00 Y3600.00 Z480.00
M851 X-50.00 Y-10.00 Z-1.12 ; (mm)
M86 B0 E0 S360 T180
M900 K1.300
M603 L10.00 U120.00 ; (mm)
M412 S0 D25.00 ; Sensor OFF
C29 L10.00 R180.00 F10.00 B210.00 N T ; 5x5  T=180°C; Physical minimums:
C100 X Y; Physical maximums:
C101 X Y Z; Bed size:
C102 X Y; Max Extruder temperature:
C104 T; Park Head:
C125 X Y Z; Filament runout sensor:
C412 M ; Active mode: HIGH; Invert Extruder:
C562 E ; No inverted; Toolbar configuration:
C810 A B C D E; Probe Z Speed and Multiple Probing:
C851 S Mok
M500
```

elfl0rd, 2025-05-18:
>HC32_Marlin-Tests.zip 

should I reflash _screen_ for it?

classicrocker883, 2025-05-18:
no need to reflash, once is all you need unless there is an update, which there hasnt one in over a year

> > HC32_Marlin-Tests.zip
> 
> should I reflash _screen_ for it?

classicrocker883, 2025-05-18:
so I dont know why it would want to Home for that Gcode, but lets remove some code that isnt necessary,  
delete the following lines:
```
G21 ; (mm)
M149 C ; Units in Celsius
M420 S0 Z0.00 ; Leveling OFF

; and all the ones starting with G29
G29 W I0 J0 Z0.07550
...
G29 W I8 J8 Z0.00000

; not sure why some C-codes have no numerical values,   
; but we can remove all those as well
C29
...
C851
```

then just to satisfy the Homing if it keeps going and going, add `G28` to the beginning, and `G4 S5` which will add 5sec delay:
```
; Gcode settings
G28
G4 S5
M200 S0 D1.75
... and so on ...
```

yrabbit, 2025-05-18:
> M420 S0 Z0.00 ; Leveling OFF
> G29 W I0 J0 Z0.07550
> G29 W I1 J0 Z-0.00800
> G29 W I2 J0 Z-0.06100
> G29 W I3 J0 Z-0.03850
> G29 W I4 J0 Z-0.00450
> G29 W I5 J0 Z0.00000
> G29 W I6 J0 Z0.00000
> ```

I could be wrong and that's fine, but with UBL homing is not infinite at all - it's just that every time you give a G29 mentioning I and J there is a calibration, no? 

https://marlinfw.org/docs/gcode/G029-ubl.html

classicrocker883, 2025-05-18:
UBL and Bilinear (BLT) leveling are different even in their Gcode, so when you save your bed level offsets it will be totally different from one to another unfortunately. itll be best just to rebuild the Auto Mesh once youre ready to print, for testing it should be fine leaving all the values to 0.0 or Resetting to Zero. i mean it seems like there is very little variation for these values to begin with, such as +/-0.10 difference should be fine. Sometimes ill just print without leveling on/enabled because I releveled the bed and dont feel like going through the whole auto leveling thing and print first layers are fine, especially when its a small print and in the center.

classicrocker883, 2025-05-18:
ok i see why the C100 and so on C-codes dont give values, thats because `HAS_CGCODE` is disabled, it takes up a ton of memory, and for HC32 with UBL and ProUI-EX, it wont fit. not a big deal, just wont be able to read/write those things in terminal/Gcode

elfl0rd, 2025-05-18:
Okay, after clearing gcode it applies correctly.
I flashed HC32_Marlin_UBL-Test1.bin

Looks like screen wants a reflash, but still able to test print

[image]

elfl0rd, 2025-05-18:
hm.
Looks like it work fine on this firmware.
printing without creating mesh is fine, creating mesh now for next try.

elfl0rd, 2025-05-18:
I cant create bed mesh on marlin.
bed leveling just closes in middle at probing.

elfl0rd, 2025-05-18:
Yeah, same issue as before, so it is HC32 issue.

classicrocker883, 2025-05-18:
oh I had forgot about the thing to change which takes the LCD Icons from another file, that is why the screen is without them, not to worry.

since the issue remains on Marlin then yes, should be reported there.
also, for now switch to the good working firmware that was BLT right?

if theres anything else let me know

elfl0rd, 2025-05-18:
> also, for now switch to the good working firmware that was BLT right?

Yep, with HC32_BLT-ProUI-EX-05-12.bin all fine

> since the issue remains on Marlin then yes, should be reported there.
reference this issue there which should help anyone trying to fix it.

I can do it, but I'll need help with nuances like
Describes your environment
Include information about:
    The Marlin version (please test [the Development branch] before submitting reports)
    Configuration.h and Configuration_adv.h files (in a ZIP file)

A bit complex to me, lol.

classicrocker883, 2025-05-18:
yea I can help out with that. 
so I fixed the Disable Motors on Abort issue, i can make a firmware for you
upon looking through some differences between UBL and BLT, I found something which may have contributed to the issue of it freezing.

Ill post another firmware if you wanna try and see

classicrocker883, 2025-05-18:
[HC32_BLT_UBL-05-18.zip](https://github.com/user-attachments/files/20275409/HC32_BLT_UBL-05-18.zip)

here is the updated firmware, _BLT doesnt have any real change, except Disable motors on Abort (renamed to Disable on Abort, moved up on list), it should be working.  
then there is _UBL, only change is I disabled `NONLINEAR_EXTRUSTION` which I suspect may be the issue, if not, then its for sure with Marlin

elfl0rd, 2025-05-18:
> if you wanna try and see

Ofc! Thx you :)

I'll test it both after sleep :)

elfl0rd, 2025-05-19:
>then there is _UBL Test, only change is I disabled NONLINEAR_EXTRUSTION which I suspect may be the issue, if not, then its for sure with Marlin

First print without mesh - success.
Second print with mesh - success.

I'll print something bigger and take a look.

classicrocker883, 2025-05-19:
[HC32_BLT_NONLIN-Test.zip](https://github.com/user-attachments/files/20281568/HC32_BLT_NONLIN-Test.zip)
BLT with `NONLINEAR_EXTRUSION` enabled - if fails, then that would be issue with UBL

elfl0rd, 2025-05-19:
>BLT with NONLINEAR_EXTRUSION enabled - if fails, then that would be issue with UBL

Is there any difference in screen meny between this BLT and previous UBL? I flashed it, looks similar.
Mesh created, test print started :)

elfl0rd, 2025-05-19:
Small print with HC32_BLT_NONLIN-Test.bin fine.
Starting larger one (I'll return after hour).

elfl0rd, 2025-05-19:
Still fine on HC32_BLT_NONLIN-Test.bin 🤷‍♂️