classicrocker883/MRiscoCProUI #216: [BUG] Freezing during bed leveling (G29), especially when using the inductive probe

Dimetroslav, 2025-05-23:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

Hello MRiscoCProUI Team! 
Английский

I used the firmware: Aquila_N32_IND_UBL-ProUI-EX-05-11 , but I also tried other firmware that generally work, but the problem remains:(in short: when calibrating the table, the hotend can start moving downwards, ignoring the operation of the inductance sensor, or press down slightly and move to the next measuring point). I tried to build the firmware myself, but I can't do anything at almost all stages, I don't have enough knowledge.
I’m requesting official support for my Voxelab Aquila S2 Pro 3D printer with the N32G455REN7 MCU (Nation STM32-compatible Cortex-M4F).

Key Issues:

Freezing during bed leveling (G29), especially when using the inductive probe.

The firmware needs adaptation for my hardware configuration.

Technical Details
1. Hardware Specifications:
Printer: Voxelab Aquila S2 Pro

MCU: Nation N32G455REN7 (ARM Cortex-M4F @144 MHz)

Mainboard: FFP0173_Aquila_Main_Board_V1.0.1 (20200811)

Probe: Inductive sensor (offset: X-49 Y-32 Z-1.5)

Extruder: Direct drive (steps: E93.00)

### Printer Model

Voxelab Aquila

### Model Type

Aquila X2 Pro 

### Your Mainboard

Aquila N32

### Other Mainboard Type

FFP0173_Aquila_Main_Board_V1.0.1

### Add-ons that could be involved

IND/BLT/LA/

### ProUI-EX | Extra Features?

Yes

### Bed Leveling

UBL - Unified Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Clock Configuration.cpp.txt](https://github.com/user-attachments/files/20412915/Clock.Configuration.cpp.txt)
[platfomio.ini.txt](https://github.com/user-attachments/files/20412916/platfomio.ini.txt)
By the way, I added a ferrite ring, it didn't help, and there is also an incomprehensible patch between the sensor and the motherboard.

[image]

[image]
[image]
[image]
[image]

Dimetroslav, 2025-05-24:
Good afternoon.  Everything works perfectly on the native firmware, but it does not support LA, I only have bin files for the old firmware, there is no source. When calibrating the table, the hotend may go down ignoring the sensor's operation (probably system suspension), it may not push hard and move to the next point, or it may push until the steps on the stepper motor are skipped, you have to turn off the entire printer urgently, or a message appears on the screen:Printer damage turn off immediately. The LED on the sensor always lights up when metal is approaching and goes out, as it was on the old (stock) firmware, the sensor works the same way on the new one. Multiple sensor checks are performed with perfect accuracy. I didn't change the bed. The red and black wires do not lead to anything, a long time ago I tried to install the firmware from Alex Aquila X2 without IND sensor support, they used to go to the self-made Z end cap button :) It didn't work out ,

Dimetroslav, 2025-05-24:
Today I ran the G29 P1 test from the terminal, the first time it passed without errors and interruptions, I was already happy, I ran it again to make sure that the G29 P1 command was working as it should, but I was disappointed the printer crashed.

[image]

Mik-S-UMS, 2025-05-24:
The probing sequence seems a bit odd to me but I don't have an ABL so have nothing to compare it to. I just thought it did it a line at a time on a grid but could be different for UBL.

I thought the sensor was not triggering till late which could mean there is something between the bed surface and the aluminium plate around the back left corner, but seeing it doing it a second time like at 5:26 you can clearly see the probe is triggered (light goes off) but the printer does not react and can hear the click as the nozzle hits the bed, and it happens many more times.

At a guess I think your bed may not be well levelled to begin with so when probing it is outside the range it can compensate for, but this should not continue to drive the nozzle into the bed so there is definitely a firmware issue here.

Try running the tramming wizard to get the bed better physically levelled and see if this improves probing.

classicrocker883, 2025-05-25:
I can upload a vanilla Marlin version of the firmware, this will help narrow down the issue if its on this ProUI repo, or something deeper with the Marlin source code.

Dimetroslav, 2025-05-25:
I am a former electronics technician and repaired laptops and computers. I have a feeling that all this is due to incorrect frequencies and timings integrated into the firmware for my chip.`#include "stm32f4xx.h"

void SystemClock_Config(void) {
  RCC_OscInitTypeDef RCC_OscInitStruct = {0};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {0};

  // Configure HSE (8MHz external crystal)
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSE;
  RCC_OscInitStruct.HSEState = RCC_HSE_ON;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSE;
  RCC_OscInitStruct.PLL.PLLM = 8;
  RCC_OscInitStruct.PLL.PLLN = 144;
  RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV2;
  RCC_OscInitStruct.PLL.PLLQ = 4;
  HAL_RCC_OscConfig(&RCC_OscInitStruct);

  // Configure clock dividers
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK|RCC_CLOCKTYPE_SYSCLK
                              |RCC_CLOCKTYPE_PCLK1|RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV4;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV2;
  HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_4);
}`

classicrocker883, 2025-05-25:
[Marlin_N32_IND_UBL-Test.zip](https://github.com/user-attachments/files/20429489/Marlin_N32_IND_UBL-Test.zip)

Dimetroslav, 2025-05-25:
Perhaps you need to use such specific flags for the N32G455 chip: 
`build_flags =
  -DN32G455xx
  -DUSE_HAL_DRIVER
  -DHSE_VALUE=8000000
  -DHSI_VALUE=8000000
  -DHAL_ADC_MODULE_ENABLED
  -DHAL_TIM_MODULE_ENABLED`

classicrocker883, 2025-05-25:
whoops I forgot about the ICON, uses 9.ICO, not 7.ICO.  

[Marlin_N32_IND_UBL-Test1.zip](https://github.com/user-attachments/files/20431222/Marlin_N32_IND_UBL-Test1.zip)

Dimetroslav, 2025-05-25:
I agree with you. This [firmware:](https://github.com/user-attachments/files/20431222/Marlin_N32_IND_UBL-Test1.zip) I also passed the table calibration test several times, no freezes, no errors, everything was done well.  I've done tests on the M48, and the result is good. I noticed some oddities on the screen: the test was performed on 49 points and for some reason it shows 37,It seems that the printer does not check all the points when calibrating the table. But I am glad that there are no collisions of the nozzle with the table and there are no critical errors or firmware freezes. Sometimes, after calibration from the terminal with the G29 command, there is an invalid Mesh message on the screen.
Test: Tilt Mesh for some reason performs only on three points (a triangle, two front on the sides and one with the back in the center)

classicrocker883, 2025-05-25:
[ProUI_IND_UBL_Tests.zip](https://github.com/user-attachments/files/20431703/ProUI_IND_UBL_Tests.zip)

here are some more tests using this ProUI.

one is NoProUI, which doesn't use ProUI-EX,  
Test1 and 2 I reverted some things.

let me know if any work

Dimetroslav, 2025-05-25:
I will check all test firmwares and write in detail about the results of their work. Only I will be able to answer tomorrow after work 😓, it is 1:30 am now. Thank you for doing this.

Dimetroslav, 2025-05-26:
All three firmware versions were checked, and the result was good. No. 3. Each one passed 3 calibrations of the table, the M48 test, and two calibrations from the terminal (G29 P1).  The first firmware calibrates faster, of course, but after changing the allowed maximum table position of 230 on the X-axis, the printer began to malfunction during calibrations and the nozzle knocked on the table (3 failures were noticed). The second one showed a slightly better result, with the Tilt Mesh at the last point, the nozzle rested on the table. and two touches of the table. The third one didn't seem to fail, I even tried to print a part on it: it worked.

Dimetroslav, 2025-05-27:
Dear classicrocker883, could you send me the files of the third test firmware before compilation, I want to find out why I couldn't build and compile it. I also want to add parameters and settings to it right away so that I don't have to reconfigure the printer after other firmware updates every time (sometimes I need to print something urgently)

Dimetroslav, 2025-05-28:
Thank you. Yesterday I compiled several firmware versions, adding only the settings that I entered through the terminal. With these firmware updates, the hotend hits the table again or crushes it. perhaps I got confused and did not make changes to the stm32f1-maple.ini, today I will check and try again to install the Test 3 firmware and then calibrate it several times to make sure.

classicrocker883, 2025-05-29:
I have a feeling it very may well be with the probe offset and size of the bed causing the issue.

so by not adjusting any parameters there is no failure and it is fine? but once you change them this causes the problem?

so, mriscoc has come out with a new release, here is a test file. the menu item placements are not in the usual area, but you should be able to find all you need.

[NewProUI_IND_UBL_Test.zip](https://github.com/user-attachments/files/20503100/NewProUI_IND_UBL_Test.zip)

let me know if this test works

Dimetroslav, 2025-05-29:
You were absolutely right in your guesses. :

I installed the firmware and set the offsets for the probe.  the miracle did not happen. I carried out calibration several times (5). There were always failures or touches of the bed-nozzle.  I started setting different values for the number of points on the X and Y axes, and after each change, I performed a calibration test. I noticed that the touches were no longer random and occurred strictly in the center of the X-axis, but along the entire Y-axis, it became easier here, I calculated approximately and adjusted the dimensions: the safe displacement of the probe and the number of points.  This is the sign that came out:

1. Calculation of the safe boundaries for the sensor (in nozzle coordinates):
By X:
Max. Sensor X = X_MAX_POS + offset_x = 230 + (-49) = 181 mm
Offset from the right edge of the table:
225 - (181 - 0) = 44 mm > 10 mm
(Explanation: nozzle at X=181 → sensor at X=132 mm from left edge → to right edge 225-132=93 mm)

By Y:
Max. Sensor Y = Y_MAX_POS + offset_y = 245 + (-32) = 213 mm
set MESH_MAX_Y=208 (a careful approach).
Indentation from the front edge of the table:
225 - (208 - 0) = 17 mm > 10 mm 
(Nozzle at Y=208 → sensor at Y=176 mm from the rear edge → to the front edge 225-176=49 mm)

Calculation of the 5×6 grid:
By X (5 points):
Range: 0 → 181 mm
Step: 181 / (5 - 1) = 45.25 mm
Nozzle coordinates: 0, 45.25, 90.5, 135.75, 181.
By Y (6 points):
Range: 0 → 208 mm
Step: 208 / (6 - 1) = 41.6 mm
Nozzle coordinates: 0, 41.6, 83.2, 124.8, 166.4, 208.

the parameters (X_MAX=230, Y_MAX=245, grid 5×6, borders 181×208) are mathematically optimal:
Match the physics of the printer and sensor.
Exclude points where the sensor is off the table.
Provide safe margins ≥10 mm.
A balance between accuracy (25 points) and calibration speed.
Physical size of the table is 225x225 mm Actual printing area
X_MAX_POS 230 mm Max. X movement (mechanical limitation)
Y_MAX_POS 245 mm Max. Y movement (mechanical limitation)
Sensor offset (-49, -32)	Sensor to the left and back of the nozzle
Sensor requirement ≥10 mm Minimum distance from the edge of the table
Scan boundaries X: 181, Y: 208 Maximum nozzle coordinates during calibration
Grid points 5×6 5 by X, 6 by Y

After that, I ran about 10-15 tests with and without preheating. There was one touch on the right edge, but I think it's a coincidence or within acceptable limits.

[OctoPrintLog.txt](https://github.com/user-attachments/files/20506569/OctoPrintLog.txt)