classicrocker883/MRiscoCProUI #217: [BUG] Aquila x3 is not homing after flash

afgp87, 2025-05-30:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

Hi,

Thank you for all the hard work you're putting into this project — I really appreciate it!

I wanted to try out the firmware version Aquila_N32_IND_UBL-ProUI-EX-05-11, but for some reason, the Z-probe doesn't trigger during auto-homing. I've tried everything I can think of, but nothing seems to work.

I reverted back to the stock firmware, and the Z-probe works properly there. When I check the probe status, it shows as "open," and when I bring a piece of metal close to it, it detects and triggers correctly.

Any help or suggestions would be greatly appreciated. Thanks again!

### Printer Model

Voxelab Aquila

### Model Type

x3

### Your Mainboard

None

### ProUI-EX | Extra Features?

Yes

### Bed Leveling

UBL - Unified Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Mik-S-UMS, 2025-05-30:
What exactly is it doing when homing Z? Does it only move up? 
Did you use the switch diagnostic in Mriscoc to see it the sensor is being detected as triggered or was this on stock FW? 
Does the same thing happen in an earlier version of the FW?

afgp87, 2025-05-30:
> What exactly is it doing when homing Z? Does it only move up? Did you use the switch diagnostic in Mriscoc to see it the sensor is being detected as triggered or was this on stock FW? Does the same thing happen in an earlier version of the FW?

afgp87, 2025-05-30:
Yes in the MRiscoC the probe is being detected and if I put something metal trigger the z, but when I do home z it just go up not down on the stock fw, It goes down and trigger z, do you have any idea how to fix it

Mik-S-UMS, 2025-05-30:
I'm not sure but could be related to #216, from what I understand it may be the offsets are not correct but from the OP of that issue it was homing but would not detect at certain spots.

Definitely try a going back a few versions to make sure there is not a bug that has crept in here recently and make sure the filename of the .bin you use contains `_IND`
If that suddenly works then you can move to the next release and see which one breaks.

afgp87, 2025-05-30:
> Could you try the NoPro test in this zip file and let me know if the issue is there?
> https://github.com/user-attachments/files/20431703/ProUI_IND_UBL_Tests.zip

Thank you Sir. I will try tonight and let you know

classicrocker883, 2025-05-30:
on second thought, the probe not triggering make me think that the connection is not in the right ports. such as if you look at the BL or CR touch, one way of connecting is it has a 3pin and a 2 pin connector, and some have a single 5 pin. so this could be in a different place where it is supposed to register the trigger.

afgp87, 2025-05-30:
> on second thought, the probe not triggering make me think that the connection is not in the right ports. such as if you look at the BL or CR touch, one way of connecting is it has a 3pin and a 2 pin [...]

I'm using a 3-wire probe (black, brown, and blue wires). The probe itself appears to be functioning correctly: it lights up when open and turns off when triggered. custom fw MRiscoCProUI also recognizes the probe's status, showing 'open' in green and 'triggered' (by metal) in red. So, the probe is definitely recognized and working.
However, when I try to home the printer with the custom firmware, the print head moves to the center of the bed and then moves up instead of down. When I revert to the stock firmware, the printer homes correctly, moving the print head down until the probe is triggered

Mik-S-UMS, 2025-05-30:
> on second thought, the probe not triggering make me think that the connection is not in the right ports. such as if you look at the BL or CR touch, one way of connecting is it has a 3pin and a 2 pin [...]

The X3 has 3 wires in a 5 way plug for the BLtouch port so this will be correct, it does not need the last 2 as there is no probe to deploy. It uses the optocoupler board to convert the trigger signal from the sensor to the BLtouch port.

One thought I had was that somehow the configuration of the latest build had mixed up the trigger pin as BLtouch uses `OUT` but the inductive sensor only uses `IN`. As this was working properly for people before this is my guess guess for this version.

However if the screen diagnostic shows the probe triggering I don't know what could be at fault as otherwise the hardware is working correctly. 

Maybe that for homing it is using the wrong input for Zmin.

@afgp87 did you try an earlier version?

afgp87, 2025-05-30:
> The X3 has 3 wires in a 5 way plug for the BLtouch port so this will be correct, it does not need the last 2 as there is no probe to deploy. It uses the optocoupler board to convert the trigger [...]

I try all 3 files and all the versions back till November, M48 probe test failed 
[image]

https://github.com/user-attachments/ahttps://github.com/user-attachments/assets/72c746a9-38be-4d5f-895d-0755e7485779

https://github.com/user-attachments/assets/49479acc-eb9c-4ec7-a22a-0520716b8d5d

ssets/9f51f49d-e627-4857-ac5d-9a356d165ceb

afgp87, 2025-05-30:
https://github.com/user-attachments/assets/be159455-1a47-42ce-bf8f-94a8f45f708a

Mik-S-UMS, 2025-05-30:
Not quite sure if this is it but `Z_probe` should show it being `closed` (pulled low) when the sensor is not triggered then `open` (pulled high) when it is. I don't have an ABL so don't know if what this shows is correct.

When the Z limit switch, ABL or sensor is open this is considered as being triggered as this is normally closed so the nozzle does not drive into the bed if it is disconnected, this results it in only moving up when homing.

That thread explains the same thing that says the signal needs to be inverted for this sensor.

> I also needed to revert Z_MIN_ENDSTOP_HIT_STATE and Z_MIN_PROBE_ENDSTOP_HIT_STATE to HIGH

Could you try going back to versions even earlier than November as this has been working fine for others with the same printer.

Z-offset looks to be about right from what I have seen of this sensor but not sure of the typical value.

afgp87, 2025-05-31:
How i can do this i really don't understand how to do anything in marlin there is something like the printer.cfg? And how i access to that or if any one can do it a past me the .bin [image]

classicrocker883, 2025-05-31:
ah ok it seems to me that this is the issue:
the trigger is inverted, even though it may show correctly for `Open` and `TRIGGERED`

could it be possible that the probes pin connector is backwards?  
can you take a look, take a picture if you want and we can compare to another.  
even from factory sometimes they could have swapped the pins around.

[is this the same one?](https://www.amazon.com/dp/B08K8NC5WC), it doesnt show the actual connector, but does show the pinout

because the firmware appears to work fine for others using the same one, as @Mik-S-UMS mentioned in the other recent similar issue, using the same exact firmware.

did you get this printer new or was it someone elses before you?

classicrocker883, 2025-05-31:
[N32_IND_UBL-Inverted-Test.zip](https://github.com/user-attachments/files/20532738/N32_IND_UBL-Inverted-Test.zip)

I made a test you can try, I just inverted to
```
#define Z_MIN_ENDSTOP_HIT_STATE LOW
#define Z_MIN_PROBE_ENDSTOP_HIT_STATE LOW
```

afgp87, 2025-06-01:
> [N32_IND_UBL-Inverted-Test.zip].
Today I install this and unfortunately didn't work i also install klipper to try and surprisingly it did the same, klipper show the probe is working correctly open and triggered when i put a metal under the sensor, but when I do home it still goes up, I took a few pictures of the boar and the 3 wires goes to small pcb and on one red wire goes to the main power and the other end 3 wires goes to the sensor black brown blue [image]

[image]

[image]

[image]

[image]

[image]

Mik-S-UMS, 2025-06-01:
Everything looks correct wiring wise. 
I am suspecting it could be a hardware problem on your printer if the same thing happens to Klipper and it is configured properly, but I don't know what could be causing it. I found [this](https://www.reddit.com/r/VoxelabAquila/comments/1bbd641/comment/kutwbuq) post on how to set up the probe in Klipper so you can confirm it is correct.

It may be an issue with that converter board if the signal is not fully set high or low it might not be enough for the FW to read it as triggered. The board has pull up resistors to limit the 5V supply from the BLtouch port to 3v3 for the signal but the CPU also has internal pull up resistors that can be enabled so maybe the problem is with this.

From pictures I have seen of this board there is a LED on it but not sure exactly how it reacts. I think it will be fully lit when the sensor is not triggered (when the light on the sensor itself is on) but will be dim when triggered. I have not seen this working so it is mostly a guess. Could you make a video of what this LED does when you trigger the sensor?

classicrocker883, 2025-06-01:
> I am suspecting it could be a hardware problem on your printer if the same thing happens to Klipper and it is configured properly, but I don't know what could be causing it.

this

classicrocker883, 2025-06-07:
thats great news!

Mik-S-UMS, 2025-06-07:
Its good that it is working but I am worried if there is any fault on the sensor it will send 24v straight into the CPU frying it. That board was an isolator to protect the motherboard.

This does make me think the issue was with this converter board and it was not giving a clean signal through the opto-coupler.

Is there any resistance between the black and blue wires when not triggered? AFAIK this acts like the normally closed contacts on a relay but not sure if this is the case or some electronic switch. If it is the latter then it makes it more likely a fault will send 24v though the signal wire.

afgp87, 2025-06-08:
> thats great news!

Is working perfectly thank yo [image]

I have sherpa mini do i have to do any changes i trying to find the rotation distance there is a setting to input the new e-steeps

Mik-S-UMS, 2025-06-10:
I don't think it will be the thermistor but may have to do with the reference voltage for it. This also might be the reason why homing was not working properly, or the way the inductive sensor is wired up is affecting the 3v3 voltage rail.

The supply voltage on the BLtouch port is 5v but the `IN` signal is expected to be 3v3 so cutting out the converter board means it now has 5v on here and could mess up the thermistor ref voltage. I suspect that unplugging the sensor will make the temps correct.

classicrocker883, 2025-06-11:
try the firmware that has  *_SPDY5*  Spider Speedy Ceramic Hotend

afgp87, 2025-06-11:
> try the firmware that has  *_SPDY5*  Spider Speedy Ceramic Hotend , this uses thermistor # 5

Perfect did a quick test with the the firmware  spdy5 and it work extruder is Extruding properly, mil gracias por toda tu ayuda