classicrocker883/MRiscoCProUI #220: [BUG] Bad Auto Mesh Leveling on Aquila X2 N32

gusogon, 2025-09-21:
### Did you test the latest release build?

Yes, and the problem still exists.

### Bug Description

Hi, my name is Gustavo

Hopefully you can help me, i've been trying to run perfect mesh Leveling with 3DTouch for almost 2 years💀

I have a Voxelab Aquíla X2 (Board: N32 N32G455) with aftermarket 3D Touch. (Firmware "Aquila_N32_UBL-ProUI-EX-08-24")

Studied the wiki (https://github.com/classicrocker883/MRiscoCProUI/wiki) and after setting all the physical parameters and gcode start with M420(also tried G29A... and same result); i follow this sequence:

-Bed Tramming (to adjust bed screws)

-Auto Mesh Generating (dont know if It has to be centered in the bed, but none of it works correctly)

-Adjusting Z-Offset firstly with a paper to avoid hits and (more accurate) when printing bed test.

•My conclusion: It applies the Mesh, but either displaced or (more probable) completely wrong, maybe because of interferences. (Last time i had to edit the mesh adding 1mm on the right side of the bed to compensate this error)

I've read that in some Ender 3V2 the way to fix this is using a Probe support with 0 Y offset (MRiscoC said this is because of twisted x axis), so this is the last change i'm going to try. 

Some others say there is nothing to do but editing the Mesh manually; If you have any additional information, I would love to read it.

Thanks in advance for your help,
Best regards.

### Printer Model

Voxelab Aquila

### Model Type

Aquila X2 N32

### Your Mainboard

Aquila N32

### Add-ons that could be involved

3D-Touch

### ProUI-EX | Extra Features?

Yes

### Bed Leveling

UBL - Unified Bed Leveling

### Did you include your own configuration files?

- [ ] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2025-09-22:
so the issue is then the mesh isn't as level as you'd like it? the print layer could be better?

here is what I would do.
- disable auto leveling, remove-comment out `M420 / G29A` , or instead of `M420 S1` just do `M420 S0`
- make a test print, something 1 layer high, either in the center of the bed or the whole bed wide.
- observe how this prints without any mesh leveling to gauge what sort of mesh correction you may need.
- start over, zero all the mesh points.
- - do an Auto Build Mesh in Leveling menu

the probe should do it's job and get the new value for every point. depending on what your offset it and how many points (like 7x7, 5x5) you may notice that one column of points (right side) may not get probed, this is okay because it will automatically calculate the missing values pretty well. and you can also manually change them later.

you should be able to add more mesh inset to the sides as well. and if you have bed clips, you want to make sure this is enough to clear them.

if you think that the bed isn't quite leveled/trammed:
- do your best bed tramming again. this is how I do it - without using paper 
- 1. what I do first is make sure your Z-offset is true
- 2. go to the Z-offset wizard and have it so the nozzle just touches the bed. lower the nozzle until it does but just barely the surface. you can also use paper in between, but subtract the thickness to compensate for the offset. not as accurate than using without in my opinion.
- 3. go to Tramming Wizard, use the Manual Tramming option checkbox. go to each corner and adjust the knobs so the nozzle and bed are just about to touch.
- 4. then check the center, rehome Z, and double check the 4 corners and center again

you may need to readjust the Z-offset to get it as close as you can.  
also, when doing the test print, be prepared you may need to do change a live Z-offset

gusogon, 2025-09-24:
Dear Andrew, sorry for not responding before.

I already printed Bed level test without mesh correction and it gets a little better. 

Don't know if what you said is previous to get a better Auto-Mesh or if is an alternative way

Wanted to know if there is a way to tramming and meshing automatically with the 3DTouch 
(Nowadays, even with 9x9 mesh, i get an error of almost 1mm [compared with getting nozzle to bed in those points] in the right part of the bed] Checked all offsets and gcode; But still getting this huge error that ruins bed level tests)

Thanks for your time and help!
Gustavo

Mik-S-UMS, 2025-09-24:
> Wanted to know if there is a way to tramming and meshing automatically with the 3DTouch

If you are using the UBL version putting `G29 P1`  in your start Gcode should do this. If it is the BLT version the just `G29` should work.
This is instead of using `M420 S1` as that applies the current mesh in memory instead of automatically probing.
Make sure this is after G28 (home)

You should run the tramming wizard first though to get the bed as level as possible so the mesh does not need to make large corrections.

classicrocker883, 2025-09-25:
if you get that much of an error, you want to make sure the Z offset is set correctly. basically when you manually make Z=0.00 on the Prepare>Move menu movie Z axis or the Z Offset Wizard, the nozzle should be just touching the bed. this is how the printer knows where to print. so any little variations on the bed can be corrected with Auto mesh leveling. but remember you may be able to actually probe the last row (or two with 9x9), you can tell by watching it do the auto build.

if there is such a huge error, you can still fix it by manually editing the mesh point. adjust the height by manually setting it